opencardev/crankshaft #325: How to connect with mobile via bluetooth.

ShashankAdur, 2019-03-23:
If you need support: Please post on https://www.reddit.com/r/crankshaft/ as this is a bug tracker, not a support forum.

Note: Reporting issues not based on official published builds will be closed without any comment.

- What phone do you have?one plus 6-

- What OS version?latest

- What crankshaft version?alpha 4

- How to connect Crankshaft using Android Auto to mobile phone via Bluetooth?
  I can able to connect it to mobile via Hot Spot.

andrewip1027, 2019-04-02:
sorry, but does this post mean that bluetooth still not available yet? only usb can connect to the Raspberry pi to activate AA ? 

A new guy passing by sorry for stupid questions.

hawkeyexp, 2019-04-22:
You can connect your phone via cable or via wifi to cs for the android auto porjection.
Audio is routed via bluetooth.

closed 2021-11-27 by github-actions[bot]