opencardev/crankshaft #326: Lux value on main window doesn't update

hfd9876, 2019-03-23:
If you need support: Please post on https://www.reddit.com/r/crankshaft/ as this is a bug tracker, not a support forum.

Note: Reporting issues not based on official published builds will be closed without any comment.

- What phone do you have? Pixel 2

- What OS version? Android P

- What crankshaft version? 299C16C 2019-03-03

- Steps to reproduce the bug, if you're planning to report a bug. Please indicate whether the bug is always repoducible.

Install and enable light sensor (confirmed working OK - night mode enabled correctly etc.).

Go to TSL2561 tab and enable option "Show Lux value from TSL2561 on Main Window. Save settings. 

Initial lux value displayed but doesn't update as lux value changes. Is this by design or is it supposed to update?

- Please provide any further information that you might find helpful if available.

- Starting with pre4: Boot into dev mode (can be selected in settings).
  Reproduce the possible bug. Create the debug logs by pushing the grey button on bottom of screen.
  Add the created file (/boot/debug.zip) to the issue.

  Warning: if you have configured wifi passwords they will be blanked by script during creation.
  We can't make absolutly sure you passwords are removed so do a personal check of the files inside the zip.

Grawa, 2019-03-25:
Same problem,i solved it turning on the RTC (select ds3231,even if it's not connected obviously) then reboot it and after about 1/2 minutes the light sensor will work

hawkeyexp, 2019-04-22:
seems i have to check the check if i2c is active to read from sensor :-)

hawkeyexp, 2019-05-10:
is your sensor @ i2c address 0x39 ?

hfd9876, 2019-05-10:
Hi

Yes it is.

Thanks
Simon

hawkeyexp, 2019-06-08:
released

closed 2019-06-08 by hawkeyexp