opencardev/crankshaft #327: VIDIOC_REQBUFS error 12 when firing cam_overlay

enriquedelpino, 2019-03-27:
Hi there,

I'm just trying to get started using crankshaft to display the rear cam in my car. I'm getting a memory issue with the webcam which I'm using to test my setup, and I am not sure whether the problem is coming from my setup, or rather if this is a problem someone has come across before. 

After failing to start the rearcam using the command `crankshaft rearcam show` I tried to dig into crankshaft's code and figure out what is the ultimate command to be run in order to get the USB webcam up and running. I noticed its done through executing cam_overlay, but in any case I'm getting this `VIDIOC_REQBUFS error 12 when firing cam_overlay`

The details of my setup are:
RasbperryPi 3B+
CrankshaftNG Alpha-4 | 2019-03-03-crankshaft-ng-299C16C (the latest public release)
LifeCam HD-3000

So the results of the execution are:
```
pi@CRANKSHAFT-NG:/opt/crankshaft/cam_overlay $ ./cam_overlay.bin 
Opened device '/dev/video0'
Dimensions: 640 x 480 Pixel Format: YUYV
VIDIOC_REQBUFS error 12, Cannot allocate memory
```

Also I can provide the details of the camera used:
```
pi@CRANKSHAFT-NG:/opt/crankshaft/cam_overlay $ v4l2-ctl --all
Driver Info (not using libv4l2):
	Driver name   : uvcvideo
	Card type     : Microsoft® LifeCam HD-3000: Mi
	Bus info      : usb-3f980000.usb-1.3
	Driver version: 4.14.98
	Capabilities  : 0x84200001
		Video Capture
		Streaming
		Extended Pix Format
		Device Capabilities
	Device Caps   : 0x04200001
		Video Capture
		Streaming
		Extended Pix Format
Priority: 2
Video input : 0 (Camera 1: ok)
Format Video Capture:
	Width/Height      : 640/480
	Pixel Format      : 'YUYV'
	Field             : None
	Bytes per Line    : 1280
	Size Image        : 3085365528
	Colorspace        : sRGB
	Transfer Function : Default
	YCbCr/HSV Encoding: Default
	Quantization      : Default
	Flags             : 
Crop Capability Video Capture:
	Bounds      : Left 0, Top 0, Width 640, Height 480
	Default     : Left 0, Top 0, Width 640, Height 480
	Pixel Aspect: 1/1
Selection: crop_default, Left 0, Top 0, Width 640, Height 480
Selection: crop_bounds, Left 0, Top 0, Width 640, Height 480
Streaming Parameters Video Capture:
	Capabilities     : timeperframe
	Frames per second: 0.003 (1/353)
	Read buffers     : 0
                     brightness (int)    : min=30 max=255 step=1 default=-8193 value=142
                       contrast (int)    : min=0 max=10 step=1 default=57343 value=5
                     saturation (int)    : min=0 max=200 step=1 default=57343 value=83
 white_balance_temperature_auto (bool)   : default=1 value=1
           power_line_frequency (menu)   : min=0 max=2 default=2 value=0
      white_balance_temperature (int)    : min=2800 max=10000 step=1 default=57343 value=4500 flags=inactive
                      sharpness (int)    : min=0 max=50 step=1 default=57343 value=25
         backlight_compensation (int)    : min=0 max=10 step=1 default=57343 value=5
                  exposure_auto (menu)   : min=0 max=3 default=0 value=3
              exposure_absolute (int)    : min=5 max=20000 step=1 default=156 value=156 flags=inactive
                   pan_absolute (int)    : min=-201600 max=201600 step=3600 default=0 value=0
                  tilt_absolute (int)    : min=-201600 max=201600 step=3600 default=0 value=0
                  zoom_absolute (int)    : min=0 max=10 step=1 default=57343 value=0
```

hawkeyexp, 2019-04-22:
possible you can try to assign more gpu mem in /boot/config.txt if it possible will help.

hawkeyexp, 2019-04-23:
seems to be duplicate of #300 -> closing

closed 2019-04-23 by hawkeyexp