opencardev/crankshaft #334: Crankshaft displayed on LCD. OpenAuto open on HDMI.

SHOmike, 2019-04-08:
Hi.
I am unable to get openauto to open onto the LCD.
I boot the Pi, crankshaft loads and its shown onto the LCD but when i connect the phone nothing happens. But if the HDMI is plugged into the Pi, openauto will open with the LCD resolution on my TV.

What i have:
-samsung s8, android 9
-Raspberry pi 3+ model b
-LCD TFT 3.5" KUMAN ( http://www.kumantech.com/kuman-35quot-inch-480x320-rgb-tft-lcd-pixels-touch-screen-with-touch-pen-for-raspberry-pi-3-2-model-b-b-a-a-module-spi-interface-sc06_p0014.html )
-2019-03-03-crankshaft-ng-299C16C

what i have done:
-installed cranshaft
-installed LCD driver(which put "dtoverlay=tft35a" in /boot/config.txt)
-changed some adjustment(enabled X11, changed line from "driver "fbturbo" " to "driver "fbdev"" in /usr/share/X11/xorg.conf.d/99-fbturbo.conf ,calibrated touchscreen)

this got me to have crankshaft working properly on LCD, but plugging the phone into the Pi will display openauto through HDMI whether or not it is plugged into something.(touch features still works properly on LCD display)

I attached debug files.HDMI version is with the HDMI cable plugged in. just in case it is needed.
[debug-HDMI.zip](https://github.com/opencardev/crankshaft/files/3054539/debug-HDMI.zip)
[debug.zip](https://github.com/opencardev/crankshaft/files/3054540/debug.zip)

Thanks a lot 
Mike
-Sorry for my english-

vsvipul, 2019-04-16:
Have the same problem.. need help for a project.. @SHOmike if u find something, please comment here.. I need help

SHOmike, 2019-04-16:
Sorry.no way with the 3.5 lcd.i bought a 4" hdmi lcd from waveshare, higher resolution, quite similar dimension.plugged in and worked straight away.

vsvipul, 2019-04-16:
Can you point me to a link for the 4 inch display you used? And what modifications were required in that to make it work? Did it come with an hdmi cable?

SHOmike, 2019-04-16:
Ok.so the lcd is this one:
https://www.amazon.com/Raspberry-Resistive-Supports-Revsions-Provided/dp/B078N8SHHQ/ref=mp_s_a_1_1?keywords=waveshare+4+inch&qid=1555414570&s=gateway&sr=8-1

Donno where u from so i picked it from amazon.com.

You only need to activate x11 mode, and add a couple of line on 10-evdev.conf which is located on usr/share/x11/xorg.conf.d.

If you get the LCD, let me know i will send you what i did in order to have it fully working. Sorry but now i am on my phone i dont have my stuff here to check.

vsvipul, 2019-04-16:
@SHOmike ok sure.. thank u for your help.. I am ordering the LCD. Please post the changes you made (When you are free and have access to your computer) as a comment here or you can mail me to vsvipul555@gmail.com . Thanks for your help.

vsvipul, 2019-04-19:
@SHOmike can u please list the changes you made. I finally got the LCD. can't wait to try it out. Thanks.

vsvipul, 2019-04-20:
@SHOmike ping

SHOmike, 2019-04-20:
yep.sorry.
1.install crankshaft

2.while you add your wifi's credential to the file /boot/crankshaft/crankshaft_env.sh, also change START_X11 and DEV_MODE and DEV_MODE_APP values from 0 to 1.

3.ssh into the pi

4.add these lines at the end of the file /boot/config.txt       

hdmi_group=2
hdmi_mode=87
hdmi_cvt 480 800 60 6 0 0 0
dtoverlay=ads7846,cs=1,penirq=25,penirq_pull=2,speed=50000,keep_vref_on=0,swapxy=1,pmax=255,xohms=150,xmin=200,xmax=3900,ymin=200,ymax=3900
display_rotate=3

5 make the relevant section look like this:

Section "InputClass"
        Identifier "libinput touchscreen catchall"
        MatchIsTouchscreen "on"
        MatchDevicePath "/dev/input/event*"
        Driver "libinput"
        Option  "SwapAxes" "1"
        Option  "InvertX"       "1"  
        Option  "InvertY"       "0"
        Option  "Calibration" "208 3905 288 3910"
EndSection

6.reboot and try it out.

vsvipul, 2019-04-20:
I followed all the above steps but The display doesn't seem to light up. It requires external power , right? I did that from my 2 phone adapters to both pi and display. What should i do? @SHOmike .. any help?

vsvipul, 2019-04-20:
@SHOmike working fine now.. Thank you so much.. Just need to callibrate the touch.

Grandement, 2019-05-01:
Hi guys,
I got it to work. I modified the tutorial for the retropie project. https://www.hackster.io/112938/kuman-tft-3-5-retropie-2018-a9b0c8

What I did:

sudo apt-get install libraspberrypi-dev

wget https://raw.githubusercontent.com/adafruit/Raspberry-Pi-Installer-Scripts/master/adafruit-pitft.sh

chmod +x adafruit-pitft.sh

git clone https://github.com/goodtft/LCD-show.git

 sudo chmod +x adafruit-pitft.sh
 cd LCD-show
 sudo chmod +x LCD35-show

sudo ./adafruit-pitft.sh

Options:
----- 4 -----
----- 1 ----- 90
----- n ----- select no to console settings
Select Y for HDMI mirror
DO NOT REBOOT!!!!!!!! N

cd LCD-show
sudo ./LCD35-show

Finish :) 

libraspberrypi-dev is required because adafruit run into a missing File error: bcm_host.h

Touchscreen works as well. Let me know if it worked at your side as well.

MarcGodard, 2019-06-25:
This got my waveshare screen to work: https://github.com/opencardev/crankshaft/issues/351

closed 2021-11-27 by github-actions[bot]