opencardev/crankshaft #335: Is this project still alive?

Maju3, 2019-04-08:
If you need support: Please post on https://www.reddit.com/r/crankshaft/ as this is a bug tracker, not a support forum.

Note: Reporting issues not based on official published builds will be closed without any comment.

- What phone do you have?

- What OS version?

- What crankshaft version?

- Steps to reproduce the bug, if you're planning to report a bug. Please indicate whether the bug is always repoducible.

- Please provide any further information that you might find helpful if available.

- Starting with pre4: Boot into dev mode (can be selected in settings).
  Reproduce the possible bug. Create the debug logs by pushing the grey button on bottom of screen.
  Add the created file (/boot/debug.zip) to the issue.

  Warning: if you have configured wifi passwords they will be blanked by script during creation.
  We can't make absolutly sure you passwords are removed so do a personal check of the files inside the zip.

Last post from hawkeyexp is over 1 month ago.
So i really wonder about the state of this project.

parkerlreed, 2019-04-08:
Upstream (OpenAuto/aasdk) stagnated and went their own route for a "pro" version. It's speculated that they are updating it for the paid version but not pushing the changes to the public repo.

Crankshaft is tying those projects into its own experience. I suspect there isn't much to maintain at the moment. Open source projects can take a beackseat depending on what life throws at us. I don't believe the project is dead.

Maju3, 2019-04-08:
Sure.
Crankshaft is a open source project and normal life has priority.
Anyway the last post from hawkeyexp was 1 month ago.
That makes me wondering whats going on.
Maybe he lost interest in this?

Maju3, 2019-04-08:
A statement from hawkeyexp about this would be helpful.

JiseOT, 2019-04-09:
> Upstream (OpenAuto/aasdk) stagnated and went their own route for a "pro" version. It's speculated that they are updating it for the paid version but not pushing the changes to the public repo.
> 
> Crankshaft is tying those projects into its own experience. Open source projects can take a beackseat depending on what life throws at us. I don't believe the project is dead.

I am using both 'Openauto Pro' and 'Crankshaft' in my cars. I can confirm that guys from Open Auto Pro are alive, maintain and develop their software very good. They release updates every few weeks. 
On the other hand 'Crankshaft' is free and i hope it will be still developed.
I wrote to blue studio with question when they update open source version of the openauto. Maybe this will be some trigger for other projects.

hawkeyexp, 2019-04-22:
I'm still present but some things of real life needs more time. I don't lost the interest on CS but it is what it is - a free project which get's love if the time allows it :-)

Maju3, 2019-04-23:
Nice to see that hawkeyexp is still present :)
issue closed ;)

closed 2019-04-23 by Maju3