opencardev/crankshaft #336: WiFi and Bluetooth Connections won't work immediately

renkemeyer, 2019-04-29:
So when i start Crankshaft on the newest Version my Phone can not connect with the USB Cable, because i would connect and disconnect and connect over and over again until Crankshaft crashes. But that might be an issue with the Phone. When i want to connect over WiFi on the Phone it says Authentification failed like 5 times before it connects. The password ist correct (1234567890). When AA finally started it can happen that my Phone disconnects and the audio is stuck (annoying sound bug) so i have to restart the pi.

Another Issue is that when i connect the phone over Bluetooth, without AA it ask every time on the phone if i really want to connect (ok or cancel box) like the phone was never connected before. And everytime i have to try connecting multiple times because my phone says "can not connect) i tryed with pairing enabled and disabled.

Phone: Galaxy S8 European Verison 
Pi: normal Raspberry Pi 3B

Sorry for the bad english, i'm from Germany

hawkeyexp, 2019-05-10:
Beschreib es mir noch mal in deutsch :-)

alextytan, 2019-06-18:
u can flash latest img on your sdcard, let it boot. in crankshaft's settings (the cog wheel) go to the last tab (hw something) and choose for bluetooth either builtin or external, depending on what bluetooth u want to use. save, reboot and try to connect via bluetooth from android device. that's what i did with a oneplus3t mobile

closed 2021-11-27 by github-actions[bot]