opencardev/crankshaft #337: Google won't let me navigate anywhere, I have a USB mic and it is detected in crankshaft

ThePacificGamer, 2019-05-01:
If you need support: Please post on https://www.reddit.com/r/crankshaft/ as this is a bug tracker, not a support forum.

Note: Reporting issues not based on official published builds will be closed without any comment.

- What phone do you have?
LG G3 (D850)
- What OS version?
CRDroid with android 9 (Custom ROM)
- What crankshaft version?
Latest
- Steps to reproduce the bug, if you're planning to report a bug. Please indicate whether the bug is always repoducible.
Me: Hey google
Google Shows up
Me: Navigate to so and so
Google won't do anything
- Please provide any further information that you might find helpful if available.
I have a USB Mic plugged in
- Starting with pre4: Boot into dev mode (can be selected in settings).
  Reproduce the possible bug. Create the debug logs by pushing the grey button on bottom of screen.
  Add the created file (/boot/debug.zip) to the issue.

  Warning: if you have configured wifi passwords they will be blanked by script during creation.
  We can't make absolutly sure you passwords are removed so do a personal check of the files inside the zip.

Maju3, 2019-05-02:
Take a look into your PHONE settings for google assistant.
There is a option to select voice input between bluetooth or CABLE headset.
To use your usb mic you need to select cable headset.

ThePacificGamer, 2019-05-02:
It doesn't seem to be working even when I enable cable in the assistant

Maju3, 2019-05-02:
Then i have no other idea,sorry.

egisz, 2021-01-12:
Hi, I have same issue, USB microphone, sound tests are working fine.
AA running on RPI4, hotspot mode.
Google assistant is recognizing words like "navigate home", but it nothing happens. I believe AA doesnt have internet connection, that's why. If I connect RPI to home LAN, then it works fine. But not in the car.
I can search addresses only using keyboard. This is annoying...

matt2005, 2021-01-12:
I'll take a look when my pi4 arrives. I accidentally shorted out my old one.