opencardev/crankshaft #339: Android auto WIFI not working

EVPower2k13, 2019-05-02:
If you need support: Please post on https://www.reddit.com/r/crankshaft/ as this is a bug tracker, not a support forum.

Note: Reporting issues not based on official published builds will be closed without any comment.

- What phone do you have?
Samsung S9

- What OS version?
Android 9.0

- What crankshaft version?
Alpha-4 2019-03-03

- Steps to reproduce the bug, if you're planning to report a bug. Please indicate whether the bug is always repoducible.

I enable crackshaft to theter to my phone and use the developper setting to enable android auto to start the the head server unit.

I had the 4.1 android auto version that seem to not be able to connect more than 1 week of usage. Android auto is showing on the raspberry but when I try to press on the icon nothing append. I erase the car, clear the data and cache. Didn't work. I uninstall everything than reconfig it, then work for again 1 week. I try to latest update 4.2 of android auto and it work for 2 days. Try the delete car, clear date, cache, uninstall et reinstall but nothing works. Same issue.
Android auto is showing but when clickng on it, nothing append 

Hope that maybe you'll be able to find something in the debug.

Thanks for your help.
BTW, Thank for your work for this project.

[debug.zip](https://github.com/opencardev/crankshaft/files/3140270/debug.zip)

- Please provide any further information that you might find helpful if available.

- Starting with pre4: Boot into dev mode (can be selected in settings).
  Reproduce the possible bug. Create the debug logs by pushing the grey button on bottom of screen.
  Add the created file (/boot/debug.zip) to the issue.

  Warning: if you have configured wifi passwords they will be blanked by script during creation.
  We can't make absolutly sure you passwords are removed so do a personal check of the files inside the zip.

EVPower2k13, 2019-05-09:
Seem after more than one week, the android auto and crackshaft decided to talk to each other. I received 2 last prompt to complete the android setup. Now, it works wireless.

Will keep you posted if the case close after 1 week of usage.

Thanks.

hawkeyexp, 2019-05-10:
i remember some longer time ago i had a similar effect casue the prompt was not in foreground so not visible to accept. Nothing where cs can something do.

EVPower2k13, 2019-05-14:
Thanks hawkeyexp for your reply. I can understand. I will close the thread even of the problem is still occurring. Hope to have more in log to be able to fix it.

closed 2019-05-14 by EVPower2k13