opencardev/crankshaft #344: .iso file not readable

agnoam, 2019-05-23:
The latest release: `Crankshaft OpenCarOS NG Alpha-4 2019-03-03` not working on etcher.
and the windows cannot open it.
The error is: the file is corrupted or unreadable

parkerlreed, 2019-05-23:
Seems to work fine here. Are you using the img? You mentioned iso

[image]

agnoam, 2019-05-23:
yes it is

agnoam, 2019-05-23:
But I got this error from Etcher: 
[image]

What should I do ?

parkerlreed, 2019-05-23:
Maybe try Win32DiskImager/Rufus? I haven't dealt with Windows in so long, I'm honestly not sure.

agnoam, 2019-05-23:
Thanks a lot its worked with Rufus 👌

closed 2019-05-24 by agnoam