opencardev/crankshaft #346: Attaching steering wheel button to the Pi

agnoam, 2019-05-25:
Is there a way to activate the Google Assistant by my steering wheel voice command button ?
I also have an answer and rejection call buttons in my steering wheel, can I use them too ?

[image]

toufou54, 2019-05-31:
Same for me, I have an ibus USB interface and nodejs.
How can I interface it ?

gcharlie, 2019-06-03:
I don't think this is something that can be addressed in the software and will vary vehicle to vehicle. Ultimately it's a hardware issue and the easiest way to make use of these buttons would be to trace the leads for the button and connect them to gpio pins. This is harder on newer vehicles that make use of more integrated pcb and fragile ribbon cables.

toufou54, 2019-06-03:
Hi,
I wanted to know how to programaticely interface with AA and crankshaft.
My USB interface coupled with node can interprète and decide ibus from my vehicle, I just don't want use gpio but addressing directly by soft.
How can't I retrieve all crankshaft commands triggered by gpio ?
I saw a lot of script named service.xxxx in opt but it seems not to be full commands.
For example I already could launch the rearcam command when a capture ibus event from that car thru :
crankshaft rearcam show | hide
But how simulate AA navigation for example or Google voice activation ?

ryanjblajda, 2019-06-13:
use an arduino, create a voltage divider to read the resistive SWC and connect it to your PI using USB and sending keyboard commands. That is what i am doing. I am still trying to figure out how to control the GPIO pins, but i think i will use a solid state relay (controlled by arduino) to connect the raspberry pi volume GPIO pins to ground.

priitpi, 2019-06-17:
> use an arduino, create a voltage divider to read the resistive SWC and connect it to your PI using USB and sending keyboard commands. That is what i am doing. I am still trying to figure out how to [...]

I solved it same way. No need to mess with GPIO. With arduino HID project you can send keyboard media commands like volume up and down. https://github.com/NicoHood/HID

in arduino you can send command like this (with HID library added):
`Consumer.write(MEDIA_VOLUME_UP);`

Then define key in triggerhappy.conf  (boot -> crankshaft -> triggerhappy.conf) and add

`
KEY_VOLUMEUP		1		amixer -q sset 'Master' 5%+
KEY_VOLUMEUP		2		amixer -q sset 'Master' 5%+
KEY_VOLUMEDOWN	1		amixer -q sset 'Master' 5%-
KEY_VOLUMEDOWN	2		amixer -q sset 'Master' 5%-
`
I think triggerhappy let's to define some keyboard combination aslo if you wish to skip setting it up acting as media keys.

meekys, 2019-06-18:
Plenty of examples out there, I used this one as a starting point.
https://www.instructables.com/id/Digispark-Volume-Control/
Then created another that did the whole voltage divider thing to figure out which button was pressed and send that command instead, then it's just like using the keyboard shortcuts in OpenAuto/Crankshaft.

ryanjblajda, 2019-06-18:
will this allow the volume controls to work outside of Android Auto? with the GPIO they only work when it's running correct?

malcolm-fendick, 2019-06-18:
Just tested and it worked in and out of Android Auto.

ryanjblajda, 2019-06-18:
awesome.

ryanjblajda, 2019-06-20:
> Just tested and it worked in and out of Android Auto.

did you remove the GPIO mapping? i just tried this and it doesnt seem to work. the other keyboard mappings seem to work 'H' brings me to the audio screen in the main menu for some reason, but it works, but the Media keys are not.

malcolm-fendick, 2019-06-20:
I'm using an Arduino Micro. The keyboard input is via the Arduino's USB connection which is independent of GPIO configurations.

ryanjblajda, 2019-06-20:
> I'm using an Arduino Micro. The keyboard input is via the Arduino's USB connection which is independent of GPIO configurations.

i am using an an Adafruit trinket, doing the same thing. using the same library. i wrote a basic sketch to just increment the volume as a test and i didnt see it working.

priitpi, 2019-06-21:
> > I'm using an Arduino Micro. The keyboard input is via the Arduino's USB connection which is independent of GPIO configurations.
> 
> i am using an an Adafruit trinket, doing the same thing. using the same library. i wrote a basic sketch to just increment the volume as a test and i didnt see it working.

I'm also using Pro Micro which has  32U4 chip. Your device has different chip and probably isn't supported by HID project

You said before that you are still able to send keyboard commands. Maybe just try sending it as function keys or any other key that Trinket can send. Then configure triggerhappy.conf 

e.g vol up = F7, vol down = F8
`KEY_F7	1	amixer -q sset 'Master' 5%+`
`KEY_F7	2	amixer -q sset 'Master' 5%+`
`KEY_F8	1	amixer -q sset 'Master' 5%-`
`KEY_F8	2	amixer -q sset 'Master' 5%-`

ryanjblajda, 2019-06-21:
> > > I'm using an Arduino Micro. The keyboard input is via the Arduino's USB connection which is independent of GPIO configurations.
> > 
> > 
> > i am using an an Adafruit trinket, doing the same thing. using the same library. i wrote a basic sketch to just increment the volume as a test and i didnt see it working.
> 
> I'm also using Pro Micro which has 32U4 chip. Your device has different chip and probably isn't supported by HID project
> 
> You said before that you are still able to send keyboard commands. Maybe just try sending it as function keys or any other key that Trinket can send. Then configure triggerhappy.conf
> 
> e.g vol down = F7, vol up= F8
> `KEY_F8	1	amixer -q sset 'Master' 5%+`
> `KEY_F8	2	amixer -q sset 'Master' 5%+`
> `KEY_F7	1	amixer -q sset 'Master' 5%-`
> `KEY_F7	2	amixer -q sset 'Master' 5%-`

it works fine, i plugged the trinket into my windows PC and it responded to the Media keys.

ryanjblajda, 2019-06-25:
just wanted to let anyone who cares know, this is working, i set up exactly as described using media keys and it works great.

iofirag, 2019-06-30:
With arduino?

thrill85, 2019-07-04:
Doesnt have anyone an examplecode to impliment this for a can bus based SWC?

closed 2020-10-02 by matt2005