opencardev/crankshaft #347: Using vnc to use tablet or phone display instead of lcd

toufou54, 2019-05-31:
- What phone do you have?
OnePlus 3T
- What OS version?
8.1
- What crankshaft version?
Latest (2019-03-03)

I'm using a kulan 7' display

Using x11vnc, when connecting remotely with a vnc viewer, works fine until Android auto launch: the x11 server always display crankshaft interface instead Android auto.

- Please provide any further information that you might find helpful if available.

I understood that the Android auto is on another layer, but is it possible to stream the screen too ? If yes : how ?

Thank you

hawkeyexp, 2019-06-01:
The video stream from android auto is pushed to the gpu directly so no chance.

closed 2019-06-01 by hawkeyexp

toufou54, 2019-06-03:
ok so I have a try with dispmanx_vnc, after some very quick configuration it's working perfectly :)

So like we said here: "impossible n'est pas Français" ;)

 https://github.com/hanzelpeter/dispmanx_vnc

ankorez, 2019-12-08:
@toufou54 Hello. peux tu stp nous dire comment tu as fait ? J'ai installé dispmanx mais je n'arrive pas à le lancer 

`pi@CRANKSHAFT-NG:~/dispmanx_vnc $ ./dispmanx_vncserver
Open display[0]...
Display is 1920 x 1080
Server bpp:16
Server bigEndian:0
Server redShift:0
Server blueShift:10
Server greeShift:5
08/12/2019 07:30:07 Listening for VNC connections on TCP port 5900
08/12/2019 07:30:07 rfbListenOnTCP6Port: error in bind IPv6 socket: Address family not supported by protocol
open /dev/uinput returned -1.
First write returned -1.
ioctl UI_DEV_CREATE returned -1.
Error create uinput device -1.`