opencardev/crankshaft #348: Crankshaft NG on 10.1inch LCD touchscreen

tesmnorth, 2019-06-08:
Hello,

I am using last version of Crankshaft NG with raspberry pi 3 and waveshare 10.1 inch LDC touch screen. Touch screen is working very well and screen size is ok. But, there is a problem like this: 

[image]

There is a blue rectangle in the middle of screen. How can I fix this?

Thanks.

hawkeyexp, 2019-06-08:
it's the non scaled default background of 800x480. As quickfix replace the images in /boot/cran kshaft-*.png with resized to your native screen size.

hawkeyexp, 2019-06-08:
for next release a auto resize will be added

tesmnorth, 2019-06-08:
> for next release a auto resize will be added

It will be more user-friendly. Thanks. Now, I resized png files.

tesmnorth, 2019-06-08:
I have an one more question about touch screen. If I want to click any button by touch screen, button is not working. But, If I double-click, It is working. I search any option for this. But I couldnt find. How can I resolve this double-click problem.

hawkeyexp, 2019-06-08:
bg resize released. for double clock issue please create a new issue. thx.

closed 2019-06-08 by hawkeyexp