opencardev/crankshaft #351: X11 Mode Problem

tesmnorth, 2019-06-10:
Unfortunely, X11 mode is not working. I changed START_X11=0 to START_X11=1 in `crankshaft_env.sh` file. I need to X11 mode for touching screen of my lcd screen.

JoelBrenstrum, 2019-06-21:
I had a similar issue. I found if I booted in EGL, SSH'd in and enabled X11 then ran 
`sh /opt/crankshaft/service_openauto.sh` I would get a crash message.
setting the environment variable 
`export QT_DEBUG_PLUGINS=1` 
then gave me detailed logging.

I found I needed to install 
`sudo apt-get install libxcb-icccm4 libxcb-image0 libxcb1 libxcb-keysyms1 libxcb-render-util0 libxcb-xinerama0 libx11-xcb1`
all these libs and then it worked! 

Once you get it successfully working via SSH you can reboot with X11 turned on. 
This allowed me to run `xinput-calibrator` to get my LCD touch working correctly. 

Hopefully this helps

chucktastic, 2019-06-23:
In my case I had to install all the packages JoelBenstrum lists, plus libqt5x11extras5.  This was with Alpha-5.

MarcGodard, 2019-06-25:
This did the trick for me :) Thanks guys.

tesmnorth, 2019-06-25:
Thanks @JoelBrenstrum . But we need to install  libqt5x11extras5 as @chucktastic. Thanks to all.

closed 2019-06-25 by tesmnorth