opencardev/crankshaft #352: Touch Screen stopped working after Google Play Service updated

Stargazzer, 2019-06-21:
If you need support: Please post on https://www.reddit.com/r/crankshaft/ as this is a bug tracker, not a support forum.

Note: Reporting issues not based on official published builds will be closed without any comment.

- What phone do you have?

- What OS version?

- What crankshaft version?

- Steps to reproduce the bug, if you're planning to report a bug. Please indicate whether the bug is always repoducible.

- Please provide any further information that you might find helpful if available.

- Starting with pre4: Boot into dev mode (can be selected in settings).
  Reproduce the possible bug. Create the debug logs by pushing the grey button on bottom of screen.
  Add the created file (/boot/debug.zip) to the issue.

  Warning: if you have configured wifi passwords they will be blanked by script during creation.
  We can't make absolutly sure you passwords are removed so do a personal check of the files inside the zip.

Stargazzer, 2019-06-21:
I have a raspberry pi3 b+ and the official touchscreen 7" display which worked flawlessly until the point where touchscreen works only in the chrankshaft menu but not in Android Auto like it did until this point.

Everything worked fine until Google Play Services updated.

I have run into a Catch-22.

I removed the Google Play Service updates successfully.

Android Auto won't work unless Google Play Services is updated.

Updated Google Play Services makes my touch screen not work in Android Auto.

Phone: Umidigi Z2 Pro

Android Version: 8.1.0

Build Number: UMIDIGI_Z2_PRO_V1.020180823

Raspberry Pi 3+ with 7" Touch Screen

CrankShaft: Crankshaft OpenCarOS NG Alpha-5 2019-06-08

Novocaine85, 2019-06-21:
Same identical issue on my Galaxy Note 8 EU version.

Android version: 9.0

Build number: PPR1.180610.011.hadesRom_OneUI_v7.0

Raspberry Pi 3+ with 7" Touch Screen

CrankShaft: Crankshaft OpenCarOS NG Alpha-5 2019-06-08

JoelBrenstrum, 2019-06-21:
Same issue on my OnePlus 5T

Android version: 9.0

Raspberry Pi 3+ with generic 7" screen and digitiser ("Mouse USB TouchController")

CrankShaft: Crankshaft OpenCarOS NG Alpha-5 2019-06-08

Getuwer, 2019-06-22:
I got the same issue on my Asus X00QD

Android version: 9.0

Raspberry Pi 3+ with 7" Touch Screen SKU:Z-0017 

Crankshaft OpenCarOS NG Alpha-4 2019-03-03
Crankshaft OpenCarOS NG Alpha-5 2019-06-08

Crankshaft : both not working. I was using the Alpha-4 wich worked using touchscreen on Android Auto and crankshaft . 
One day Android Auto touchscreen was not responding but crankshaft worked and playing Music with Spotify on Android Auto by starting Spotify via Android (smartphone) worked too.
I updated to Alpha-5 and this has had no changes.

[debug.zip](https://github.com/opencardev/crankshaft/files/3317174/debug.zip)

hawkeyexp, 2019-06-23:
needs some debug what was changed by play services....

chucktastic, 2019-06-23:
the f1xpl/aasdk repository had an issue report a few days ago stating the sdk would no longer work with android auto due to a protocol buffer incompatibility.  Something about AA using proto2 and aasdk using proto3 and play services starting to be more strict about the data it receives.  Issue reports have been disabled on f1xpl/aasdk for some reason and the debug for that issue is no longer visible.

Hopefully that helps a little.

Getuwer, 2019-06-23:
I tryed to do debug files 

I uninstalled Play Services updates on my phone and i catch some apk from apkmirror.com 

I first installed Google Play Services _17.4.55_ (100400-248795830) version, i plug my phone to my rpi and Android Auto started and touch screen was working , i then using touchscreen returning to crankshaft-ng menu and do debug file using the button debug while usb still plug

Then installed Google Play Services _17.5.28_ (100400-252519129) version, i plug my phone to my rpi and Android Auto started and touch screen was working , i then using touchscreen returning to crankshaft-ng menu and do debug file using the button debug while usb still plug 

Then installed Google Play Services _17.7.84_ (100400-252519124) version, i plug my phone to my rpi and Android Auto started but  **touch screen was not working anymore in AA** so i cannot reach crankshaft-ng menu with my touchscreen so i unplugged my phone , launched a debug and pluged my phone right after , AA started and debug was still in progress . 
So i got a debug file but it should not be really good. I should do my debug using ssh when my phone is plugged but i do not now how to debug via ssh.

[debug_working_17.4.55.zip](https://github.com/opencardev/crankshaft/files/3317911/debug_working_17.4.55.zip)
[debug_working_17.5.28.zip](https://github.com/opencardev/crankshaft/files/3317912/debug_working_17.5.28.zip)
[debug_not_working_17.7.84.zip](https://github.com/opencardev/crankshaft/files/3317910/debug_not_working_17.7.84.zip)

ishan747, 2019-06-24:
Same identical issue on my Galaxy S8 Qualcomm 

Android version: 9.0

Raspberry Pi 3 B+ with Other 7" touchscreen

CrankShaft: Crankshaft OpenCarOS NG Alpha-5 2019-06-08

iofirag, 2019-06-25:
Same identical issue on my Galaxy S7

Android version: 7.0

Raspberry Pi 3 B with  7" touchscreen

CrankShaft: 2019-03-03-crankshaft-ng-299C16C.zip

ulido, 2019-06-26:
Same issue with my Nokia 7 Plus

Android version: 9

RPi 3 A+ with official 7" touchscreen

CrankShaft latest
Android Auto version 4.3.591854
Google Play Services version 17.7.85

xxxjajoxxx, 2019-06-26:
Same issue with my Xiaomi Mi6

Android version: 8.0

RPi 3 B+ with official 7" touchscreen

CrankShaft latest
Android Auto version 4.1.590944
Google Play Services version 17.7.85 (040408-253824076)

eye2web, 2019-06-26:
Same issue with Huawei mate 20 pro
Android version: 9.0

RPi 3 B+ with 10.1" Waveshare touchscreen

CrankShaft latest
Android auto 4.4.592344
Google play services 17.7.85

Touchscreen is working in the crankshaft menu's. And stops working when connecting Phone.

tompeel, 2019-06-27:
Same Issue
Pixel 2XL
Android Q beta Beta 4
GPS - 17.7.85 (100400-253824076)
AA - 4.3.591854

xxxjajoxxx, 2019-06-27:
downgraded google play services to 17.1.22 and touchscreen went crazy random touches all over the android auto app works fine in crankshaft

josemaes, 2019-06-27:
Same issue.
Samsung S8+ Android 9

CrankShaft: 2019-03-03-crankshaft-ng-299C16C.zip

7" screen + USB touch panel

KK1423, 2019-06-27:
I am experiencing the same issue.

Phone: Oneplus 6T

Crankshaft: May 2018

KK1423, 2019-06-27:
A workaround is to open the application settings for Google Play Services and uninstalling all updates. Maybe a newer apk could be sideloaded, but the factory version is all that I tested.

Richard6360, 2019-06-28:
Same here on samsung A3 2017
Cant un install google play services

gcharlie, 2019-06-28:
I'll pull a debug later tonight, but I just got the update on my Google pixel 3 and same is happening to me. 
Additionally I noticed that Android auto had also stopped responding to  gpio mapped button commands

ashwaganda15, 2019-06-29:
Same.
Samsung S9+
Android 9
7" touchscreen
crankshaft 4 8 18

Work around is to use "OK Google" and tell AA to play the music you want.
 
now to reflash crankshaft as I figured my sd was corrupt.

Thanks for the info everyone-  I'll wait for someone smarter to post the fix...hopefully.

SonofAdam, 2019-06-30:
Same here, Pixel XL  after google play services update touch screen stopped working .

atjohnson23, 2019-06-30:
Same issue: 

Phone - Samsung S9 Plus on Android 9.0
CrankShaft -  latest
Android auto - 4.4.592344
Google play services - 17.7.85

Definitively a problem with google play services update it seems. I will downgrade and report results.

tompeel, 2019-07-01:
Im having issues downgrading my play services, with out formatting the phone its basically not going to happen so cant test this temporary fix.
Any progress on a fix?

josemaes, 2019-07-01:
> Im having issues downgrading my play services, with out formatting the phone its basically not going to happen so cant test this temporary fix.
> Any progress on a fix?

You have to disable "locate my phone" and google pay in the device administrator apps. Then you can remove the app updates but, even I disabled automatic updates and offline data for the app, it updates again after some hours. It is a very temporary fix.

rhatfield-bushel, 2019-07-02:
First you'll need to download the google play services APK version for your phone, here's the guide I used (pay attention to the sub version numbers):
https://itechify.com/2018/01/12/install-right-google-play-services-android-device/
And the apk downloads link:
https://google-play-services.en.uptodown.com/android/download/2007365

I tried `17.7.81` but that didn't work, I ended up using `17.5.28`.
Here's the settings I changed:
* `Settings > Google Services > Location` turn off location entirely
* `Settings > Google Services > Security > Find my device` turn off entirely
* `Settings > Apps > App Info > Google Play Services` first disable, then force stop
* `Settings > Apps > App Info > Google Play Services > Storage > Manage Space`  Clear all data, then on the previous screen Clear Cache

At this point your phone will likely be freaking out with tons of notifications. I'd disable all sounds before you start, it gets really annoying with it buzzing every 2 seconds.

Enable the app again, and install the APK you downloaded. If disable worked, it should install fine.

Go to Google Store (after install) and disable auto update. I think side loading an APK might prevent auto update anyways.

You'll need to re-enable all location services, but just open the Android Auto app and let it do it's thing. This works for me so far, we'll see if it tries to update itself anyways.

I got this working on the 7" touchscreen, Raspberry Pi 3 B, standard release SD image, and a Moto Z2 Force phone.

lukeb-uk, 2019-07-02:
Just as a warning, rolling back google play services may break other apps.

goon818, 2019-07-02:
The touch don't work with my phone with crankshaft.
I testet die Viktorgino headunit Project. The Touch working with it. I hope this information can help to find the problem.

gregorhollmig, 2019-07-03:
Is this issue openauto related? 
Is the openauto fork used in crankshaft? It seems the openauto project does not get any source updates anymore. maybe this could be helpful? 
https://github.com/f1xpl/openauto/pull/170

krisstakos, 2019-07-03:
> 
> 
> I got this working for the time, so I figured I'd share.
> 
> First you'll need to download the google play services APK version for your phone, here's the guide I used (pay attention to the sub version numbers):
> https://itechify.com/2018/01/12/install-right-google-play-services-android-device/
> And the apk downloads link:
> https://google-play-services.en.uptodown.com/android/versions
> 
> I tried `17.7.81` but that didn't work, I ended up using `17.5.26`.
> Here's the settings I changed:
> 
>     * `Settings > Google Services > Location` turn off location entirely
> 
>     * `Settings > Google Services > Security > Find my device` turn off entirely
> 
>     * `Settings > Apps > App Info > Google Play Services` first disable, then force stop
> 
>     * `Settings > Apps > App Info > Google Play Services > Storage > Manage Space`  Clear all data, then on the previous screen Clear Cache
> 
> 
> At this point your phone will likely be freaking out with tons of notifications. I'd disable all sounds before you start, it gets really annoying with it buzzing every 2 seconds.
> 
> Enable the app again, and install the APK you downloaded. If disable worked, it should install fine.
> 
> Go to Google Store (before or after install) and disable auto update. I think side loading an APK might prevent auto update anyways.
> 
> You'll need to re-enable all location services, but just open the Android Auto app and let it do it's thing. This works for me so far, we'll see if it tries to update itself anyways.
> 
> I got this working on the 7" touchscreen, Raspberry Pi 3 B, standard release SD image, and a Moto Z2 Force phone.

I roll back the Google Services but after a while they update again. Even after I turned background data usage off

hawkeyexp, 2019-07-03:
> 
> 
> Is this issue openauto related?
> Is the openauto fork used in crankshaft? It seems the openauto project does not get any source updates anymore. maybe this could be helpful?
> [f1xpl/openauto#170](https://github.com/f1xpl/openauto/pull/170)

Yes the problem is related to openauto or aasdk interacting with play services. The ping problem is solved longer time ago - only the touch problem was a new effect with latest play service update. Openauto has stopped any source updates (the original announce was the gpl version will get fixes also when openauto pro - commercial - is available - but as you will see - no updates for gpl based version.

But i try the changes cause we skipped the ping pong to fix the bug which was working for long time now. Thx for the hint.

xxxjajoxxx, 2019-07-03:
> I roll back the Google Services but after a while they update again. Even after I turned background data usage off

Wont work for me :/ random touches all over the screen and very laggy :/ (Mi6 Xiaomi.eu Rom)

peetereczek, 2019-07-03:
I just installed the project on my RPI 3+ with custom touch screen.
System works fine with Huawei P30 Pro, as it is supposed to, but also noticed issue with touch only in Android Auto

borconi, 2019-07-03:
Guys I've posted about this on AASDK, I thought you will pick it up, you need to convert all proto3 to proto2 and recompile the protobuf, that's all.
Sorry I haven't mentioned it here as well... :( 
I've looked it up now and the issue had been removed ( this was the url: https://github.com/f1xpl/aasdk/issues/32 )

chucktastic, 2019-07-03:
I don't think f1xpl is even remotely interested in backporting fixes to aasdk/openauto from "open"auto pro.  They're too busy charging for it and not releasing any code.

borconi, 2019-07-03:
I've updated the changed protobuf here: https://github.com/borconi/AAProto just change the package name accordingly and recompile should be all working.
It is possible that you will need to adjust the code as well in some places to reflect method changes between proto3 and proto2.

chucktastic, 2019-07-03:
Wow.  Thank you!  I've been a long time HUR user so I know how much work you've put into supporting the AA community.  I didn't have much luck compiling my own aasdk/openauto a while back.  I may just give it another shot when I've some free time again.

roffe, 2019-07-04:
why is a protobuff downgrade needed, has someone accidently made a typo and everyone just follows?

Proto3 is the successor of Proto2, why would google downgrade their protobuffers all of a sudden?

borconi, 2019-07-04:
O don't think it's a downgrade I think it was never intended to be 3 in the first place. The problem is caused by how 2 and 3 are treating empty / 0 fields. Recent Google play services made the checks more strict and where we where able to get away with no value set previously now it needs to see the 0, hence the downgrade needed. Let's not forget that AA was launched officially in 2015 so specs sent to car manufacturers etc are probably dated 2014 or so, that was the time of proto2 not 3.

krisstakos, 2019-07-04:
> 
> 
> I've uploaded the changed protobuf here: https://github.com/borconi/AAProto just change the package name accordingly and recompile should be all working.
> It is possible that you will need to adjust the code as well in some places to reflect method changes between proto3 and proto2.

Can you gives a bit more detailed instructions how to do it properly?

borconi, 2019-07-04:
@krisstakos I'm not sure as I'm not really familiar with c/c++ development neither had I looked at all the source code of crankshaft. But generally you need to download the proto compiler from Google, compile the proto to your language (c, java, python, etc) import the created classes in your project and use it.

krisstakos, 2019-07-04:
> @krisstakos I'm not sure as I'm not really familiar with c/c++ development neither had I looked at all the source code of crankshaft. But generally you need to download the proto compiler from [...]
 I'm a bit lost rigt now. Your proto files are downloaded and stored under aasdk_proto folder. Then I removed proto3 and compiled proto 2.6.1 for the RPI. And when I try to compile aasdk it gives me a ton of errors that can't find proto version and mismatch cimpiler proto3 version. Also I noticed that is some of the files says syntax=proto3 and syntax=proto2 is that normal?

borconi, 2019-07-04:
> I'm a bit lost rigt now. Your proto files are downloaded and stored under aasdk_proto folder. Then I removed proto3 and compiled proto 2.6.1 for the RPI. And when I try to compile aasdk it gives me [...]

Never do work at 2AM as the chances are you will do STUPID things. I've uploaded the wrong version of the files, sorry, just pushed a commit now, please re-download and try to compile again you shouldn't see more errors and all files should be proto2 syntax.

krisstakos, 2019-07-04:
That is the output. Any help to get it going?
`sudo cmake -DCMAKE_BUILD_TYPE=Release ../aasdk
CMake Error at /usr/share/cmake-3.7/Modules/FindProtobuf.cmake:357 (file):
  file STRINGS file "/usr/include/google/protobuf/stubs/common.h" cannot be
  read.
Call Stack (most recent call first):
  aasdk_proto/CMakeLists.txt:1 (include)

CMake Error at /usr/share/cmake-3.7/Modules/FindProtobuf.cmake:363 (math):
  math cannot parse the expression: " / 1000000": syntax error, unexpected
  exp_DIVIDE, expecting exp_OPENPARENT or exp_NUMBER (2)
Call Stack (most recent call first):
  aasdk_proto/CMakeLists.txt:1 (include)

CMake Error at /usr/share/cmake-3.7/Modules/FindProtobuf.cmake:364 (math):
  math cannot parse the expression: " / 1000 % 1000": syntax error,
  unexpected exp_DIVIDE, expecting exp_OPENPARENT or exp_NUMBER (2)
Call Stack (most recent call first):
  aasdk_proto/CMakeLists.txt:1 (include)

CMake Error at /usr/share/cmake-3.7/Modules/FindProtobuf.cmake:365 (math):
  math cannot parse the expression: " % 1000": syntax error, unexpected
  exp_MOD, expecting exp_OPENPARENT or exp_NUMBER (2)
Call Stack (most recent call first):
  aasdk_proto/CMakeLists.txt:1 (include)

CMake Warning at /usr/share/cmake-3.7/Modules/FindProtobuf.cmake:387 (message):
  Protobuf compiler version 3.0.0 doesn't match library version ..
Call Stack (most recent call first):
  aasdk_proto/CMakeLists.txt:1 (include)

CMake Error at /usr/share/cmake-3.7/Modules/FindProtobuf.cmake:357 (file):
  file STRINGS file "/usr/include/google/protobuf/stubs/common.h" cannot be
  read.
Call Stack (most recent call first):
  aasdk_proto/CMakeLists.txt:2 (find_package)

CMake Error at /usr/share/cmake-3.7/Modules/FindProtobuf.cmake:363 (math):
  math cannot parse the expression: " / 1000000": syntax error, unexpected
  exp_DIVIDE, expecting exp_OPENPARENT or exp_NUMBER (2)
Call Stack (most recent call first):
  aasdk_proto/CMakeLists.txt:2 (find_package)

CMake Error at /usr/share/cmake-3.7/Modules/FindProtobuf.cmake:364 (math):
  math cannot parse the expression: " / 1000 % 1000": syntax error,
  unexpected exp_DIVIDE, expecting exp_OPENPARENT or exp_NUMBER (2)
Call Stack (most recent call first):
  aasdk_proto/CMakeLists.txt:2 (find_package)

CMake Error at /usr/share/cmake-3.7/Modules/FindProtobuf.cmake:365 (math):
  math cannot parse the expression: " % 1000": syntax error, unexpected
  exp_MOD, expecting exp_OPENPARENT or exp_NUMBER (2)
Call Stack (most recent call first):
  aasdk_proto/CMakeLists.txt:2 (find_package)

CMake Warning at /usr/share/cmake-3.7/Modules/FindProtobuf.cmake:387 (message):
  Protobuf compiler version 3.0.0 doesn't match library version ..
Call Stack (most recent call first):
  aasdk_proto/CMakeLists.txt:2 (find_package)

-- Found Protobuf: /usr/lib/arm-linux-gnueabihf/libprotobuf.so;-lpthread;-lpthread (found version "..")
-- Boost version: 1.62.0
-- Found the following Boost libraries:
--   system
--   log
--   unit_test_framework
--   date_time
--   log_setup
--   filesystem
--   thread
--   regex
--   chrono
--   atomic
-- Found libusb-1.0:
--  - Includes: /usr/include/libusb-1.0
--  - Libraries: /usr/lib/arm-linux-gnueabihf/libusb-1.0.so
CMake Error at /usr/share/cmake-3.7/Modules/FindProtobuf.cmake:357 (file):
  file STRINGS file "/usr/include/google/protobuf/stubs/common.h" cannot be
  read.
Call Stack (most recent call first):
  CMakeLists.txt:40 (find_package)

CMake Error at /usr/share/cmake-3.7/Modules/FindProtobuf.cmake:363 (math):
  math cannot parse the expression: " / 1000000": syntax error, unexpected
  exp_DIVIDE, expecting exp_OPENPARENT or exp_NUMBER (2)
Call Stack (most recent call first):
  CMakeLists.txt:40 (find_package)

CMake Error at /usr/share/cmake-3.7/Modules/FindProtobuf.cmake:364 (math):
  math cannot parse the expression: " / 1000 % 1000": syntax error,
  unexpected exp_DIVIDE, expecting exp_OPENPARENT or exp_NUMBER (2)
Call Stack (most recent call first):
  CMakeLists.txt:40 (find_package)

CMake Error at /usr/share/cmake-3.7/Modules/FindProtobuf.cmake:365 (math):
  math cannot parse the expression: " % 1000": syntax error, unexpected
  exp_MOD, expecting exp_OPENPARENT or exp_NUMBER (2)
Call Stack (most recent call first):
  CMakeLists.txt:40 (find_package)

CMake Warning at /usr/share/cmake-3.7/Modules/FindProtobuf.cmake:387 (message):
  Protobuf compiler version 3.0.0 doesn't match library version ..
Call Stack (most recent call first):
  CMakeLists.txt:40 (find_package)

-- Found Protobuf: /usr/lib/arm-linux-gnueabihf/libprotobuf.so;-lpthread (found version "..")
-- Configuring incomplete, errors occurred!
See also "/home/pi/aasdk_build/CMakeFiles/CMakeOutput.log".
See also "/home/pi/aasdk_build/CMakeFiles/CMakeError.log".
`

borconi, 2019-07-04:
@krisstakos - sorry I don't know, I would need to download and understand how the whole crankshaft is built, but I do not have the time and capacity now. 
I think that's the problem here: Protobuf compiler version 3.0.0 doesn't match library version ..
Your protoc isn't quite right, you can use the new protoc compiler to compile proto2 syntax, that's not an issue, as soon as the syntax is changed from 3 to 2 the compile will know what to do.
If you give me the package name I can compile the protobuf to c++ for you, but I need to know the exact package name.

tompeel, 2019-07-04:
Just as i look to openauto pro life breaths again in the CS community! 
I was taking a look at trying to get this to compile on my PC but im not a dev and am not sure how to open any of it... im used to opening solutions in VS hahah! 
Anyway I commend your work gentlemen!

roffe, 2019-07-04:
I managed to get `aasdk_proto` to compile on a generic linux host using proto2 gut on a Pi running crankshaft it complains it cannot do it any one. only difference i noted is that the Pi version is running version 3.0.0 of protobuf. sitting and compiling it by hand as we speak and hope for results soon

krisstakos, 2019-07-04:
> 
> 
> I managed to get `aasdk_proto` to compile on a generic linux host using "proto2" instead of "proto3" in the protocol files. But on a Pi running crankshaft it complains it cannot do it any one. only difference i noted is that the Pi version is running version 3.0.0 of protobuf. sitting and compiling protobuf compiler by hand as we speak and hope for results soon

How you managed to compile successfully? RPI just yelling at me with errors in around 84%

borconi, 2019-07-04:
Again, I can compile the protos for you to C++ if you give me the exact package name I should use, it will literally take me 2 mins.

roffe, 2019-07-04:
@krisstakos i copied borconi's proto files into the aasdk_proto folder and ran `cmake CMakeLists.txt` in the repo root, then `make -j3` (-j3 for number of cores on machine -1)

roffe, 2019-07-04:
no dice

`Jul  4 18:37:21 CRANKSHAFT-NG service_openauto.sh[4279]: /opt/crankshaft/service_openauto.sh: line 182:  4856 Segmentation fault      stdbuf -i0 -o0 -e0 /usr/local/bin/autoapp >> /tmp/openauto.log`

when i connect the phone after rebuilding libaasdk.so & libaasdk_proto.so with @borconi's protobuf files

borconi, 2019-07-04:
@roffe probably some of the functions needs slight adjustment as well, but I really cannot help there. I know I needed to make some small adjustment in my Java code when I've changed it, but trust me the reason of touch not working is the usage of proto3 instead of proto2.

hawkeyexp, 2019-07-04:
proto2 build is in progress - stay tuned...
@borconi thx for your hint and help

tompeel, 2019-07-05:
@hawkeyexp hanging on the end of a thread here! On the plus side i did in this period of broken decided to try Open Auto Pro and have been pretty underwhelmed. Its just not working...

borconi, 2019-07-05:
@tompeel you should have tried HUR instead 😄  you would been pretty overwhelmed 😁

tompeel, 2019-07-05:
@borconi whats HUR?

borconi, 2019-07-05:
HUR = Headunit Reloaded, the app before Crankshaft, basically the same thing but for Android instead of Raspberry. Hence knowing exactly what's the problem as I've faced it in my app as well, about 3 weeks ago, I've just was lucky enough to be on Google Play Service beta so I caught the problem before it became general issue and managed to push out an update for my app before the majority of users got affected. Also that's when I've posted about this on the AASDK warning that this will happen and that it needs addressing, again sorry I haven't warned this community as well.

lukeb-uk, 2019-07-05:
@borconi If they hadn't have hidden the issues on AASDK then it would have been easier for the crankshaft team to see, no blame on you!

scylla-xi, 2019-07-05:
> proto2 build is in progress - stay tuned...
> @borconi thx for your hint and help

any update?

tompeel, 2019-07-05:
@borconi im on GPS BETA too, weirdly i had it all working fine with the same GPS version, then i downloaded the new CS on my test pi logged in and it must have flicked a switch because then it wouldn't work with either alpha 4 or 5... i confirmed it with the release date. very odd.

abraha2d, 2019-07-05:
So before this, I was able to run openauto: the screen/video output worked, but there was no touch input. I merged @borconi's AAProto files into aasdk, and made a small change to openauto to get it to compile. Now touch works, but there's no video output (i.e. I can touch the place where the music play button should be, and music starts playing, but I can't actually see anything other than the "Waiting for device" screen).

Anyone have any ideas on what's causing this? These are the changes I made:
https://github.com/f1xpl/aasdk/compare/development...abraha2d:development
https://github.com/f1xpl/openauto/compare/development...abraha2d:development

[111 more replies]

closed 2019-08-08 by abraha2d