opencardev/crankshaft #357: “Return to Crankshaft-NG” button in AA not working

abraha2d, 2019-07-08:
- What phone do you have?

OnePlus 6T

- What OS version?

OxygenOS Beta 13

- What crankshaft version?

Alpha-5, with fixes for #352.
https://github.com/abraha2d/aasdk.
https://github.com/abraha2d/openauto.

- Steps to reproduce the bug, if you're planning to report a bug. Please indicate whether the bug is always repoducible.

Connect phone, and wait for Android Auto to launch. Press the rightmost button on the toolbar. Click "Return to Crankshaft-NG". Screen freezes.

- Please provide any further information that you might find helpful if available.

This line is output to /tmp/openauto.log when "Return to Crankshaft-NG" is pressed:
`[2019-07-08 11:27:55.174485] [0x6f8fd300] [error]   [OpenAuto] [VideoService] channel error: AaSdk error code: 32, native code: 0`

- Starting with pre4: Boot into dev mode (can be selected in settings).
  Reproduce the possible bug. Create the debug logs by pushing the grey button on bottom of screen.
  Add the created file (/boot/debug.zip) to the issue.

tompeel, 2019-07-12:
Yea i see this too, fortunately a way around this is in the notifications bar in your phone there is a disconnect from car notification which you can tap and it will return back to the CrankShaftNG menu.
It doesn't seem to actually crash the app, rather send a command with no response so everything is still running in the background.

abraha2d, 2019-07-12:
Huh, never noticed that before. I'll have to look out for it next time I drive.
I'm also looking into fixing the return functionality in OpenAuto.

abraha2d, 2019-07-12:
My initial guess at what the issue might be:
- The "Return to..." button tells the phone "hey i don't want to see AA anymore"
- Phone says "ok, i'm stopping the video feed"
- [what should happen] OpenAuto says "hey the video feed stopped, let's hide AA"
- [what's happening] OpenAuto says "what?! why did the video feed stop?!?!"
Now to figure out where in code this is handled...

abraha2d, 2019-07-12:
Full logs:
- Phone connected: https://gist.github.com/abraha2d/192b94584ce3991b13278d342dddeacd
- Return to Crankshaft-NG pressed: https://gist.github.com/abraha2d/a68022edac27f24b30957f6240412756
- Phone disconnected: https://gist.github.com/abraha2d/5d79bb2665b516ea726210b9b44298c4

abraha2d, 2019-07-12:
Found another log message. It's erroring with:
`Can't parse message of type "f1x.aasdk.proto.messages.VideoFocusRequest" because it is missing required fields: disp_index`
Saw this when I ran `systemctl status openauto`. Why are all the logs not in 1 place?
Anyway I know where to fix this!

abraha2d, 2019-07-12:
Cool, that works! PR coming soon...

closed 2019-08-07 by abraha2d