opencardev/crankshaft #358: terminal half in front of crankshaft mainscreen

Bishamon1987, 2019-07-09:
Hi,
Since the new update and the fix of the touchscreen issue, my crankshaft main screen looks like this.
[image]

It's only there in X11 mode not in EGL

Any idea if it's a setting issue or a bug?

tompeel, 2019-07-10:
Is the fix not to just use egl?

Bishamon1987, 2019-07-11:
my touchscreen doesn't work in EGL.
So I'm forced to use X11.
But thanks ^^

tompeel, 2019-07-12:
Perhaps you have boot to old style menu selected by default in the config? if so try to set it back to its default.
Otherwise this is a pretty odd one, have you tried a new image from scratch?
There is a way of doing this over USB or OTA i think by placing a file in the crankshaft folder even to jsut restore to the same version as your on but fresh. 
You would have to do the 45 mins or so of building with the Protobuf2 fix....

Bishamon1987, 2019-07-12:
I just did this whole new image. Like described in the touch screen doesn't
work because of Google service update.
I try the menu on default part first.

abraha2d, 2019-07-12:
Occasionally I have one character show through at the bottom-left corner of the screen. I've never been able to successfully reproduce it. I'm also using X11 mode because of my touchscreen.

Never had almost the whole thing be hidden before. What's your screen resolution?

abraha2d, 2019-07-12:
Also, is the screen still hidden when you plug in your phone, or does AA show over the full screen?

Bishamon1987, 2019-07-12:
Well I'm not sure about the screen resolution.
The car dealer can't tell me what the specs are. I think it's something
like 560p

Android auto does go over AA. I use it wirelessly.

When I change to day/night or classic or NG the screen is fully back.
So I think that the terminal just loads later then AA? And it pops up or
something.

Aaron407s, 2019-07-24:
I had reported this same issue a few months ago (issue #308). No real solution that I've found so far.

homestar1217, 2019-08-10:
I've been able to resolve this issue in Alpha4 with the ethernet solution provided by @Grawa in issue #308. I'm going to try this again today to see if it resolves it for Alpha5.1. If anyone knows a command to perform this from within the Crankshaft console that would be awesome, otherwise it looks like I may be installing Linux on an old computer today.

Bishamon1987, 2019-08-10:
I use this app on my phone called Termius. It works great.

homestar1217, 2019-08-10:
@Bishamon1987 @abraha2d I can confirm the workaround in #308 works for me with Alpha5.1 as well.

add the text below to the \etc\network\interfaces file:

auto eth0
iface eth0 inet static
address 192.168.1.50
netmask 255.255.255.0
gateway 192.168.1.1

You can obviously change the IP to match your network if you need to.

Grawa, 2019-08-10:
@homestar1217 nice to see someone that have used my workaround after 5 months :D

in the meanwhile I started to learn python and bought OA Pro XD

homestar1217, 2019-08-10:
It works to remove the terminal from in front of the UI, but now AA won't display when I plug my phone in. Crankshaft sees the phone plug in and locks the UI, but AA never actually displays. I think this is a UI issue as I can start music from my phone and it plays through the Pi so I think the AA process is actually running, just not displaying on the screen. I'm about done with Crankshaft, The audio coming out of the composite RPi output is really bad anyway.

abraha2d, 2019-08-12:
@homestar1217 did you try the steps here to get better audio? https://github.com/opencardev/crankshaft/wiki/3.5mm-Audio-Jack

abraha2d, 2019-08-12:
@homestar1217 @Bishamon1987 without using the workaround, if you touch where the settings icon is supposed to be, do you see the settings UI?

If so, if you press cancel, do you get the crankshaft UI back?

Bishamon1987, 2019-08-12:
Yes, but if the terminal gives a low voltage warning, this line will show in front of crankshaft again

abraha2d, 2019-08-12:
How about https://join.slack.com/t/crankshaft-ng/shared_invite/enQtNzI2MDE0MDE3NDMxLWU4MWUxNzIxNjM2YWMyM2VjZmQzYjE3YTBiMDc0NDhjNzIwZDY0YjU5N2Y0ZmMzYTU2YjU4ZGRiNTEwNjU2YmQ

jreneew2, 2019-08-14:
> It works to remove the terminal from in front of the UI, but now AA won't display when I plug my phone in. Crankshaft sees the phone plug in and locks the UI, but AA never actually displays. I think [...]

I am having the exact same issue as you so you are not alone. I ran into the overlay and used your workaround and same result. I can stream audio through BT and it connects and locks the phone, but android audio isn't showing up on the screen

homestar1217, 2019-08-16:
@abraha2d I had not tried the audio workaround for the 3.5mm jack. I use it for both audio and video, would that make a difference? It was really bad, it sounded like it was clipping all the time but I don't have that problem with other devices so if that would be improved with this workaround it's definitely something I will try.

@jreneew2 to be honest I got fed up with chasing issues with Crankshaft and I have gone back to using my Asus TinkerBoard running Android 6.0 and the HeadUnit Reloaded app. I had to spend $15 on an HDMI to RCA converter, but the ease of use is worth it. I hope the Crankshaft devs can fix a bunch of these issues with a new release soon, but as it is right now it was pretty much unusable in my setup. I'm disappointed because I really wanted to use it.

jreneew2, 2019-08-16:
@homestar1217 Hmm. I hadn't thought of doing that. I think I have one somewhere around I can try that.

abraha2d, 2019-08-16:
> Yes, but if the terminal gives a low voltage warning, this line will show in front of crankshaft again.

Hmm, a temporary solution would probably be to add `loglevel=0 vt.global_cursor_default=0` in /boot/cmdline.txt, to get rid of all console messages.

The proper fix would be to start X11 on its own tty. I'll look into it.

abraha2d, 2019-08-16:
@homestar1217 Using the 3.5mm jack for video as well shouldn't make a difference as long as you've got the pinout correct.

What volume do you have set in crankshaft? Maybe that could be the source of any clipping? I've got mine set to 50%.

abraha2d, 2019-08-31:
For some unknown reason this problem has started happening on my system as well. Will update if/when I fix it.

closed 2021-11-27 by github-actions[bot]