opencardev/crankshaft #359: Use gpio.Out for Backlight and more / Busy

WhereisBielefeld, 2019-07-10:
- What phone do you have? Not depending on phone 

- What OS Version? Not depending on phone 

- What crankshaft version? Alpha 3

Maybe this is not a real problem, but is there a way to use one of the GPIOs as a switch. I use a mosfet to trigger the backlight brigthness as it's not possible on stock.
But as I want to change the state in GPIO exports it won't let me.

And ideas?

gcharlie, 2019-07-10:
If I'm understanding correctly you should be able to map that in the setting menu. I have a similar setup where I use an octocoupler wired up to  the illumination signal coming out of the wire harness.
Make sure you are mapping it to the right gpio number, since the gpio numbers are not assigned relative to the physical pin number

WhereisBielefeld, 2019-07-11:
In the settings menu I was only able manage incoming signals for certain actions in crankshaft/AA. But couldn't finde a way to get an internal event, like switching from day to night, as output for a gpio pin.
To be a bit more precise. If I change from day to night mode I'd like to get a pin set from high to low in oder to block a mosfet.

Bishamon1987, 2019-07-11:
I got some question about this too.
I want to use a switch with 3 positions for my rear and frontcam.

I will either run the cam_overlay bin twice or use some small script to
mount and unmount specific USB devices (videograbbers).

I don't really understand how this gpio thing works. But can I put this
little script inside the gpio mapper?

gcharlie, 2019-07-11:
Ohh I misunderstood, I thought you meant having the unit brighten or dimm depending on the state of the car, like the cars headlights being on or tapping into the cars day/night sensor. 

Your right there's not really a means of doing that from any of the forward facing setting, all of the gpio setting are used more or less as inputs.

Whats the mosfet blocking exactly?

WhereisBielefeld, 2019-07-12:
A low on the gpio will cut the Source/Drain and a high will connect them. The backlight of my display is connected over a resistor, which will be skipped by the mosfet to get a higher screen brightness.

Nevertheless. Not idea to get on of the GPIOs as output? A separate skript didn't work.

Besides: Using the signal from headlight won't help because there are enough country's which force you to keep the lights on at daytime.

gcharlie, 2019-07-12:
Yeah if a custom script has been off no use, I don't know what to tell you. But I would causion against having something as critical as the headlights dependant on functionality from the raspberry pi. If the pi unit loses power momentarily for what ever reason it will take a while to power back on and you'll be without headlights for that time

WhereisBielefeld, 2019-07-13:
Its not my intention to use a signal from my headlight, nor make them dependent on my rpi. I think we might misunderstood.

WhereisBielefeld, 2019-07-20:
No idea? Better closing this threat or am i asking at the wrong place? I don´t want to hurry because i love this project!

closed 2021-11-27 by github-actions[bot]