opencardev/crankshaft #364: Lag

ArvindSami, 2019-08-01:
If you need support: Please post on https://www.reddit.com/r/crankshaft/ as this is a bug tracker, not a support forum.

Note: Reporting issues not based on official published builds will be closed without any comment.

- What phone do you have?
Xperia Xz2 Premium

- What OS version?
Android P

- What crankshaft version?
NG5

- Steps to reproduce the bug, if you're planning to report a bug. Please indicate whether the bug is always repoducible.
Driving around, All reproducible, after about half an hour maps playing catchup with the voice! (so the voice says turn right but the maps show a dead straight road, or I finish driving and according to the screen I am still 10 minutes away). Tried pressing buttons such as music or telephone or crankshaft and these take a while to a) recognise inpt and b) load

- Please provide any further information that you might find helpful if available.
Tried using a case to cool, no effect. 

- Starting with pre4: Boot into dev mode (can be selected in settings).
  Reproduce the possible bug. Create the debug logs by pushing the grey button on bottom of screen.
  Add the created file (/boot/debug.zip) to the issue.

  Warning: if you have configured wifi passwords they will be blanked by script during creation.
  We can't make absolutly sure you passwords are removed so do a personal check of the files inside the zip.

Crescendo-BLYAT, 2019-08-12:
maybe your Pi is overheating?

kolonuk, 2019-12-16:
I had this on an old Nexus 5, moved to Nokia 7.1 and issue went away.  I personally think I had too much running on my old Nexus 5 as it was slow-ish normally anyway...  You could try a different phone to see is it is the rpi or your phone, or try resetting phone and see if that helps.  Also, try running Android Auto not connected to rpi and see what happens over time.

closed 2020-10-01 by matt2005