opencardev/crankshaft #366: NG 5.1 pre-release OpenAuto crashes in X11

Bishamon1987, 2019-08-09:
If you need support: Please post on https://www.reddit.com/r/crankshaft/ as this is a bug tracker, not a support forum.

Note: Reporting issues not based on official published builds will be closed without any comment.

- What phone do you have?
Nokia 6.1
- What OS version?
Android Pie
- What crankshaft version?
NG5.1
- Steps to reproduce the bug, if you're planning to report a bug. Please indicate whether the bug is always repoducible.
I just put the new image on my usb stick, put it in my pi and turned on my pi with crankshaft NG 5.0 
It ran like it should. But it seems not to be able to load open auto in X11 mode.
I didnt put any thing else on the new installation.
I tried running in devapp and default mode.

abraha2d, 2019-08-09:
@Bishamon1987 Can you post the output of running `crankshaft syslog` from an SSH connection?

Bishamon1987, 2019-08-09:
Here it is!

[Syslog.zip](https://github.com/opencardev/crankshaft/files/3488002/Syslog.zip)

abraha2d, 2019-08-09:
Hmm, the error seems to be `qt.qpa.plugin: Could not load the Qt platform plugin "xcb" in "" even though it was found.`
I've encountered this error when I set up my system, although I forgot what I did to fix it.

Let me see if I can figure out what I did to fix it.

Bishamon1987, 2019-08-09:
I found this but it doesnt say much to me :P 
https://forum.qt.io/topic/93247/qt-qpa-plugin-could-not-load-the-qt-platform-plugin-xcb-in-even-though-it-was-found/

abraha2d, 2019-08-09:
Ah, I remember now. 

Run this: `ldd /usr/lib/qt/plugins/platforms/libqxcb.so`

What does it say?

Bishamon1987, 2019-08-09:
This is not possible
/usr/lib/**qt** does not exist.

Bishamon1987, 2019-08-09:
with `find / -name libqxcb `
i get no result either

abraha2d, 2019-08-09:
Sorry, this should work:

`ldd /usr/local/qt5/plugins/platforms/libqxcb.so`

Bishamon1987, 2019-08-09:
```
ldd /usr/local/qt5/plugins/platforms/libqxcb.so        linux-vdso.so.1 (0x7eff8000)
	/usr/lib/arm-linux-gnueabihf/libarmmem.so (0x76f1f000)
	libQt5XcbQpa.so.5 => /usr/local/qt5/plugins/platforms/../../lib/libQt5XcbQpa.so.5 (0x76e1a000)
	libfontconfig.so.1 => /usr/lib/arm-linux-gnueabihf/libfontconfig.so.1 (0x76dc9000)
	libfreetype.so.6 => /usr/lib/arm-linux-gnueabihf/libfreetype.so.6 (0x76d2e000)
	libz.so.1 => /lib/arm-linux-gnueabihf/libz.so.1 (0x76d07000)
	libQt5Gui.so.5 => /usr/local/qt5/plugins/platforms/../../lib/libQt5Gui.so.5 (0x76803000)
	libQt5DBus.so.5 => /usr/local/qt5/plugins/platforms/../../lib/libQt5DBus.so.5 (0x76783000)
	libQt5Core.so.5 => /usr/local/qt5/plugins/platforms/../../lib/libQt5Core.so.5 (0x76277000)
	libbrcmGLESv2.so => /opt/vc/lib/libbrcmGLESv2.so (0x76252000)
	libbrcmEGL.so => /opt/vc/lib/libbrcmEGL.so (0x76219000)
	libpthread.so.0 => /lib/arm-linux-gnueabihf/libpthread.so.0 (0x761f0000)
	libX11-xcb.so.1 => /usr/lib/arm-linux-gnueabihf/libX11-xcb.so.1 (0x761de000)
	libxcb-icccm.so.4 => not found
	libxcb-image.so.0 => not found
	libxcb-shm.so.0 => /usr/lib/arm-linux-gnueabihf/libxcb-shm.so.0 (0x761cb000)
	libxcb-keysyms.so.1 => not found
	libxcb-randr.so.0 => not found
	libxcb-render-util.so.0 => not found
	libxcb-render.so.0 => /usr/lib/arm-linux-gnueabihf/libxcb-render.so.0 (0x761b0000)
	libxcb-shape.so.0 => not found
	libxcb-sync.so.1 => /usr/lib/arm-linux-gnueabihf/libxcb-sync.so.1 (0x7619a000)
	libxcb-xfixes.so.0 => /usr/lib/arm-linux-gnueabihf/libxcb-xfixes.so.0 (0x76183000)
	libxcb-xinerama.so.0 => not found
	libxcb.so.1 => /usr/lib/arm-linux-gnueabihf/libxcb.so.1 (0x76154000)
	libXrender.so.1 => /usr/lib/arm-linux-gnueabihf/libXrender.so.1 (0x7613b000)
	libXext.so.6 => /usr/lib/arm-linux-gnueabihf/libXext.so.6 (0x7611c000)
	libX11.so.6 => /usr/lib/arm-linux-gnueabihf/libX11.so.6 (0x75ff9000)
	libm.so.6 => /lib/arm-linux-gnueabihf/libm.so.6 (0x75f7a000)
	libxkbcommon.so.0 => /usr/lib/arm-linux-gnueabihf/libxkbcommon.so.0 (0x75f31000)
	libdl.so.2 => /lib/arm-linux-gnueabihf/libdl.so.2 (0x75f1e000)
	libstdc++.so.6 => /usr/lib/arm-linux-gnueabihf/libstdc++.so.6 (0x75dd6000)
	libgcc_s.so.1 => /lib/arm-linux-gnueabihf/libgcc_s.so.1 (0x75da9000)
	libc.so.6 => /lib/arm-linux-gnueabihf/libc.so.6 (0x75c6a000)
	libxcb-icccm.so.4 => not found
	libxcb-image.so.0 => not found
	libxcb-keysyms.so.1 => not found
	libxcb-randr.so.0 => not found
	libxcb-render-util.so.0 => not found
	libxcb-shape.so.0 => not found
	libxcb-xinerama.so.0 => not found
	libgthread-2.0.so.0 => /usr/lib/arm-linux-gnueabihf/libgthread-2.0.so.0 (0x75c58000)
	libglib-2.0.so.0 => /lib/arm-linux-gnueabihf/libglib-2.0.so.0 (0x75b50000)
	libexpat.so.1 => /lib/arm-linux-gnueabihf/libexpat.so.1 (0x75b1e000)
	/lib/ld-linux-armhf.so.3 (0x76f35000)
	libpng16.so.16 => /usr/lib/arm-linux-gnueabihf/libpng16.so.16 (0x75ae4000)
	libdbus-1.so.3 => /lib/arm-linux-gnueabihf/libdbus-1.so.3 (0x75a91000)
	libicui18n.so.57 => /usr/lib/arm-linux-gnueabihf/libicui18n.so.57 (0x75881000)
	libicuuc.so.57 => /usr/lib/arm-linux-gnueabihf/libicuuc.so.57 (0x75717000)
	libicudata.so.57 => /usr/lib/arm-linux-gnueabihf/libicudata.so.57 (0x73e8a000)
	libdouble-conversion.so.1 => /usr/lib/arm-linux-gnueabihf/libdouble-conversion.so.1 (0x73e68000)
	libbcm_host.so => /opt/vc/lib/libbcm_host.so (0x73e3f000)
	libvchiq_arm.so => /opt/vc/lib/libvchiq_arm.so (0x73e29000)
	libvcos.so => /opt/vc/lib/libvcos.so (0x73e0e000)
	librt.so.1 => /lib/arm-linux-gnueabihf/librt.so.1 (0x73df7000)
	libXau.so.6 => /usr/lib/arm-linux-gnueabihf/libXau.so.6 (0x73dec000)
	libXdmcp.so.6 => /usr/lib/arm-linux-gnueabihf/libXdmcp.so.6 (0x73dd7000)
	libpcre.so.3 => /lib/arm-linux-gnueabihf/libpcre.so.3 (0x73d5e000)
	libsystemd.so.0 => /lib/arm-linux-gnueabihf/libsystemd.so.0 (0x73ce3000)
	libbsd.so.0 => /lib/arm-linux-gnueabihf/libbsd.so.0 (0x73cba000)
	libselinux.so.1 => /lib/arm-linux-gnueabihf/libselinux.so.1 (0x73c87000)
	liblzma.so.5 => /lib/arm-linux-gnueabihf/liblzma.so.5 (0x73c56000)
	liblz4.so.1 => /usr/lib/arm-linux-gnueabihf/liblz4.so.1 (0x73c33000)
	libgcrypt.so.20 => /lib/arm-linux-gnueabihf/libgcrypt.so.20 (0x73b62000)
	libgpg-error.so.0 => /lib/arm-linux-gnueabihf/libgpg-error.so.0 (0x73b42000)
```

abraha2d, 2019-08-09:
`sudo apt install libxcb-icccm4 libxcb-image0 libxcb-keysyms1 libxcb-randr0 libxcb-render-util0 libxcb-shape0 libxcb-xinerama0`

Bishamon1987, 2019-08-09:
Lets see

abraha2d, 2019-08-09:
Out of curiosity, did you not have this problem on NG5? I ask because nothing's supposed to be different between 5 and 5.1, with the exception of the prebuilts being updated (autoapp, libaasdk, etc.)

Bishamon1987, 2019-08-09:
Yea I did have this problem in NG5 after running a camoverlay.bin which some how crashed.... really weird.

It seems to work! NICE

Bishamon1987, 2019-08-09:
Maybe it started after the some update, not sure

abraha2d, 2019-08-09:
Cool. I'll see if I can add those packages to the package list for the next release.
I assume Android Auto is working properly as well? Then I can probably go ahead and release 5.1.

Bishamon1987, 2019-08-09:
GIve me a sec. 

Something offtopic: Is it possible to add some extra buttons to android auto, like the one going back to crankshaft?

abraha2d, 2019-08-09:
#362? As far as I know, I don't think you can. But I'm not sure. Maybe someone with more knowledge will comment on that issue.

Bishamon1987, 2019-08-09:
hmmm... it doesnt seem to connect

abraha2d, 2019-08-09:
Over WiFi or USB?

Bishamon1987, 2019-08-09:
Over Wifi Hotspot

abraha2d, 2019-08-09:
Oh, ok. Testing on my side now.

Bishamon1987, 2019-08-09:
It doesn't even detect my phone via USB

Bishamon1987, 2019-08-09:
after changing to client wifi, i got the usb detection back. But it doesn't connect either

homestar1217, 2019-08-10:
I am having the same issue. I updated to Alpha 5.1 from Alpha4 via USB. I am running in X11 mode and have modified the /etc/X11/xorg.conf.d/99-calibration.conf to work with my touchscreen overlay as well. OpenAuto attempts to start in X11 mode then crashes. I don't have a Linux machine so I'm not sure if I can SSH into the Pi and send commands without that.

Hardware FYI:  Pi 3 B+, Xiaomi Mi Mix 2S on Android 9.0.

abraha2d, 2019-08-10:
@Bishamon1987 I'm not able to get wifi mode working (Pi crashes, probably because I'm using a 2B, and it doesn't support hotspot for my USB wifi adapter). But I just verified with a fresh installation of NG5.1, after installing the packages listed [here](https://github.com/opencardev/crankshaft/issues/366#issuecomment-520072487), everything works with USB.

abraha2d, 2019-08-10:
@homestar1217 If you have a USB keyboard attached, you can press Ctrl-Alt-F2 to switch to another virtual console on the Pi. Default user/pass is pi/raspberry, and then you can run the commands there.

homestar1217, 2019-08-10:
@abraha2d I do show the same error @Bishamon1987 got.  

"qt.qpa.plugin: Could not load the Qt platform plugin "xcb" in "" even though it was found."

I attempted to ran this command: "ldd /usr/local/qt5/plugins/platforms/libqxcb.so" and got pretty much the same output as well.

Ran the command that worked for @Bishamon1987 but unfortunately I just get all kinds of errors. Mostly it looks like due to the files being read-only.

abraha2d, 2019-08-10:
@homestar1217 run `crankshaft system unlock` first. By default everything boots into a read-only state to prevent filesystem corruption if the power is abruptly removed.

homestar1217, 2019-08-10:
@abraha2d thank you that did it. I still have the login screen coming up after the UI appears, I had fixed that on the previous build I was running. Hopefully I can fix that without Linux as well. Thank you again.

Bishamon1987, 2019-08-10:
@homestar1217
It would be nice if you share the fix in this issue -> #358

matt2005, 2020-10-17:
Is this an issue in alpha 55?

matt2005, 2020-10-26:
fixed in alpha7

closed 2020-10-26 by matt2005