opencardev/crankshaft #368: Night Mode - Is It Supposed to Work?

Aaron407s, 2019-08-14:
Is pushing the 'night' button on the CS home screen supposed to effectively switch to AA night mode, or is this not fully implemented? I noticed that it hasn't switched anything to night mode in AA for quite a few versions, only the CS wallpaper. I believe the last time that it worked for me was with CS classic.

- What phone do you have? Samsung Galaxy S9

- What OS version? Android 9.0

- What crankshaft version? 2019-06-08 NG base + recompiled with patch for recent proto issues, but issue has existed for a solid year

Running in X11.

My sincere apologies if this has been captured somewhere else. I scoured the open and closed issues but couldn't find it.

gcharlie, 2019-08-14:
I can confirm this should still works. I have  the night mode function mapped to a gpio pin and when it activates google maps goes dark. A lot of the new UI though is dark by default, i think to be less distracting when driving but as a result less things change visually between day and night mode.

Aaron407s, 2019-08-14:
Thanks. Switching to night mode through the menu, then connecting my phone, ends up with maps still in the normal daytime mode. Maybe I'll experiment with the GPIO pin trigger option.

gcharlie, 2019-08-14:
There might be some options too that delegate the day/night over to the phone, so regardless of crankshaft AA only triggers night mode if the phone says so ...I thought I remember there being some setting like that. I'll dig through my settings tonight and see what I have things set to.

gcharlie, 2019-08-14:
Looks like this may be a samsung issue too
https://support.google.com/androidauto/thread/2482734?hl=en

Aaron407s, 2019-08-14:
Ah, interesting. I also just changed it in the AA developer settings to be car controlled rather than app controlled, so I'll see if it makes a difference. If not, maybe it is a samsung phone issue as you mentioned.

Aaron407s, 2019-08-14:
Well, it would appear that it was the developer setting for day/night control that was getting in the way. I had previously set it to be car controlled, but I must have reinstalled at some point. At any rate, it looks like there's no issue and it's working as expected now. Thanks.

closed 2019-08-14 by Aaron407s