opencardev/crankshaft #369: NG5.1 crash after pluging Phone

Darkcat78, 2019-08-14:
If you need support: Please post on https://www.reddit.com/r/crankshaft/ as this is a bug tracker, not a support forum.

Note: Reporting issues not based on official published builds will be closed without any comment.

- What phone do you have?
  OnePlus 6
- What OS version?
 Android 9
- What crankshaft version?
 NG5.1 08.08 (last version)
- Steps to reproduce the bug, if you're planning to report a bug. Please indicate whether the bug is always repoducible.
 Plug USB, device detected then crash when loading AA
- Please provide any further information that you might find helpful if available.

Google Play Services : 18.3.82

Hi,

Just did a fresh install of NG5.1 06.08, then updated to 08.08.
Running with BerryBoot.
Worked perfectly before on NG4 before touchscreen issues.

Checked Syslog and oalog
Can't create a debug.log file since im with Berryboot

Thanks for the help :)
[syslog.txt](https://github.com/opencardev/crankshaft/files/3503144/syslog.txt)

abraha2d, 2019-08-16:
@Darkcat78 Can you also attach the openauto log after the crash has occured?

abraha2d, 2019-08-16:
Looks like #371 is a duplicate of this one. Closing this in favor of that one since it has more info. Please comment there.

closed 2019-08-16 by abraha2d