opencardev/crankshaft #371: NG5.1 crashes if phone is connect via USB or Wlan

deejaypro, 2019-08-14:
- What phone do you have?
**One Plus 7 Pro**
- What OS version?
**Android 9**
- What crankshaft version?
**NG5.1**
- Steps to reproduce the bug, if you're planning to report a bug. Please indicate whether the bug is always repoducible.
**Open Auto crashes everytime i want to connect. It doesn't matter if its via USB or Wlan.
This is the errormessage in when issuing the "crankshaft syslog" command:
_Aug 14 23:08:22 CRANKSHAFT-NG service_openauto.sh[4167]: /opt/crankshaft/service_openauto.sh: line 182:  4751 Segmentation fault      stdbuf -i0 -o0 -e0 /usr/local/bin/autoapp >> /tmp/openauto.log
Aug 14 23:08:22 CRANKSHAFT-NG pi: SERV-LOGGER: --------------------------------------------------------------------
Aug 14 23:08:22 CRANKSHAFT-NG pi: SERV-LOGGER: Caller: /opt/crankshaft/service_openauto.sh
Aug 14 23:08:22 CRANKSHAFT-NG pi: SERV-LOGGER: Openauto crashed
Aug 14 23:08:22 CRANKSHAFT-NG pi: SERV-LOGGER: --------------------------------------------------------------------_**
- Please provide any further information that you might find helpful if available.
**Tested this with a fresh install and install which was updated from NG5 to NG5.1. Also i tried two different SD-Cards.**
[openauto.log](https://github.com/opencardev/crankshaft/files/3503703/openauto.log)
[syslog.txt](https://github.com/opencardev/crankshaft/files/3503704/syslog.txt)

Crescendo-BLYAT, 2019-08-15:
- What phone do you have?
**Mi 5s Plus (natrium)**
- What OS version?
**Lineage OS 16.0 - nightly build 20190809 - Stock Gapps**
- What crankshaft version?
**csng-alpha5.1**
- Steps to reproduce the bug, if you're planning to report a bug. Please indicate whether the bug is always repoducible.
**2 out of 5 when first time plugging in phone after Crankshaft boot will crash it.**
- Please provide any further information that you might find helpful if available.
**I'll get the log later, device's at home.**

Bishamon1987, 2019-08-15:
So I tried this also, 
It seems that Crankshaft NG 5.1 doesn't open AA when its connected via mini-jack (RCA) plug. But when I test it using HDMI there is no issue, it works flawlessly.

So maybe it has something to do with the video output part?

Do you guys use it over **HDMI or Mini-jack** (RCA) @Crescendo-BLYAT and @deejaypro ?

deejaypro, 2019-08-15:
I use the original Pi display via dsi cable. I can test it later If i can
reproduce it with an HDMI display.

deejaypro, 2019-08-15:
Just testet it with an hdmi display. It still crashes for me with this error:
Aug 15 17:32:41 CRANKSHAFT-NG service_openauto.sh[1682]: /opt/crankshaft/service_openauto.sh: line 182:  2567 Segmentation fault

Maybe i need to clarify it a littlebit better. Crankshaft doesnt crash when i'm connecting my phone to it. It crashed when i press the button to launch OpenAuto

[syslog_2.txt](https://github.com/opencardev/crankshaft/files/3506611/syslog_2.txt)
[openauto_2.txt](https://github.com/opencardev/crankshaft/files/3506612/openauto_2.txt)
[debug.zip](https://github.com/opencardev/crankshaft/files/3506614/debug.zip)

VaticanUK, 2019-08-16:
I'm also seeing the above behaviour every time I tap the AA button within Crankshaft. Tried connecting via WiFi and usb but neither works.

What phone do you have?
OnePlus 6T
What OS version?
Android 9 (OxygenOS 9.0.15)
What crankshaft version?
csng-alpha5.1
Steps to reproduce the bug, if you're planning to report a bug. Please indicate whether the bug is always repoducible.
Connect via WiFi or usb, them tap the AA button within crankshaft.
Please provide any further information that you might find helpful if available.
I'll get the log later, I'm away for a few days and don't have my rpi and screen with me.

SergioMuinhos, 2019-08-19:
What phone do you have?
OnePlus 6T
What OS version?
Android 9 (OxygenOS 9.0.16)
What crankshaft version?
csng-alpha5.1
Exactly the same thing happens to me, but I connect the phone after starting Crankshaft and stop trying to connect with the phone.
I have also tried to update it via Wi-Fi but it doesn’t take hours "downloading" but nothing happens

deejaypro, 2019-08-19:
Just a heads up:
For me everything works with the NG5.0 version except the touchscreen within AA. If i apply the workaroung, which was implemented with NG5.1, manually ([workaround](https://gist.github.com/eduardofilo/eced05b5fcafb86178834f4d708dcfa9)) everyting works fine. I can connect to my phone and the touchscreen within AA is working.

SergioMuinhos, 2019-08-22:
@deejaypro  that solution has not worked for me in NG5.1

pottsy-nz, 2019-08-30:
What phone do you have?
Pixel 3a + Generic 7" inch touchscreen via HDMI/usb (touch)
What OS version?
Android 9 
What crankshaft version?
csng-alpha5.1
Steps to reproduce the bug, if you're planning to report a bug. Please indicate whether the bug is always repoducible.
Connecting via usb, then tap the AA button within crankshaft. Crankshaft crashes with segfault.

Bishamon1987, 2019-08-30:
I had this same issue before. Me and @abraha2d found out some dependencies were missing. I will find it and add the issue in this post

abraha2d, 2019-08-31:
@Bishamon1987 added a Known Issues section to the changelog. https://github.com/opencardev/crankshaft/wiki/General-Changes---Changelog

pottsy-nz, 2019-08-31:
> I had this same issue before. Me and @abraha2d found out some dependencies were missing. Issue #366
> 
> `sudo apt install libxcb-icccm4 libxcb-image0 libxcb-keysyms1 libxcb-randr0 libxcb-render-util0 libxcb-shape0 libxcb-xinerama0`

That's for X11, I run EGL.  I'll try switching to X11 (which gives me a different crash) then run the workaound,

abraha2d, 2019-08-31:
@pottsy-nz Logs of the EGL crash would be nice.

matt2005, 2020-10-26:
fixed in alpha7

closed 2020-10-26 by matt2005