opencardev/crankshaft #372: Crankshaft and RCA-cable problem.

Bkmz2017, 2019-08-17:
What i have:
-samsung a3 2016, android 7
-Raspberry pi 3+ model b
-2019-08-08-crankshaft-ng-E2B068B

I'm trying to use crancshaft via RCA cable. When I use EGL-mode, crankshaft is crashed. When I use X11, crankshaft gui is starting, but my phone can't connect to crankshaft. I mean my phone is starting "Android Auto" and crankshaft is locking buttons and see my "Samsung Android", and then nothing happend.
When I use HDMI cable everything goes well, but I need to use RCA.
Please help!

Bishamon1987, 2019-08-17:
Got the same issue. 
And when i use the camoverlay.bin it says it cant find the displaybcm. Which i think indicates it tries to use a different video output (the HDMI). 
So maybe something is wrong in the settings with open auto also (which display output to use)

andreacampanella, 2019-08-19:
If I have to be honest with you, the RPi Composite output is garbage, if you really are keen to use it, use an HDMI to composite output, you will get way better results.

Bishamon1987, 2019-08-19:
It maybe so but I drove 5 hours today and my HDMI to composite converter died after 3 hours by overheating I think.
I rather have the raspi rca output because it works.

Bkmz2017, 2019-08-21:
> 
> 
> If I have to be honest with you, the RPi Composite output is garbage, if you really are keen to use it, use an HDMI to composite output, you will get way better results.

Maybe you are right, but why should I buy another HDMI-device?, I want to use onboarding device. And it perfectly work in older version of Crankshaft. Perfectly for me.

andreacampanella, 2019-08-24:
Because you get a way better image for you expirience, plus, that makes your setup more flexible, this way you are able to connect any HDMI capable to your car (?)

ivanixiv, 2019-08-29:
I got the same issue... I hope someone can solve it

abraha2d, 2019-08-31:
For future reference, there's a video of the problem over on #380 (thanks @ivanixiv)

matt2005, 2020-10-17:
@egisz I know you are using RCA. Have you experienced this issue with the latest release?

egisz, 2020-10-20:
I am using rca output on rpi4, egl mode.

Everything works fine, I only needed to enable in [config.txt](https://www.raspberrypi.org/documentation/configuration/config-txt/video.md) (only needed for rpi4).

`enable_tvout=1`

closed 2020-10-20 by matt2005