opencardev/crankshaft #373: problem connecting to WiFi hotspot of phone

Bishamon1987, 2019-08-18:
I changed the WiFi information in crankshaft_env.conf  (not sure this is spelled right)
I tried the first one which should be used for home network.
Of course I tried the second also.
Both have me no luck after changing the settings and rebooting.

ldehaas1612, 2019-08-20:
Hi Bishamom, I've had the same issue a while back and figured out that the file the wifi passwords are saved in does not get renewed. What I've done to get this fixed was reflash the SD card with the latest version of crankshaft and before continuing editing the SSID and password in the appropriate file. Then after first time boot these credentials are saved somewhere else and won't be reloaded untill reinstall. Hope this helps you out!

Bishamon1987, 2019-08-20:
Where should this file be located, then I can put it in with sudo nano filename I guess

ldehaas1612, 2019-08-20:
Okay, I've looked it up for you againand here it is:
Normally you would edit the settings for the wireless network in the /crankshaft/crankshaft_env.sh file. However after initial boot these settings are exported and encrypted into wpa_suplicant.conf, network0.conf and network1.conf if you got a second network setup.
Just reinstall crankshaft, edit the crankshaft_env file with the appropriate settings on your computer, then insert the card into the pi and boot. Hope this helps!

abraha2d, 2019-08-20:
When changing WiFi related settings in crankshaft_env.sh, make sure you also set `WIFI_UPDATE_CONFIG` to `1` to properly update the WiFi information in wpa_supplicant.

ldehaas1612, 2019-08-20:
Aaah, that explains the reloading settings from the file issue! Thanks for clarifying!

abraha2d, 2019-08-21:
@Bishamon1987 does this solve the issue?

Bishamon1987, 2019-08-21:
It did thanks!

closed 2019-08-21 by Bishamon1987