opencardev/crankshaft #374: Pulseaudio has detected no imputs!

SergioMuinhos, 2019-08-19:
If you need support: Please post on https://www.reddit.com/r/crankshaft/ as this is a bug tracker, not a support forum.

Note: Reporting issues not based on official published builds will be closed without any comment.

- What phone do you have?
OnePlus 6T
- What OS version?
OxygenOS 9.0.16
- What crankshaft version?
5.1
- Steps to reproduce the bug, if you're planning to report a bug. Please indicate whether the bug is always repoducible.
When you start CrankShaftNG it puts message:
-Pulseaudio has detected no imputs!
-Add an audio imput device
And it does not detect connected microphone and the microphone works correctly on PC.

- Please provide any further information that you might find helpful if available.
anymore

abraha2d, 2019-08-20:
What kind of microphone is it, and how are you connecting it?

agustinbonelli29, 2019-08-21:
I have the same problem, I think it should be some system configuration problem

SergioMuinhos, 2019-08-26:
@abraha2d Hi, I conected the microphone via USB, the screen shows Pulseaudio not detects inputs and Pulseaudo server stopped