opencardev/crankshaft #375: Enhancement: reduce audio judder when exiting Android Auto

gcharlie, 2019-08-19:
Theirs a very jarring audio glitch that happens when you exit Android Auto while audio is playing. I assume this has to do with the sound buffer and/or the audio stream coming from the phone, since the glitch just sounds like the last few bits of audio looping for a second or two. I'm not sure if this is a phone side discrepancy, but has anyone else experienced this?

abraha2d, 2019-08-20:
Personally, I haven't. Do you mean exiting android auto by pressing the Exit / Return to Crankshaft-NG inside Android Auto, or just by pulling the USB cable?

smartroad, 2019-08-23:
This happened on "professional" equipment to. I have a hire car at the moment and it will sometimes leave the audio playing for a short time. I think it is an issue with AA rather than CS.

abraha2d, 2019-08-23:
@gcharlie Tested both ways, and wasn't able to reproduce the glitch. What combination of phone/hardware are you using? If other people are experiencing the same thing, maybe we can find that it only happens with certain setups...

smartroad, 2019-08-23:
It happens with me both with CS and with Mitsubishi head unit. Phone wise, S10e, Nokia 8 and Samsung A(some number, its works phone lol)

gcharlie, 2019-08-24:
Fair enough, that makes sense. i guess there no need to keep this open then.

closed 2019-08-24 by gcharlie