opencardev/crankshaft #380: Strange problem changing video output

ivanixiv, 2019-08-29:
If you need support: Please post on https://www.reddit.com/r/crankshaft/ as this is a bug tracker, not a support forum.

Note: Reporting issues not based on official published builds will be closed without any comment.

- What phone do you have?
Xiaomi mi mix 3
- What OS version?
Android 9
- What crankshaft version?
5.1 NG
- Steps to reproduce the bug, if you're planning to report a bug. Please indicate whether the bug is always repoducible.
I just flashed the crankshaft 5.1 on my rpi3 sd, i use crankshaft without touchscreen so with a usb wireless command (smart tv like), so i run Crankshaft in x11.
If i run cranskhaft via hdmi, everyrhing works fine, i can even use Android Auto with the new interface
But if i run cranskhaft with my rpi3 videou output via RCA jack, cranskhaft goes crazy, after 5 sec it shows a black screen asking for login, like if i am in dev mode but i am in default, then if i connect my phone i just can see the name of my phone on the usb button in the crankshaft home, but nothing happens, Android Auto doesn't start... So, how it is possible that just changing video output type, crankshaft works in different way?

Here you can even see a video of the problem:

[Reddit original request](https://www.reddit.com/r/crankshaft/comments/cwr9km/strange_problem_with_video_output/?utm_medium=android_app&utm_source=share)

Bishamon1987, 2019-08-29:
#372

abraha2d, 2019-08-31:
Closing in favor of #372 (thanks @Bishamon1987)

closed 2019-08-31 by abraha2d