opencardev/crankshaft #381: Current Problem; Handy unload while Crankshaft is used

krooozong, 2019-09-10:
Maybe someone can help me. My handy unload itsself while using Crankshaft in my car.
I have a 5V 5A power adapter and use the rasperry pi 3 with a small vent on Top of it.

Furthermore i have a Mic on Usb, the 7 inch on Usb, and a Usb cable to my handy
My handy is a huawei p20 lite.

Everything works but for longer Roadtrips it is really annoying when my handy ist empty.
To solve the problem, i used to different cables but nothing changed.
I have already overwrite my config text to max Current
and i try to use two usb to support my handy but the problem stays.

Can I use the ethernet for a double power supply ?
Are there some Advices to solve the problem ?

Best regards

gcharlie, 2019-09-11:
This might be a hardware limitation. no matter how many and you have going into the pi, I don't think it's USB ports will put out more than .5amps per USB2 spec. On top of that some phones won't go past that when syncing data, only when charging exclusively.

gcharlie, 2019-09-11:
You could try using a USB hub that uses external power. That way the data lines go to the pi, but the phone will get it's power from the external source. But that may still not be enough depending on how power hungry your phone is. Since Android Auto turns on GPS continuously and is essentially streaming video to the Head unit, it might be using more power then it otherwise would be.

gcharlie, 2019-09-11:
https://www.instructables.com/id/Raspberry-Pi-3-External-Power-for-USB-Ports/

I did a similar mod on my setup and it gives my pixel 3xl just enough juice to slowly charge it.