opencardev/crankshaft #383: Crankshaft input microphone by bluetooth

agnoam, 2019-09-19:
Is there a way to set my bluetooth microphone as crankshaft's input device ?

thrill85, 2019-09-30:
I have the same issue. I mean it was working 1 or 2 month ago but now within AA when you trigger the assistant it seams like it doesn't hear anything from the mic. BTW speech and music channel  is unchecked on crankshaft. The strange thing is that AA picks up the mic when answering a message though. It has to be a crankshaft issue cause I know it still works with OEM AA devices and works when the phone is not connected to the pi

basemhegazy, 2020-06-13:
Hi, I am running the latest Crankshaft CarOS (2019-08-08 / Build E2B068B), it took me some tinkering to get things done specially for the HFP and phone calling, everything is working fine now.

To activate the USB sound card Mic and AUX over bluetooth (external BT dongle) you need to have both "Speech" and "Music" checked, otherwise it will use your phone speaker and microphone. 

From "Pulsaudio Input / Output Device" you need to select your Output (USB sound card AUX) and Input (USB sound card Mic). 

You can still test if it works by pressing on "Execute Audio Test" keep talking until your hear your voice, it's a 5 seconds record. Make sure to raise the Playback Volume & Capture Volume up.