opencardev/crankshaft #384: FM integration with rtl_fm

krisstakos, 2019-09-25:
I have different aproach for bringing regular FM radio in the crankshaft but I need some help. So to the idea:
Crankshaft supports .strm files and there is rtl_fm_streamer which turns generic rtl-sdr dongle in to internet radio streamer and can be accessed trough the assigned local ip address of the pi + there is JSON RPC api for setting the frequency.
I successfully compiled and tested the streamer. And tried making local file which contains the streaming address but the sound is choppy. So I dig around in the crankshaft source and guesed that for playing media it uses gst-launch-1.0(gstreamer)? (if anyone more familliar with the source can confirm it would be nice) I continue with my research and replicated exact choppy sound using mentioned tool and modifed my command so I can listen to the stream without problems.
Working command:
gst-launch-1.0 souphttpsrc location=http://localhost:2346/103800000/1 ! rawaudioparse use-sink-caps=false format=pcm pcm-format=s16le sample-rate=48000 num-channels=2 ! audioconvert ! audioresample ! autoaudiosink
The problem:
I need your help to integrate it. The idea is when the user press .strm, the executor looks inside the file and if there is 'localhost' in the address execute with the right command.
The goal:
You can make files with your favourite radios and just play them using only rtl-sdr dongle
Bump from #288

abraha2d, 2019-09-25:
Hey @krisstakos, i've been thinking of doing something to integrate FM radio into Crankshaft as well.
What's the difference between your command and the one that Crankshaft uses? (i'm still not very familiar with the source, couldn't find it with a quick look.)
It might be easier to just change the command across the board if it doesn't break playing back regular files.

krisstakos, 2019-09-26:
Hi, @abraha2d. I didn't find any command. From my observations I assumed that audio backend is held by gstreamer. So I started playing and replicated exact choppiness of the audio and then fixed it with above command. The problem was with encoding and some elements of the piping. 

> It might be easier to just change the command across the board if it doesn't break playing back regular files.

I don't think that would work. As you can see I supplied url to the stream in any other case it was file (mp3, strm etc)
My suggestion is to change a bit the logic as described above

gcharlie, 2019-10-10:
Hey @krisstakos I've been trying without much luck to get rtl_fm_streamer working on my crankshaft build. Did you run into issues getting it to work after building it? I logged in via ssh to my pi, went thru the build process, and it looks like it managed to build (or at least it didn't throw any obvious errors while compiling. but when I try to use the run command from the documentation `$ ./rtl_fm_streamer` I get nothing. 
Depending on where I run it from it'll say either `-bash: ./rtl_fm_streamer: Is a directory` or `-bash: ./rtl_fm_streamer: No such file or directory`

Any idea what I might be doing wrong?

gcharlie, 2019-10-11:
scratch that. got the streamer working. Build/install instructions not very clear

gcharlie84, 2019-10-16:
@krisstakos One very limited work around that is working for me is to use custom button, 5 of them using your command to set and play 5 radio stations and the 6th to shut the stream off

krisstakos, 2019-10-16:
@gcharlie84  that's cool! I managed to find that QMediaPlayer uses gstream for backend as I said. Also poke around the code and hope soon to have fully working sdr fm

gcharlie, 2019-10-16:
Awesome man, good luck getting it working! I tried poking around the backend myself, but most of it went way over my head unfortunately.

krisstakos, 2019-10-16:
I will share what I think are the key points in the code so you all can look into:
File: MainWindow.spp 
Line: 1490 (snippet 1503)
`        foreach (QString filename, mp3s) {
            // add to mediacontent
            if (filename.endsWith(".strm")) {
                QString url=configuration_->readFileContent(this->musicfolder + "/" + this->albumfolder + "/" + filename);
                content.push_back(QMediaContent(QUrl(url)));
                ui_->mp3List->addItem(filename.replace(".strm",""));`

As we can see it's getting the content from .strm files and push it in "content" variable

Line: 1360
This is triggered when you press the play button not sure about directly pressing song from album list

`void f1x::openauto::autoapp::ui::MainWindow::on_pushButtonPlayerPlayList_clicked()
{
    player->setPlaylist(this->playlist);
    playlist->setCurrentIndex(this->currentPlaylistIndex);`

// instead of calling play() we should implement our logic: 
// if 'content' at current index contains 'localhost' (that means it's our radio stream) 
//     -  Qprocess - executing the right gstream pipe with content(current index)
//     - we should somehow manage the process (kill/start new) when we choose different frequency
// else player->play()

   ` player->play();
    ui_->pushButtonBack->setIcon(QPixmap("://coverlogo.png"));
    ui_->mp3selectWidget->hide();
    ui_->PlayerPlayingWidget->show();
    ui_->pushButtonPlayerPlayList->hide();
    ui_->pushButtonList->show();
    ui_->pushButtonBackToPlayer->hide();
    ui_->pushButtonPlayerStop->show();
    ui_->pushButtonPlayerPause->setStyleSheet( "background-color: rgb(233, 185, 110); border-radius: 4px; border: 2px solid rgba(255,255,255,0.5); color: rgb(0,0,0);");
    ui_->pushButtonPlayerPause->show();
    int currentalbum = ui_->comboBoxAlbum->currentIndex();
    ui_->labelCurrentAlbumIndex->setText(QString::number(currentalbum+1));
    ui_->Info->show();
    ui_->horizontalSliderProgressPlayer->show();
    ui_->pushButtonAlbum->hide();
    ui_->mp3List->hide();
    ui_->AlbumCoverListView->show();
}`

krisstakos, 2019-10-16:
I made very ugly experiment and I can say it was successful !

krisstakos, 2019-10-19:
So that's my solution for the moment. It's kinda works. But I think you will get tha hang of it. Soon I will post better solution. I assume you already have rtl_fn_streamer compiled and running, better start it at boot of the pi.
0. Boot pi in dev mode
1. Rename attached file from *.txt to *.cpp
2. Put the file in openauto/src/autoapp/UI folder
3. Compile:
` cd ~
$ mkdir openauto_build
$ cd openauto_build
$ sudo apt-get install -y libtag1-dev libblkid-dev
$ cmake -DCMAKE_BUILD_TYPE=Release -DRPI3_BUILD=TRUE -DAASDK_INCLUDE_DIRS="/home/pi/aasdk/include" -DAASDK_LIBRARIES="/home/pi/aasdk/lib/libaasdk.so" -DAASDK_PROTO_INCLUDE_DIRS="/home/pi/aasdk_build" -DAASDK_PROTO_LIBRARIES="/home/pi/aasdk/lib/libaasdk_proto.so" ../openauto
$ make

4. Replace binary in Crankshaft:

$ sudo cp /home/pi/openauto/bin/autoapp /usr/local/bin/autoapp`

5. reboot and njoy
[MainWindow .txt](https://github.com/opencardev/crankshaft/files/3746908/MainWindow.txt)

FGameCZ, 2019-11-14:
> So that's my solution for the moment. It's kinda works. But I think you will get tha hang of it. Soon I will post better solution. I assume you already have rtl_fn_streamer compiled and running, better start it at boot of the pi.
> 0. Boot pi in dev mode
> 
> 1. Rename attached file from *.txt to *.cpp
> 2. Put the file in openauto/src/autoapp/UI folder
> 3. Compile:
>    ` cd ~
>    $ mkdir openauto_build
>    $ cd openauto_build
>    $ sudo apt-get install -y libtag1-dev libblkid-dev
>    $ cmake -DCMAKE_BUILD_TYPE=Release -DRPI3_BUILD=TRUE -DAASDK_INCLUDE_DIRS="/home/pi/aasdk/include" -DAASDK_LIBRARIES="/home/pi/aasdk/lib/libaasdk.so" -DAASDK_PROTO_INCLUDE_DIRS="/home/pi/aasdk_build" -DAASDK_PROTO_LIBRARIES="/home/pi/aasdk/lib/libaasdk_proto.so" ../openauto
>    $ make
> 4. Replace binary in Crankshaft:
> 
> $ sudo cp /home/pi/openauto/bin/autoapp /usr/local/bin/autoapp`
> 
> 1. reboot and njoy
>    [MainWindow .txt](https://github.com/opencardev/crankshaft/files/3746908/MainWindow.txt)

Please can you make a video tutorial?

krisstakos, 2019-11-14:
> Please can you make a video tutorial?
Sorry, I don't have time for video tutorials

bob5731, 2019-11-19:
But how are you guys going to connect it to the car antenna I think that's a different adapter than USB adapters I've seen online.

gcharlie, 2019-11-19:
You need an adapter, you can find them on eBay or even Amazon if you look closely enough, you just need to know exactly what type of connector your rtl sdr USB uses. This one works pretty well for me https://rover.ebay.com/rover/0/0/0?mpre=https%3A%2F%2Fwww.ebay.com%2Fulk%2Fitm%2F173677931826