opencardev/crankshaft #385: Black screen

Unactive, 2019-09-25:
Hi,

I tried different versions of Crankshaft on my pi. But on every version I just get a black screen. When I install raspbian on my USB it just works perfectly, but everytime I try to run Crankshaft it just doesnt do anything. I only see a black screen.

Bishamon1987, 2019-09-25:
Are you trying to use it with the rca output or HDMI?

Unactive, 2019-09-25:
Nope, HDMI also doesnt work.. My screen is still black

Bishamon1987, 2019-09-25:
Tried changing settings like x11 or egl. Changing config.txt

Unactive, 2019-09-25:
It just doesnt boot.. I enabled splashscreen.. I see the usual colored splash screen from the Rpi then it turns black and nothing happens..

Bishamon1987, 2019-09-25:
The screen turns of for about 1.5min then crankshaft is running. It's to hide the startup lines

Unactive, 2019-09-26:
Still no succes.. I still have the black screen. I will record it when I'm back home, and send the config file, So you can understand me better.

Unactive, 2019-09-26:
Here is the Video: https://youtu.be/EQcN-IQjbNg 
I use the latest IMG from https://github.com/opencardev/crankshaft/releases

choryuidentify, 2019-09-28:
It can be happen by SD card. Trying change card.

matt2005, 2020-10-17:
Can you try latest release alpha 55?

matt2005, 2020-10-26:
fixed in alpha7

closed 2020-10-26 by matt2005