opencardev/crankshaft #389: Pixel 4XL cant connect via Wifi

tompeel, 2019-10-23:
Note: Reporting issues not based on official published builds will be closed without any comment.

- Pixel 4XL

- Q 10 QD1A.190821.011

- Alpha 5.1

- AA version 4.7.593824 - release from the 4th October

Bug logs to follow but i think this is more of an issue with AA on googles side.

Enable Dev mode on AA app, Start head unit server all ok

If you then close AA the Head Unit Server also shuts down, like it cant run the background task...

If you dont close AA with HU enabled and press connect on the CS UI, nothing happens, it doesn't let you set up the new connection of AA.

Like i say, i will try to get the debug off the pi later but i suspect its google meddling.

abraha2d, 2019-10-29:
I don't have a Pixel to test with, but I do have a 1+6T with the Android 10 Open Beta, so maybe I will be able to reproduce it. I haven't been driving the car I have Crankshaft on lately...

iTom1192, 2019-11-07:
I'll add to this as I have the Pixel 4 (nonXL), and I  having the same issue. I'd really love to be able to use WiFi as usual USB mode doesn't keep my phone charged. =)

matt2005, 2020-10-17:
Can you try the latest release?

matt2005, 2021-02-08:
Please try following
Use latest dev release (currently alpha7.1)
Enable mobile hotspot on Phone,
Configure WIFI2 in /boot/crankshaft/crankshaft_env.sh
Set wifi_enable_update=1 in /boot/crankshaft/crankshaft_env.sh
Start crankshaft
Click on wifi connect button, it will error saying unable to connect
on phone start AA and then start headunit server
On Crankshaft click connect
Phone should beep and then AA should open in crankshaft.

closed 2021-02-08 by matt2005