opencardev/crankshaft #393: additional script error Can't open Display

Bishamon1987, 2019-10-31:
It seams crankshaft doesnt use x server right?
I'm running crankshaft in x11.
But when I use python module pyautogui. It gives errors about not finding a display.

How can I fix this?
Ps: Startx also won't work

Bishamon1987, 2019-11-03:
**### More information**
**my /etc/rc.local file contains**
`# By default this script does nothing.

RED=`tput setaf 1 -T xterm`
GREEN=`tput setaf 2 -T xterm`
YELLOW=`tput setaf 3 -T xterm`
CYAN=`tput setaf 6 -T xterm`
MAGENTA=`tput setaf 5 -T xterm`
RESET=`tput sgr0 -T xterm`
BOLD=`tput bold -T xterm`

systime=`date`
echo "[${CYAN}${BOLD} INFO ${RESET}] *******************************************************" > /dev/tty3
echo "[${CYAN}${BOLD} INFO ${RESET}] Systemtime: $systime" > /dev/tty3
echo "[${CYAN}${BOLD} INFO ${RESET}] *******************************************************" > /dev/tty3

sudo /usr/bin/python3 /home/pi/test.py $
exit 0`

**My error is the following:**
`Nov  3 13:29:23 CRANKSHAFT-NG rc.local[1859]: Traceback (most recent call last):
Nov  3 13:29:23 CRANKSHAFT-NG rc.local[1859]:   File "/home/pi/test.py", line 3, in <module>
Nov  3 13:29:23 CRANKSHAFT-NG rc.local[1859]:     import pyautogui
Nov  3 13:29:23 CRANKSHAFT-NG rc.local[1859]:   File "/usr/local/lib/python3.5/dist-packages/pyautogui/__init__.py", line 94, in <module>
Nov  3 13:29:23 CRANKSHAFT-NG rc.local[1859]:     import mouseinfo
Nov  3 13:29:23 CRANKSHAFT-NG rc.local[1859]:   File "/usr/local/lib/python3.5/dist-packages/mouseinfo/__init__.py", line 149, in <module>
Nov  3 13:29:23 CRANKSHAFT-NG rc.local[1859]:     _display = Display(os.environ['DISPLAY'])
Nov  3 13:29:23 CRANKSHAFT-NG rc.local[1859]:   File "/usr/lib/python3.5/os.py", line 725, in __getitem__
Nov  3 13:29:23 CRANKSHAFT-NG rc.local[1859]:     raise KeyError(key) from None
Nov  3 13:29:23 CRANKSHAFT-NG rc.local[1859]: KeyError: 'DISPLAY'`

I did try 
`sudo DISPLAY=:0 /usr/bin/python3 /home/pi/test.py $` 
in the rc.local file but it doesnt work either.

Anyone has any ideas?

abraha2d, 2019-11-06:
The X server hasn't started when scripts in rc.local are run. Checkout `/home/pi/.xinitrc` to run scripts when the X server is started.

closed 2021-03-12 by matt2005