opencardev/crankshaft #394: How to add mouse and keyboard support

bob5731, 2019-11-07:
How do I add mouse and keyboard support?
I'll be using it with an existing head unit.

If you need support: Please post on https://www.reddit.com/r/crankshaft/ as this is a bug tracker, not a support forum.

Note: Reporting issues not based on official published builds will be closed without any comment.

- What phone do you have?

- What OS version?

- What crankshaft version?

- Steps to reproduce the bug, if you're planning to report a bug. Please indicate whether the bug is always repoducible.

- Please provide any further information that you might find helpful if available.

- Starting with pre4: Boot into dev mode (can be selected in settings).
  Reproduce the possible bug. Create the debug logs by pushing the grey button on bottom of screen.
  Add the created file (/boot/debug.zip) to the issue.

  Warning: if you have configured wifi passwords they will be blanked by script during creation.
  We can't make absolutly sure you passwords are removed so do a personal check of the files inside the zip.

matt2005, 2020-10-01:
USB Mouse and Keybaord already work, have you tried connecting one to the Pi?

egisz, 2020-10-20:
> USB Mouse and Keybaord already work, have you tried connecting one to the Pi?

I believe users are confused as mouse cursor is not visible by default.
It has to be enabled in settings, bu it's not that easy if you dont have touch screen...

It would be nice if mouse device is detected, it would enable cursor automatically.

timriker, 2023-12-31:
How is the mouse cursor enabled in settings? Which settings?