opencardev/crankshaft #396: rpi4

thostie2002, 2019-11-08:
If you need support: Please post on https://www.reddit.com/r/crankshaft/ as this is a bug tracker, not a support forum.

Note: Reporting issues not based on official published builds will be closed without any comment.

- What phone do you have?

- What OS version?

- What crankshaft version?
when comes androidautp(crankshaft) on rpi 4?
- Steps to reproduce the bug, if you're planning to report a bug. Please indicate whether the bug is always repoducible.

- Please provide any further information that you might find helpful if available.

- Starting with pre4: Boot into dev mode (can be selected in settings).
  Reproduce the possible bug. Create the debug logs by pushing the grey button on bottom of screen.
  Add the created file (/boot/debug.zip) to the issue.

  Warning: if you have configured wifi passwords they will be blanked by script during creation.
  We can't make absolutly sure you passwords are removed so do a personal check of the files inside the zip.

RaspberryPiFan, 2020-01-02:
https://github.com/RaspberryPiFan/crankshaft

presslab-us, 2020-02-11:
From the work of @RaspberryPiFan and @matt2005 I have a Buster build working here:
https://github.com/presslab-us/crankshaft/tree/docker-buster

And prebuilts here:
https://github.com/presslab-us/prebuilts

I have tested with rpi3 and rpi4 and it's working on both.  Included are various fixes and modifications but there is undoubtedly more work to be done.  At the least Android Auto seems to be working.

modemlamer, 2020-02-12:
hi, can u give me some guidance to build that from scratch?
i tried that, but when i checkout https://github.com/presslab-us/crankshaft/tree/docker-buster
there is no config.example to modify... 
and when i download the zip, the build-script tell me there are no git rep files...

so i just build https://github.com/presslab-us/crankshaft/tree/docker-buster checkout with this script:
build-crankshaft-ng-full.sh

this build is working.. and i got an image... this image don't run on my Pi4 
but i checked with my Pi3 and there it runs with no issues... 
so.. which step I'm missing to build an working image running on my pi4 ?

can u make a step by step guide for beginners?

presslab-us, 2020-02-12:
You are checking out the main branch, that is why you do not see `config.example`, you need to switch to the other branch like "git checkout docker-buster" once you clone it.

This build is still a work in progress so not exactly beginner friendly but I can try to help where possible.

matt2005, 2020-02-15:
Wow, I'm glad my time taken to get pi-gen updated and buster working is being used. I really thought no one else would use it.

matt2005, 2020-02-18:
I've created a image release built from my fork [Crackshaft-ng on Buster](https://github.com/matt2005/crankshaft/releases/tag/v.1.0.0.20200214-alpha) in case anyone can't build it their self

WhereisBielefeld, 2020-06-07:
`I've created a image release built from my fork Crackshaft-ng on Buster in case anyone can't build it their self`

Im a bit new to Docker.. am i´m supposed to ad you image to Docker? On the other Hand i wrote your image to an SD-Card and tried to start my Rpi4 but it wont start. I thing I probably missunderstood something. 
@matt2005 Might you give me a small hint how to handel your file?

Besides this i got your Github-Files and tried to compile it myself with your build-docker.sh script. But at the end fails when copying to pigen/deploy because there seem to be no folder. Like i said, im new to docker but i really dont get it :-/

matt2005, 2020-10-01:
I've just built a new release and it'll be appearing here soon.

matt2005, 2020-10-16:
Please use latest release alpha 5.5

closed 2020-10-16 by matt2005