opencardev/crankshaft #397: disable logs from boot

agnoam, 2019-11-09:
Hey,
How do I hide logs from the boot (verbose) and show my custom `crankshaft/splash.png` screen until the system goes up ?

gcharlie, 2019-12-10:
I also have this problem haven't been able to find a solution. It only started happening after I did . a recent full reinstall on my SD card. On my prior install the  splash screen images always took precedent.

danielhanna88, 2020-02-20:
I'd also like to know if there's an easy way to fix this.

matt2005, 2021-02-10:
its due to plymouth writing to FB and then fkms kicking in with a blank fb

RuhanSA079, 2022-01-26:
Hello dear friends.
Sorry for reviving this old thread.
I have found a way during boot, that it will show the splash during init, using official Raspberry Pi 7 inch DSI display with touch.
Follow the instructions here:
https://gitlab.com/DarkElvenAngel/initramfs-splash

@matt2005 Maybe you can have a look at this?
Mine works, on boot, (1 second or so, before kernel messages overwrite, until crankshaft UI starts)
Would be nice if we can find a effective way to hide those messages, until the UI starts.