opencardev/crankshaft #399: Android auto crashes

BoothBayer, 2019-11-25:
- What phone do you have?
Galaxy S9

- What OS version?
Android version 9, Samsung G960U1

- What crankshaft version?
E2B068B, Build Date: 2019-08-08
-Specs
*pi3b
*official 7inch touchscreen
*no peripherals attached currently
*suitable power supply

- Please provide any further information that you might find helpful if available.
Earlier today, I enabled Bluetooth and connected with my phone. After doing so it caused android auto to only open for a few seconds before closing. I disabled Bluetooth, restarted, and now when I connect via USB AA only remains open for a few seconds to a minute or two. Just in case it matters, I have gpio pin 7 set to change crankshaft to night mode, I have a jumper cable running from the ground pin (34? I think) to "pin 7".
      In the android auto settings I was using the "Try the new Android Auto" mode, which launches you directly into google maps. With that enabled it would crash within a few seconds of opening android auto, now when I have it disabled it starts me on the home menu; however, when I open google maps it will crash in under a minute. This leads me to believe it could be a memory issue or some error with google maps.

Try new android auto = True:
[debug.zip](https://github.com/opencardev/crankshaft/files/3887291/debug.zip)

abraha2d, 2019-11-25:
The aasdk logs just look like a result of the phone disconnecting abruptly. Have you tried clearing app data for the Android Auto app? Had to do that recently to get it working again. I'm also using the "Try the new Android Auto" mode.

BoothBayer, 2019-11-26:
> The aasdk logs just look like a result of the phone disconnecting abruptly. Have you tried clearing app data for the Android Auto app? Had to do that recently to get it working again. I'm also using the "Try the new Android Auto" mode.

I tried the "clear all data" button in the Android Auto developer options on my phone, not sure if it did anything though.
Is there a way I can force the terminal to stay open so I can watch the device status?
Before it crashes I can see the terminal and it looks like my phone connects > disconnects > connects, but once restarted my phone isnt recognized in the usb.
This is all on a fresh build, I reflashed my sd card the other day and it worked (mostly)fine temporarily, until I enabled and disabled wifi.

BoothBayer, 2019-11-27:
I swapped to an official samsung usb cable, and that seemed to fix my problem? It must have been my phone disconnecting because of a bad cord. Doesn't explain why it crashed when using wifi before, but it now works fine with my phone as a hotspot for whatever reason.

closed 2019-11-27 by BoothBayer