opencardev/crankshaft #400: bad url in wiki for building crankshaft

Alexplose, 2019-11-27:
In this page : 
https://github.com/opencardev/crankshaft/wiki/Building-the-binary-blobs

This url is 404 :
Please refer to https://github.com/htruong/crankshaft/blob/master/COMPILE.md to start.

Maybe I'll need more help on how to build crankshaft, can you please elaborate on how to build.

Thanks

egisz, 2020-07-22:
Hi @Alexplose, 
I suggest you go to the more fresh fork of this repo https://github.com/fazerxlo/crankshaft
and build using docker container. It worked for me.

matt2005, 2020-09-22:
I'm currently reviewing the build process and will be updating once completed.

matt2005, 2020-09-22:
fixed, wiki page is for crankshaft  classic so repointed to correct file

closed 2020-09-22 by matt2005