opencardev/crankshaft #401: Running openauto using QtVideoOutput on different platforms

mozcelikors, 2020-01-02:
I am trying to run openauto/crankshaft on a different platform (TI DRA7xx) that supports Yocto. I have successfully managed to run the application with a lot of porting work. However, since my platform is not a Raspberry Pi OMXVideoOutput won't work. When I try to run using QtVideoOutput.cpp, video performance is very very slow (5 to 30 second delayed video), however the actions are instant. That means, when I click "Play" button, there is sound instantly, but the video wont refresh for 5-30 seconds.
At first look, I thought this is caused by hardware acceleration support. 
I believe MediaPlayer object used in QtVideoOutput is directly tied to gstreamer. However, I am able to play mp4 videos using gstreamer's playbin seamlessly. (Though playing .mkv through playbin is troublesome.)
Please correct me if I'm wrong: openauto/aasdk directly reads from USB (and not from the USB-Network gadget) and writes payload to a video buffer, which then is played on a VideoWidget. If this is the case, I am not familiar with this and therefore don't know any methods to improve the problem at hand.
So the issue remains and I need help identifying the possible causes for this problem.

Here is the log:

```
GST_DEBUG=3 /usr/bin/autoapp -platform eglfs                                                                                                          
Unable to query physical screen size, defaulting to 100 dpi.
To override, set QT_QPA_EGLFS_PHYSICAL_WIDTH and QT_QPA_EGLFS_PHYSICAL_HEIGHT (in millimeters).
[2020-01-02 16:32:14.213013] [0xb2795010] [info]    [OpenAuto] [OpenAuto] Display width  : 1280
[2020-01-02 16:32:14.213133] [0xb2795010] [info]    [OpenAuto] [OpenAuto] Display height  : 800
libpng warning: iCCP: known incorrect sRGB profile
[2020-01-02 16:32:14.323396] [0xb2795010] [info]    [OpenAuto] [Configuration] CS param found: BR_MIN= Value:30
[2020-01-02 16:32:14.323563] [0xb2795010] [info]    [OpenAuto] [Configuration] CS param found: BR_MAX= Value:255
[2020-01-02 16:32:14.323664] [0xb2795010] [info]    [OpenAuto] [Configuration] CS param found: BR_STEP= Value:25
[2020-01-02 16:32:14.323762] [0xb2795010] [info]    [OpenAuto] [Configuration] CS param found: BR_STEP= Value:25
inotify_add_watch("/media/USBDRIVES") failed: "No such file or directory"
[2020-01-02 16:32:14.600733] [0xb2795010] [info]    [OpenAuto] [Configuration] CS param found: WIFI_SSID= Value:sample
[2020-01-02 16:32:14.600908] [0xb2795010] [info]    [OpenAuto] [Configuration] CS param found: WIFI2_SSID= Value:sample
[2020-01-02 16:32:14.601530] [0xb2795010] [info]    [OpenAuto] [Configuration] CS param found: WIFI2_SSID= Value:sample
[2020-01-02 16:32:14.824301] [0xb2795010] [warning] [OpenAuto] [RecentAddressesList] failed to read configuration file: openauto_wifi_recent.ini, error: openauto_wifi_r.
[ConnectDialog] constructor
"[ConnectDialog] loadRecentList"
inotify_add_watch("/media/USBDRIVES/CSSTORAGE") failed: "No such file or directory"
[2020-01-02 16:32:14.998629] [0xb2795010] [info]    [OpenAuto] [App] constructor
[2020-01-02 16:32:15.030337] [0xb2795010] [info]    [OpenAuto] [App] waitForUSBDevice
[2020-01-02 16:32:15.030480] [0xafdff380] [info]    [OpenAuto] [App] Waiting for device...
[2020-01-02 16:32:15.037725] [0xaf5ff380] [info]    [OpenAuto] [App] Devices enumeration result: false
[ 8675.726638] usb 1-1.2: new high-speed USB device number 22 using xhci-hcd
[ 8675.857649] usb 1-1.2: New USB device found, idVendor=04e8, idProduct=6860
[ 8675.864560] usb 1-1.2: New USB device strings: Mfr=2, Product=3, SerialNumber=4
[ 8675.872033] usb 1-1.2: Product: SAMSUNG_Android
[ 8675.876727] usb 1-1.2: Manufacturer: SAMSUNG
[ 8675.881077] usb 1-1.2: SerialNumber: ad081603190a730447
[ 8675.914350] cdc_acm 1-1.2:1.1: ttyACM0: USB ACM device
[ 8677.225013] usb 1-1.2: USB disconnect, device number 22
[ 8677.546669] usb 1-1.2: new high-speed USB device number 23 using xhci-hcd
[ 8677.677558] usb 1-1.2: New USB device found, idVendor=18d1, idProduct=2d01
[ 8677.684469] usb 1-1.2: New USB device strings: Mfr=2, Product=3, SerialNumber=4
[ 8677.691882] usb 1-1.2: Product: SAMSUNG_Android
[ 8677.696523] usb 1-1.2: Manufacturer: SAMSUNG
[ 8677.702302] usb 1-1.2: SerialNumber: ad081603190a730447
[2020-01-02 16:32:24.844956] [0xae5ff380] [info]    [OpenAuto] [App] Device connected.
[2020-01-02 16:32:24.845004] [0xae5ff380] [info]    [OpenAuto] [App] Start Android Auto allowed - let's go.
[2020-01-02 16:32:24.858800] [0xb2795010] [debug]   [OpenAuto] [AudioInput] create.
[2020-01-02 16:32:25.271225] [0xb2795010] [debug]   [OpenAuto] [QtVideoOutput] create.
[2020-01-02 16:32:25.273760] [0xae5ff380] [info]    [OpenAuto] [AndroidAutoEntity] start.
[2020-01-02 16:32:25.273811] [0xae5ff380] [info]    [OpenAuto] [AudioInputService] start.
[2020-01-02 16:32:25.275886] [0xae5ff380] [info]    [OpenAuto] [AudioService] start, channel: MEDIA_AUDIO
[2020-01-02 16:32:25.276178] [0xae5ff380] [info]    [OpenAuto] [AudioService] start, channel: SPEECH_AUDIO
[2020-01-02 16:32:25.276281] [0xae5ff380] [info]    [OpenAuto] [AudioService] start, channel: SYSTEM_AUDIO
[2020-01-02 16:32:25.276477] [0xae5ff380] [info]    [OpenAuto] [SensorService] start.
[2020-01-02 16:32:25.276764] [0xae5ff380] [info]    [OpenAuto] [VideoService] start.
[2020-01-02 16:32:25.277027] [0xae5ff380] [info]    [OpenAuto] [BluetoothService] start.
[2020-01-02 16:32:25.277288] [0xae5ff380] [info]    [OpenAuto] [InputService] start.
[2020-01-02 16:32:25.281448] [0xaedff380] [info]    [OpenAuto] [AndroidAutoEntity] version response, version: 1.5, status: 0
[2020-01-02 16:32:25.281514] [0xaedff380] [info]    [OpenAuto] [AndroidAutoEntity] Begin handshake.
[2020-01-02 16:32:25.292984] [0xaedff380] [info]    [OpenAuto] [AndroidAutoEntity] Handshake, size: 2294
[2020-01-02 16:32:25.316870] [0xaedff380] [info]    [OpenAuto] [AndroidAutoEntity] continue handshake.
[2020-01-02 16:32:25.325794] [0xaf5ff380] [info]    [OpenAuto] [AndroidAutoEntity] Handshake, size: 51
[2020-01-02 16:32:25.326017] [0xaf5ff380] [info]    [OpenAuto] [AndroidAutoEntity] Auth completed.
[2020-01-02 16:32:25.329733] [0xaedff380] [info]    [OpenAuto] [AndroidAutoEntity] Discovery request, device name: Android, brand: samsung SM-G930F
[2020-01-02 16:32:25.329804] [0xaedff380] [info]    [OpenAuto] [AudioInputService] fill features.
[2020-01-02 16:32:25.329887] [0xaedff380] [info]    [OpenAuto] [AudioService] fill features, channel: MEDIA_AUDIO
[2020-01-02 16:32:25.329934] [0xaedff380] [info]    [OpenAuto] [AudioService] fill features, channel: SPEECH_AUDIO
[2020-01-02 16:32:25.329966] [0xaedff380] [info]    [OpenAuto] [AudioService] fill features, channel: SYSTEM_AUDIO
[2020-01-02 16:32:25.329996] [0xaedff380] [info]    [OpenAuto] [SensorService] fill features.
[... 56 lines cut ...]

```
root@dra7xx-evm:~# GST_DEBUG=3 gst-play-1.0 1.720.mp4 
Press 'k' to see a list of keyboard shortcuts.
Now playing /home/root/1.720.mp4
0:00:00.064528407 26028   0x167a60 WARN                 basesrc gstbasesrc.c:3480:gst_base_src_start_complete:<source> pad not activated yet
0:00:00.065758154 26028   0x167a60 WARN                 basesrc gstbasesrc.c:3480:gst_base_src_start_complete:<source> pad not activated yet
0:00:00.117791283 26028 0xb5dd0380 WARN                 qtdemux qtdemux.c:3008:qtdemux_parse_trex:<qtdemux0> failed to find fragment defaults for stream 1
0:00:00.118968490 26028 0xb5dd0380 WARN                 qtdemux qtdemux.c:3008:qtdemux_parse_trex:<qtdemux0> failed to find fragment defaults for stream 2
0:00:00.120738123 26028 0xb5dd0380 WARN                 basesrc gstbasesrc.c:2389:gst_base_src_update_length:<source> processing at or past EOS
0:00:00.149919580 26028 0xb5dd0380 WARN                   pulse pulsesink.c:615:gst_pulseringbuffer_open_device:<pulsesink0> error: Failed to connect: Connection refused
0:00:00.150325592 26028 0xb5dd0380 WARN                 playbin gstplaybin2.c:4660:autoplug_select_cb:<playbin> Could not activate sink pulsesink
0:00:00.159834177 26028 0xb5dd0380 WARN                    alsa conf.c:4857:parse_args: alsalib error: Unknown parameter AES0
0:00:00.159875656 26028 0xb5dd0380 WARN                    alsa conf.c:5017:snd_config_expand: alsalib error: Parse arguments error: No such file or directory
0:00:00.159906562 26028 0xb5dd0380 WARN                    alsa pcm.c:2501:snd_pcm_open_noupdate: alsalib error: Unknown PCM default:{AES0 0x02 AES1 0x82 AES2 0x00 AES3}
[... cut ...]
0:00:00.619280595 26028 0xb5dd0950 WARN                  ducati gstducatih264dec.c:423:gst_ducati_h264dec_set_sink_caps:<decoder> Using 0 frames for reordering
0:00:00.722923273 26028 0xb5dd0950 WARN                  ducati gstducatividdec.c:590:codec_process:<decoder> changing max-ref-frames in caps to 13
0:00:00.725435147 26028 0xb40019b0 WARN         audio-resampler audio-resampler.c:275:convert_taps_gint32_c: can't find exact taps
```

Granted, this looks like a platform-related platform, but I still would appreciate ideas. 
Thanks in advance.

sharavan-lab, 2020-12-23:
Hello everyone,
                      I am able to compile openauto in yocto. However I am stuck with this error with aasdk, is this due to cmake, can someone help with your insights. 

Regards
Sharavan

Compilation Issues as below..

/home/sharavan/sara_raspi/raspi_build/workspace/sources/aasdk-ice/include/f1x/aasdk/IO/Promise.hpp:35:1: error: expected class-name before '{' token
|  {
|  ^
| /home/sharavan/sara_raspi/raspi_build/workspace/sources/aasdk-ice/include/f1x/aasdk/IO/Promise.hpp:117:1: error: expected class-name before '{' token
|  {
|  ^
| /home/sharavan/sara_raspi/raspi_build/workspace/sources/aasdk-ice/include/f1x/aasdk/IO/Promise.hpp:198:1: error: expected class-name before '{' token
|  {
|  ^
| /home/sharavan/sara_raspi/raspi_build/workspace/sources/aasdk-ice/include/f1x/aasdk/IO/Promise.hpp:278:1: error: expected class-name before '{' token
|  {

matt2005, 2021-01-20:
i'm currently looking at yocto. Are you able to share your repo with your yocto work?

sharavan-lab, 2021-01-25:
Hello Matt,
    I am in warrior branch  This is issue is only due to cross compilation. I have raspbian/stretch (docker) running on top of POKY warrior  no issues observed.

bblayers.conf

BBLAYERS ?= " \
  /home/sharavan/sara_raspi/sources/poky/meta \
  /home/sharavan/sara_raspi/sources/poky/meta-poky \
  /home/sharavan/sara_raspi/sources/poky/meta-yocto-bsp \
  /home/sharavan/sara_raspi/sources/meta-raspberrypi \
  /home/sharavan/sara_raspi/sources/meta-openembedded/meta-oe \
  /home/sharavan/sara_raspi/sources/meta-openembedded/meta-perl \
  /home/sharavan/sara_raspi/sources/meta-openembedded/meta-python \
  /home/sharavan/sara_raspi/sources/meta-openembedded/meta-filesystems \
  /home/sharavan/sara_raspi/sources/meta-openembedded/meta-webserver \
  /home/sharavan/sara_raspi/sources/meta-openembedded/meta-multimedia \
  /home/sharavan/sara_raspi/sources/meta-openembedded/meta-networking \
  /home/sharavan/sara_raspi/sources/meta-openembedded/meta-gnome \
  /home/sharavan/sara_raspi/sources/meta-openembedded/meta-xfce \
  /home/sharavan/sara_raspi/sources/meta-qt5 \
  /home/sharavan/sara_raspi/sources/meta-virtualization \
  /home/sharavan/sara_raspi/raspi_build/workspace \
  "

sharavan-lab, 2021-01-25:
also you need to add c++ / g++ in the libraries list

matt2005, 2021-01-25:
Have you created a layer and recipe for aasdk and openauto?

sharavan-lab, 2021-01-26:
Yes I have the recipes for both open auto and aaasdk. Do you want me to upload

matt2005, 2021-01-26:
Yes please If you create a repo and then add the url in here i can compare against the one's i'm build at the moment. So for I've got aasdk building and have just started looking at openauto. Once I've done those then it'll be adding recipes for all to other Crankshaft applications/scripts. 

My aim is once its working I'll then add it to opencardev.

sharavan-lab, 2021-01-27:
Hello Matt,
  Thanks a ton, compiled asdk looking at your .bb file /  Working on
openauto, compilation seems to be smooth , errors are from including f1x
include directories. cmake is not finding the path for the include. If I
can be of any help let me know.

Thanks again
Regards
Sharavan

matt2005, 2021-01-27:
I'm currently having issues with rtaudio, I can't find a recipe for it.

matt2005, 2021-01-27:
@sharavan-lab suggest moving to the dedicated repo

sharavan-lab, 2021-01-28:
I have built rtaudio , will share the code

matt2005, 2021-01-28:
I've managed to build it too. Now just working on openauto.

adrianalin, 2021-02-10:
Hello matt2005, i managed to build openauto. You can find all my changes here https://github.com/adrianalin/csng_yocto.

matt2005, 2021-02-10:
Great thanks, please submit PR against my repo

GhostMonkeyFPS, 2023-04-13:
@sharavan-lab did you manage to fix this issue?

/home/sharavan/sara_raspi/raspi_build/workspace/sources/aasdk-ice/include/f1x/aasdk/IO/Promise.hpp:35:1: error: expected class-name before '{' token
| {
| ^
| /home/sharavan/sara_raspi/raspi_build/workspace/sources/aasdk-ice/include/f1x/aasdk/IO/Promise.hpp:117:1: error: expected class-name before '{' token
| {
| ^
| /home/sharavan/sara_raspi/raspi_build/workspace/sources/aasdk-ice/include/f1x/aasdk/IO/Promise.hpp:198:1: error: expected class-name before '{' token
| {
| ^
| /home/sharavan/sara_raspi/raspi_build/workspace/sources/aasdk-ice/include/f1x/aasdk/IO/Promise.hpp:278:1: error: expected class-name before '{' token
| {

I came across this problem as well.

Thanks in advance.

LeandreBl, 2023-07-02:
I'm kinda late but I found the problem, it's linked to the boost version, in which the includes needed for the template in Promise.hpp are ill formed.
I used boost 1.66.0 and it compiled

mazenelnahla, 2024-03-14:
Hello Everyone,

I'm Facing an issue when deploying open Auto on Raspberry Pi 5 (aarch64)

In file included from /home/av/aasdk/include/f1x/aasdk/IO/PromiseLink.hpp:24,
                 from /home/av/aasdk/src/Channel/Control/ControlServiceChannel.cpp:22:
/home/av/aasdk/include/f1x/aasdk/IO/Promise.hpp:35:1: error: expected class-name before ‘{’ token
   35 | {
      | ^
/home/av/aasdk/include/f1x/aasdk/IO/Promise.hpp:117:1: error: expected class-name before ‘{’ token
  117 | {
      | ^
/home/av/aasdk/include/f1x/aasdk/IO/Promise.hpp:198:1: error: expected class-name before ‘{’ token
  198 | {
      | ^
/home/av/aasdk/include/f1x/aasdk/IO/Promise.hpp:278:1: error: expected class-name before ‘{’ token
  278 | {
      | ^
make[2]: *** [CMakeFiles/aasdk.dir/build.make:174: CMakeFiles/aasdk.dir/src/Channel/Control/ControlServiceChannel.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:100: CMakeFiles/aasdk.dir/all] Error 2
make: *** [Makefile:91: all] Error 2

I checked the boost version and it is the last version Should I downgrade to boost 1.66.0, if so how can I downgrade?

Thanks in advance.