opencardev/crankshaft #402: crankshaft and messagebox

flopix77, 2020-01-10:
hello all, I have install zenity and i would like to know if it's possible to show  a messagebox on CS ?
i would like to add option with gpio triggers, to show a warning when your coolant or oil are too low ?

thanks

krisstakos, 2020-02-18:
Did you try it? What happend?

modemlamer, 2020-02-19:
I tried that too, it does not work that easy.
But if you run in X11 mode kivy will work.
BUT it will not pop up on top of Android Auto, it will be only on top of crankshaft...
AndroidAuto comes via omxplayer on the top most layer. Only the videofeed of the reversecam (also Videoplayer feed) can be seen in top of AndroidAuto.
I had the same idea, but canceld...
Maby it is possible to resize AndroidAuto to get a small taskbar....

flopix77, 2020-02-19:
i had try in SSH but nothing.

i think test Open Auto Pro, because Android Auto crash in crankshaft after few seconds since one week

robert5974, 2020-03-20:
people are working on the crashing issue.  There are two other openauto projects that do what you have suggested with resizing AA and having a frame with buttons and both are on github:

rsjudka/intelligent-auto
viktorgino/headunit-desktop

Both have their issues but worth checking out.  I've tried both.  Intelligent-Auto is actually designed currently to be a drop-in application anywhere openauto is used.  That one has a very professional layout as well.  Both of the projects are currently being worked on at present with very recent updates.