opencardev/crankshaft #404: AA returns to Crankshaft Splash Seconds after Starting

Aaron407s, 2020-02-19:
- What phone do you have? Samsung Galaxy S9
- What OS version? Android 9
- What crankshaft version? Alpha 5.1
- Steps to reproduce the bug, if you're planning to report a bug. Please indicate whether the bug is always reproducible. --Plug in phone, wait a few seconds.

- Please provide any further information that you might find helpful if available.

Testing the waters to see if anyone else is experiencing this, otherwise I'll likely try a fresh build. As of yesterday, plugging in my phone after CS is running has AA start as usual, but it abruptly returns to the CS home screen after a few seconds. The amount of time isn't consistent, usually 1 to 10 seconds, and the odd time it continues running. It appears that AA is still technically running on the phone after this (it still has the lock screen present). I tried a different cable with no change. It looks like Google Play Services was updated over a week ago, but, since no issues arose until yesterday, I assume that's not the culprit this time.

Has anyone else had this issue?

presslab-us, 2020-02-19:
I'm running an experiment rpi4 build but I get that sometimes too.  Can you look at `/tmp/openauto.log` after the crash?  I see error 29 which seems to be MESSENGER_INTERTWINED_CHANNELS.

Aaron407s, 2020-02-19:
I tried snagging a log, but had difficulties getting it into debug and saving properly after the issue would happen. After getting debug properly enabled, AA wouldn't automatically launch, and even tapping the USB button on the CS screen couldn't get it to enter AA anymore. The log shows the following error:

usb hub error: AaSdk error code: 30, native code: 0

Any suggestions? Is this possibly a failing USB port on the pi?

presslab-us, 2020-02-19:
It looks like your issue is different from mine.  30 means OPERATION_ABORTED, which probably means the USB has disconnected.

Maybe you don't have enough power?  You can also check `dmesg` to see if the kernel has registered the USB disconnecting.

Aaron407s, 2020-02-19:
Thanks for the info. The power supply is 3A and I haven't had any issues before, including no lightning bolt to suggest power issues.

I checked the kernel debug log, and the only time it shows the USB disconnecting is when I physically disconnected it. I noticed before (when I could get AA launched to start with) that it would still show the AA lockscreen on the phone after returning to the CS screen, which suggests that the phone thinks CS is still functionally projecting and is connected.

I'll see if I can get a few more/cleaner logs.

thericcer, 2020-02-19:
I received the same symptom yesterday. Made no changes to my setup (hard wired pi to 4A SMPS and display plugged into a USB adapter). I get the same error message as presslab, error 29. I'll upload the log later.

I suspect that there was an update to AA that caused this.

Aaron407s, 2020-02-19:
I was previously thinking that the original log file that I had found was an old one, but I'm remembering now that I hadn't gone through debugging since my last clean update, and the first debug file is probably legit. Lots of "channel error: AaSdk error code: 29, native code: 0" errors.

[openauto.log](https://github.com/opencardev/crankshaft/files/4227187/openauto.log)

presslab-us, 2020-02-19:
Ah, well I'm happy to hear it's not just my rpi4 build that has this problem...  The next step I was going to do in debugging was to get the debug log from the AA app on the phone, but I haven't had the time yet.

thericcer, 2020-02-19:
Here's the debug log I just pulled, same as Aaron407, lots of "channel error: AaSdk error code: 29, native code: 0" errors.
[openauto2.log](https://github.com/opencardev/crankshaft/files/4227238/openauto2.log)

thericcer, 2020-02-19:
Also, I'm using a Pixel 3 with an rpi3.

Aaron407s, 2020-02-19:
I also notice a "usb hub error: AaSdk error code: 30, native code: 0" at the bottom of your log. Definitely seems like we're fighting with the same issue overall.

Aaron407s, 2020-02-19:
Here's a full debug dump from what seemed like a clean error event. I powered up, plugged the phone in, it ran AA for a few seconds before returning to the CS screen, then I wrote the log. I didn't physically disconnect the phone after the error, contrary to what the kernel log suggests.

To be honest, I'm not sure exactly what to look for, but I'm happy to test and report back for anyone who knows more than me. I don't use wifi, so there shouldn't be anything in the logs that would be considered sensitive information (please correct me if I'm wrong).

[debug_2019-02-19_1.zip](https://github.com/opencardev/crankshaft/files/4227957/debug_2019-02-19_1.zip)

presslab-us, 2020-02-20:
I wonder if the logcat from Android will give some insight.  There is a Developer setting in AA to give a full debug log.

Jiri-sedlacek207, 2020-02-20:
I noticed very similar behavior. Started to occur yesterday. I updated AA successfully 2 weeks before.

Another clue: After several crashes I forced AA on phone to stop and cleared all data. Then connected to crankshaft. It worked normally in navigation mode. Then I switched to music and started playback. Still worked ok. Switched back to navigation -> crash within a second or such... Next reconnections -> still the same, starts and crashes almost immediately.

thericcer, 2020-02-20:
@Jiri-sedlacek207 interesting, I did the same (force stopped and cleared data) but had no change. I believe I'm on the latest (from Aug last 2019?) but I will try re-flashing this weekend.

I've also tried all screen resolutions, with and without Bluetooth enabled, and developer and release mode.

gregorhollmig, 2020-02-20:
Same for me. I tested with two phones, 13th of Feb, one (OnePlus 7) had this issue and the other one (OnePlus 6) didn't have it. Now both have it.

presslab-us, 2020-02-20:
Here is the source of the error.  It looks like the problem is that a frame is received that does not match the channel of the previous frames in that stream.

Will-SCHOTT, 2020-02-21:
Same issue here, posted issue #405 by mistake, didn't pay attention it was a duplicate from this one.

In addition to discussion, I tried to place a fresh CS install, and resulted the same.

Only thing I noticed as stated in issue #405 is that, it remains stable until sound is in use, @Aaron407s , can you try to confirm while starting AA with no audio app (Spotify, whatever..) started? then if stable, try to start any of those, at least for me, this reproduces the bug.

thericcer, 2020-02-21:
@Will-SCHOTT I can confirm this reproduces the bug also.

If I uninstall/reinstall AA it kills whatever audio program was running (not sure how otherwise) and AA will run just fine. It seems that any audio output causes this. I've tried Spotify and google maps. If google maps is muted however, AA does not crash.

DoughnutWuff, 2020-02-21:
Hey all,

I thought I was going insane! Mine started doing this at start of last week. Crash when Audio starts playing/navigation is selected. I found a workaround using Bluetooth for audio but it does crash after a couple of minutes. Best way at the minute (and the longest way) is bluetooth for audio and a WiFi hotspot from my phone to the Pi.

`[2020-02-21 18:15:56.028956] [0x6eafb300] [info]    [OpenAuto] [AndroidAutoEntity] requested audio focus, type: 1
[2020-02-21 18:15:56.029056] [0x6eafb300] [info]    [OpenAuto] [AndroidAutoEntity] audio focus state: 1
[2020-02-21 18:15:56.079010] [0x6fafd300] [info]    [OpenAuto] [AudioService] start indication, channel: MEDIA_AUDIO, session: 0
[2020-02-21 18:16:00.218744] [0x6fafd300] [error]   [OpenAuto] [VideoService] channel error: AaSdk error code: 29, native code: 0
[2020-02-21 18:16:00.218883] [0x6e2fa300] [error]   [OpenAuto] [AudioService] channel error: AaSdk error code: 29, native code: 0, channel: MEDIA_AUDIO
[2020-02-21 18:16:00.218935] [0x6e2fa300] [error]   [OpenAuto] [SensorService] channel error: AaSdk error code: 29, native code: 0
[2020-02-21 18:16:00.218970] [0x6e2fa300] [error]   [OpenAuto] [SensorService] channel error: AaSdk error code: 29, native code: 0
[2020-02-21 18:16:00.219005] [0x6e2fa300] [error]   [OpenAuto] [AudioInputService] channel error: AaSdk error code: 29, native code: 0
[2020-02-21 18:16:00.219039] [0x6e2fa300] [error]   [OpenAuto] [BluetoothService] channel error: AaSdk error code: 29, native code: 0
[2020-02-21 18:16:00.219070] [0x6eafb300] [error]   [OpenAuto] [AndroidAutoEntity] channel error: AaSdk error code: 29, native code: 0
[2020-02-21 18:16:00.219103] [0x6f2fc300] [error]   [OpenAuto] [AudioService] channel error: AaSdk error code: 29, native code: 0, channel: SYSTEM_AUDIO
[2020-02-21 18:16:00.219176] [0x6fafd300] [error]   [OpenAuto] [AudioService] channel error: AaSdk error code: 29, native code: 0, channel: SPEECH_AUDIO
[2020-02-21 18:16:00.219206] [0x6eafb300] [info]    [OpenAuto] [App] onAndroidAutoQuit.
[2020-02-21 18:16:00.219246] [0x6eafb300] [info]    [OpenAuto] [AndroidAutoEntity] stop.
[2020-02-21 18:16:00.219290] [0x6eafb300] [info]    [OpenAuto] [AudioInputService] stop.`
[oalog.txt](https://github.com/opencardev/crankshaft/files/4237471/oalog.txt)

AndAug76, 2020-02-22:
Hi folks!

Same for me. Since 2 Days, just like Aaron407s described it. Either AA falls back to CS immediately or after 10 minutes, or sometimes it runs overnight. Sure, without being served, just as a test. It's not the audio output. Once it's running, I can listen to music, navigate or even make a phone call. It's just not reliable.

I thought about the USB cable or port on my S10 mobile, first. I then tried an S8 with Android 9, an S10 with Android 10, three different USB-C cables, an RPi 3 and a replacement RPi2, as well as two different SD cards. All bring up the same error. Finally, I tested the S10 in our new car with the built-in AA headunit. There everything works perfectly.

Since I am not a software professional, I can't contribute anything, except the information that none of the mentioned constellations work, except the factory installed one.

I have activated the debug mode, but I cannot find a log-file on the SD card. To tell the truth, I even have to learn how to get a log file from the system... I just wat to use CS/OA AA in my oldtime cars. Nothing more, nothing less. I'm just a user.

I'm going to try my hand at digging up a log.

ConzM, 2020-02-22:
Hello everyone,

same problem here since 3 days.
I tried to downgrade the Google Play Services, maybe an update of Google causing it, but in my case that was no solution. If I let the sound come out of the phone everything seems to run perfect.
I hope someone can help soon.

tsukisan, 2020-02-23:
I'm experiencing the same issue with a Pixel XL 1st gen. using pi 3b+ and pi foundation touch screen.

I usually use Pandora but if I pause it before it crashes then it will work without issue, haven't tried navigation yet though, just usually have maps pulled up.

One other piece to note, the phone does not recognize that AA has disconnected and continues to play audio and has the notifications for AA still visible. It seems that it  is crashing in a weird state and the phone doesn't timeout the connection for some reason.

Aaron407s, 2020-02-24:
I guess the question is, is there any kind soul out there with the expertise to dig into this? I wish I had more of a coding background, but I'll do anything I can to support anyone who has the ability :-).

stickycitrus, 2020-02-24:
Yes I too am suddenly experiencing this issue.. Back when I used to run OpenCarDev this was a problem. (Reason I jumped to crankshaft!). That was some issue between aasdk and google play services.. When things like this happen you have to look at whats changed.. I havent looked yet but whats the bet we all received an update to play services or AA or similar on our devices??

stickycitrus, 2020-02-24:
AA was update on 20th Feb 2020.. Probably related..

Aaron407s, 2020-02-24:
I started having the issue earlier than that. I'm guessing it was an automatically updated background service that caused the issue, similar to a year ago.

stickycitrus, 2020-02-24:
and Play services 20.04.14 was released about 12 days ago.. Maybe its worth looking at that then

stickycitrus, 2020-02-24:
Ive just rolled back my google play services to 18.8.16.. Will be back out to my car in about an hour to test and see if crashing is happening

stickycitrus, 2020-02-24:
Ive rolle my play services back to 17.7.86. will be back in my car in an hour.. fingers crossed the roll back will solve it

Aaron407s, 2020-02-24:
That was a short term fix a year ago as well, but I don't believe there's a way to block the background auto updates on non-rooted devices. Even if it works temporarily, though, it could help narrow down the problem. Thanks for joining in the dig for a solution.

stickycitrus, 2020-02-24:
No dramas. well at least if its a temp workaround it will help maybe with the fix in dev potentially

stickycitrus, 2020-02-24:
NO dice! rolling back play services made no difference. Noted though that it only crashes when audio is playing as per @thericcer 's installation

tuba59, 2020-02-25:
I started having this issue on my S9+ after the Android 10 upgrade. Tried a factory reset, did not work.

sjdean, 2020-02-25:
I don't use CrankShaft myself, I use OpenAuto/AASDK and making some custom modifications for that. The issue exists there as well. I've been hit and miss using a Google Pixel 2 XL. Sometimes it works flawlessly, othertimes it kicks me out. Completely random. Today, I switched to Google Play Music, and it was working absolutely fine.

Put Waze on for a few seconds and got kicked me out.

So the question is, what does everyone use for their Navigation, and were you in Navigation when it crashed?

tuba59, 2020-02-25:
> I don't use CrankShaft myself, I use OpenAuto/AASDK and making some custom modifications for that. The issue exists there as well. I've been hit and miss using a Google Pixel 2 XL. Sometimes it [...]

I also was using Waze when the crash happens. I noticed, that when I connect my phone and Waze opens up automatically everything seems to be fine, then I press play on Spotify and do some swiping on the screen and the crash happens. If I don't play any music in the background Waze is running just fine for x amount of time, but then suddenly it crashed again when I tried this experiment. Maybe AA did some communication with Spotify which resulted in starting Spotify and causing the error.

Jiri-sedlacek207, 2020-02-25:
My experience is that crankshaft quits to UI only when combining map and music playback. While I was playing music from google music all was ok. The same applied when I opened map without music. But when I started music playback in map the crash occured almost immediately.
I've a suspicion whether the pop-up informing about played song could cause the crash.... When in navigation the playback worked for some seconds but crashed immediately when pop-up occured. But don't know, that may have been just a co-incidence.

sjdean, 2020-02-25:
Do you use Waze or Google Maps Jiri?

Aaron407s, 2020-02-25:
I have the issue and use Maps. I've also had it when switching tracks in play music.

Jiri-sedlacek207, 2020-02-25:
> 
> 
> Do you use Waze or Google Maps Jiri?

I use Google Maps

AndAug76, 2020-02-26:
I use g.maps.

sjdean, 2020-02-26:
Looks like it doesn't matter what audio app you use, or what maps you use, or indeed what you're doing. The AASDK randomly crashes.

I'll try and do some debugging. What seems to happen is that as others have said, if the frame channel doesn't match the channel, then it simply rejects the frame and resets the message. Whether that functionarlity should be changed I don't know.

moreni57, 2020-02-26:
I have same issue when i use Gmaps i switch spotify i click play music and crash at start menu.

PaoloDeMarcoManzano, 2020-02-26:
Me too same problem, at the first vocal message of the navigator (waze or google maps) or when I switch any music app, CS crash at start menu. Smartphone huawey and asus. Thanks a lot!

moreni57, 2020-02-26:
I have a Samsung A40

scavenrage, 2020-02-27:
Same here, oneplus6 with rpi3. Android auto crashes after few seconds. I'll try to upgrade crankshaft and I'll tell you the results

robert5974, 2020-02-27:
I've tried it with using the hotspot connection but there is no internet for me doing it that way so I tried a hotspot from my phone which worked until it crashed openauto.  I haven't seen anyone else report this, when my build crashes back to the menu, I get the error for USB saying " this isn't an MTP device".  Anyone else see that?

Aaron407s, 2020-02-28:
Well, a little more testing seems to be leading somewhere. It appears that it's being caused by a conflict between the speech and music channels. Everything (other then hearing speech) seems to work perfectly if you go into the CS settings, disable the speech channel, and reboot. If you're like me and use it primarily for music playback, this could be something to try for the time being until things are fixed, but you won't be able to hear the Google assistant voice (although you can still use voice commands for navigation, music control, etc.).

robert5974, 2020-02-28:
I'll try it and see.  Thanks.  I'd like to have the assistant voice but perhaps this lends to maybe an update to the assistant more than anything else??? Would have to look into that though.  It's just speculation

Aaron407s, 2020-02-28:
Sorry, may have gotten hopes up. Although it was working before with the speech channel disabled, it's still acting up again for me now.
*sigh*

robert5974, 2020-02-28:
Yeah it crashed for me too trying it. I just created a debug.zip.  I'll review that later.  If I find something I'll report back

thericcer, 2020-02-28:
+1 for the try and fail of disabling speech audio. Worked for a while, then crashed.

stickycitrus, 2020-02-28:
I notice If I dont have the map up on screen.. Im seem to be not getting the crash. Unfortunately The map comes up on startup and I just quickly go to media screen.. Then I can drive and listen to music and it doesnt crash. so Seems related to google maps for me.

PaoloDeMarcoManzano, 2020-02-28:
Nothing from the developer?

sjdean, 2020-02-28:
https://github.com/opencardev/aasdk/blob/1d91dec2401716c4f93d1cd450607d2ffab39e96/src/Messenger/MessageInStream.cpp#L73

Commenting out lines 73, 74, 75 in this block resolves the issue, but unfortunately breaks the speech channel - if you play a notification, Google tells you "Something Went Wrong".

More debugging to do, but this is the area.

For some reason.

PaoloDeMarcoManzano, 2020-02-28:
> https://github.com/opencardev/aasdk/blob/1d91dec2401716c4f93d1cd450607d2ffab39e96/src/Messenger/MessageInStream.cpp#L73
>
> Commenting out lines 73, 74, 75 in this block resolves the issue, but [...]

Sorry, on my SD card I can’t find the file MessageInStream.cpp!
Can you help me with the path.
Many thanks!

sjdean, 2020-02-28:
Sorry Paolo, you're not going to find that on the card. This is in the source code. You just get the compiled versions.

For anyone else who's looking at the code, the last crash I had, was having a message channel id of 3 and a framebuffer channel id of 4. This mismatch generates the "MESSENGER_INTERTWINED_CHANNELS ". 

Looks like Im going to have an interesting weekend of debugging. :-)

robert5974, 2020-02-28:
> [debug.zip](https://github.com/opencardev/crankshaft/files/4240729/debug.zip)
> Hi folks!
> 
> Same for me. Since 2 Days, just like Aaron407s described it. Either AA falls back to CS immediately or after 10 minutes, or sometimes it runs overnight. Sure, without being served, just as a test. It's not the audio output. Once it's running, I can listen to music, navigate or even make a phone call. It's just not reliable.
> 
> I thought about the USB cable or port on my S10 mobile, first. I then tried an S8 with Android 9, an S10 with Android 10, three different USB-C cables, an RPi 3 and a replacement RPi2, as well as two different SD cards. All bring up the same error. Finally, I tested the S10 in our new car with the built-in AA headunit. There everything works perfectly.
> 
> Since I am not a software professional, I can't contribute anything, except the information that none of the mentioned constellations work, except the factory installed one.
> 
> I have activated the debug mode, but I cannot find a log-file on the SD card. To tell the truth, I even have to learn how to get a log file from the system... I just wat to use CS/OA AA in my oldtime cars. Nothing more, nothing less. I'm just a user.
> 
> I'm going to try my hand at digging up a log.
> 
> Cheers!
> Andreas
> 
> Edit: OK, found that Button.... But wich file from the debug.zip is needed?

I look through everything personally...at least the newest entries of each file created in the debug.zip

robert5974, 2020-02-28:
There is a new Android Auto update on the play store.  Changelog says something about turning off media notifications and more.  I updated but I'll have to check it when I get home

Aaron407s, 2020-02-28:
I'm already running the new version. No change for me, unfortunately.

parkerlreed, 2020-02-28:
The phone side update causes it to completely not connect anymore.

```
[parker@stealth ~]$ autoapp 
[2020-02-28 12:17:57.798398] [0x00007f6aebac9300] [warning] [OpenAuto] [RecentAddressesList] failed to read configuration file: openauto_wifi_recent.ini, error: openauto_wifi_recent.ini: cannot open file. Empty list will be used.
[2020-02-28 12:17:57.893602] [0x00007f6ae92bd700] [info]    [OpenAuto] [App] Waiting for device...
[2020-02-28 12:17:58.015291] [0x00007f6ae37fe700] [info]    [OpenAuto] [App] Devices enumeration result: true
[2020-02-28 12:18:17.396185] [0x00007f6ae37fe700] [info]    [OpenAuto] [App] Device connected.
[2020-02-28 12:18:17.398496] [0x00007f6aebac9300] [debug]   [OpenAuto] [AudioInput] create.
[2020-02-28 12:18:17.404909] [0x00007f6aebac9300] [debug]   [OpenAuto] [QtVideoOutput] create.
[2020-02-28 12:18:17.487755] [0x00007f6ae37fe700] [info]    [OpenAuto] [AndroidAutoEntity] start.
[2020-02-28 12:18:17.487772] [0x00007f6ae37fe700] [info]    [OpenAuto] [AudioInputService] start.
[2020-02-28 12:18:17.487837] [0x00007f6ae37fe700] [info]    [OpenAuto] [AudioService] start, channel: MEDIA_AUDIO
[2020-02-28 12:18:17.487853] [0x00007f6ae37fe700] [info]    [OpenAuto] [AudioService] start, channel: SPEECH_AUDIO
[2020-02-28 12:18:17.487860] [0x00007f6ae37fe700] [info]    [OpenAuto] [AudioService] start, channel: SYSTEM_AUDIO
[2020-02-28 12:18:17.487867] [0x00007f6ae37fe700] [info]    [OpenAuto] [SensorService] start.
[2020-02-28 12:18:17.487876] [0x00007f6ae37fe700] [info]    [OpenAuto] [VideoService] start.
[2020-02-28 12:18:17.487888] [0x00007f6ae37fe700] [info]    [OpenAuto] [BluetoothService] start.
[2020-02-28 12:18:17.487896] [0x00007f6ae37fe700] [info]    [OpenAuto] [InputService] start.
[2020-02-28 12:18:17.493011] [0x00007f6ae8abc700] [info]    [OpenAuto] [AndroidAutoEntity] version response, version: 1.5, status: 0
[2020-02-28 12:18:17.493021] [0x00007f6ae8abc700] [info]    [OpenAuto] [AndroidAutoEntity] Begin handshake.
[2020-02-28 12:18:17.518106] [0x00007f6ae8abc700] [info]    [OpenAuto] [AndroidAutoEntity] Handshake, size: 2282
[2020-02-28 12:18:17.519463] [0x00007f6ae8abc700] [info]    [OpenAuto] [AndroidAutoEntity] continue handshake.
[2020-02-28 12:18:17.530676] [0x00007f6ae3fff700] [info]    [OpenAuto] [AndroidAutoEntity] Handshake, size: 51
[2020-02-28 12:18:17.530758] [0x00007f6ae3fff700] [info]    [OpenAuto] [AndroidAutoEntity] Auth completed.
[2020-02-28 12:18:17.535437] [0x00007f6ae37fe700] [info]    [OpenAuto] [AndroidAutoEntity] Discovery request, device name: Android, brand: Google Pixel 3 XL
[2020-02-28 12:18:17.535466] [0x00007f6ae37fe700] [info]    [OpenAuto] [AudioInputService] fill features.
[2020-02-28 12:18:17.535477] [0x00007f6ae37fe700] [info]    [OpenAuto] [AudioService] fill features, channel: MEDIA_AUDIO
[2020-02-28 12:18:17.535485] [0x00007f6ae37fe700] [info]    [OpenAuto] [AudioService] fill features, channel: SPEECH_AUDIO
[2020-02-28 12:18:17.535491] [0x00007f6ae37fe700] [info]    [OpenAuto] [AudioService] fill features, channel: SYSTEM_AUDIO
[2020-02-28 12:18:17.535497] [0x00007f6ae37fe700] [info]    [OpenAuto] [SensorService] fill features.
[2020-02-28 12:18:17.535505] [0x00007f6ae37fe700] [info]    [OpenAuto] [VideoService] fill features.
[2020-02-28 12:18:17.535513] [0x00007f6ae37fe700] [info]    [OpenAuto] [BluetoothService] fill features
[2020-02-28 12:18:17.535520] [0x00007f6ae37fe700] [info]    [OpenAuto] [InputService] fill features.```

Phone does nothing and it just sits on the main OpenAuto screen

[75 more replies]

closed 2021-11-27 by matt2005