opencardev/crankshaft #407: camera question

yop16, 2020-03-22:
hello im a question.
How conect this camera in my raspberrry for a crankshaft?

https://ibb.co/g6y0kBz

tsukisan, 2020-03-22:
Hi Yop16,

That camera is not easy (maybe impossible) for Raspberry pi. The camera you listed provides an analog signal which would need a converter to USB for the Pi. After that you would need a way to integrate that into the software yourself. It would be much simpler to use a USB webcam type of camera as the Pi/Crankshaft are setup for that already.

Sorry it's not simple to my knowledge, maybe someone else will have experience with a camera like this.

Bishamon1987, 2020-03-23:
Not totally true.
I have 2 analog cameras.
I use 2 videograbbers.
But not all the chipsets are supported.
One is arkmicro and the other fuchia007 or something like that.
It depends on the type of signal it gives.

egisz, 2020-07-30:
@Bishamon1987 what chipset is arkmicro grabber? 
Mine is this: 
```
#pi@raspberrypi:/home $ lsusb
Bus 001 Device 009: ID 18ec:5555 Arkmicro Technologies Inc.
```
I'm strugling to make it work on Crankshaft... Cam overlay does not work.

Ramortada, 2023-02-05:
Hello, @matt2005 . Thank you for your great work.

 Is it possible to use this video grabber listed above with cam_overlay and crankshaft ?

Geekyadz, 2024-06-23:
> Hi Yop16,
>
> That camera is not easy (maybe impossible) for Raspberry pi. The camera you listed provides an analog signal which would need a converter to USB for the Pi. After that you would need a [...]

Seems pretty simple enough to me: https://youtu.be/QF2X8LUEzLY