opencardev/crankshaft #408: Custom Splash Screen not working

Basline-LineTech, 2020-04-11:
If you need support: Please post on https://www.reddit.com/r/crankshaft/ as this is a bug tracker, not a support forum.

Note: Reporting issues not based on official published builds will be closed without any comment.

- What phone do you have?
Xiaomi Redmi Note 7

- What OS version?
Android 9

- What crankshaft version?
Lastest

- Steps to reproduce the bug, if you're planning to report a bug. Please indicate whether the bug is always repoducible.
Power on the pi

- Please provide any further information that you might find helpful if available.
I tried to replace all the default and custom 'splash' and 'shutdown' files to my custom splash screen, but nothing worked. The startup and shutdown still shows only the logs instead to my splash screen

matt2005, 2020-10-01:
Please retry with latest build

BodoMinea, 2021-01-02:
This still seems to be an issue as of the latest release. I am running on CRANKSHAFT-NG 5.4.72-v7l+ #1356 (2020-10-26 / Build f8d7ba3), and no matter what I select in settings (Animation/Classic/Custom under the Boot Screen section of the CS Base tab), I only get the text-based log boot and shutdown sequence. I have tried with both the default splash/shutdown images and my own, both with dev mode enabled and disabled. The result is always the same - text boot only.

Is this a known issue on your side or is there anything I could try? Maybe enabling the graphic boot from a configuration file and not the GUI, if you could point me to it. Thanks.

matt2005, 2021-01-03:
I haven't  tried it. I'm currently working on a few core issues at the moment.

jakka351, 2021-07-27:
You can work around by using omxplayer in a service set to play video on boot

hjaltioj, 2021-09-24:
> You can work around by using omxplayer in a service set to play video on boot

How do you change to omxplayer? :)

matt2005, 2021-11-27:
closing as same as #411

closed 2021-11-27 by matt2005