opencardev/crankshaft #409: Crankshaft NG randomly shuts down or screen goes grey

rixis19, 2020-05-10:
- What phone do you have? Irrelevant

- What OS version? Irrelevant

- What crankshaft version? E2B068B

When using crankshaft in under a minute after it boots, it was suddenly start saving settings and bluetooth settings, and shuts off completely. Or, when going into settings, it will instead show a screen of grey lines/color. I haven't had the chance to connect my phone yet, so it cannot be related, all I have managed to do is turn on BT search upon boot, and tried to change ignition pins. I'm unsure of what else to say or provide, as it posts but stays on no long enough for me to do anything. It isn't undervolted.

Here is what happens when it crashes(?) in settings: 
[image]

robert5974, 2020-05-11:
You could try changing from EGL to X for the windowing system under the video tab but idk for sure.

gcharlie, 2020-05-14:
do you have any gpio hats on the pi, or power hungry usb devices? even if you have a pretty beefy power supply, the Raspberry Pi. Is only going to take in (i think) 3amps thru the 5v rail; which may or may not be enough for the board and all its peripherals. this was happening in my setup up until i moved all the usb devices to a externally powered hub, and fed power to the screen from a separate 12v buck convertor.

Basline-LineTech, 2020-05-16:
Have you disabled the screen timeout settings?

closed 2020-09-09 by rixis19