opencardev/crankshaft #411: Animation during boot up

arkendu, 2020-06-26:
Is there any way we can add an animation while Crankshaft boots up? will be really look sleek. 

If you need support: Please post on https://www.reddit.com/r/crankshaft/ as this is a bug tracker, not a support forum.

Note: Reporting issues not based on official published builds will be closed without any comment.

- What phone do you have?

- What OS version?

- What crankshaft version?

- Steps to reproduce the bug, if you're planning to report a bug. Please indicate whether the bug is always repoducible.

- Please provide any further information that you might find helpful if available.

- Starting with pre4: Boot into dev mode (can be selected in settings).
  Reproduce the possible bug. Create the debug logs by pushing the grey button on bottom of screen.
  Add the created file (/boot/debug.zip) to the issue.

  Warning: if you have configured wifi passwords they will be blanked by script during creation.
  We can't make absolutly sure you passwords are removed so do a personal check of the files inside the zip.

matt2005, 2020-10-03:
I'm looking into this in the latest release.

Jpower2023, 2023-07-04:
i found a solution for the custom splash which was mentioned in this thread https://github.com/opencardev/crankshaft/issues/408.
To get it work you need copy the png's manually in "/usr/share/plymouth/theme/custom" dictonary. After that you need to update the initramfs and reboot