opencardev/crankshaft #412: I've created a image release built from my fork [Crackshaft-ng on Buster](https://github.com/matt2005/crankshaft/releases/tag/v.1.0.0.20200214-alpha) in case anyone can't build it their self

EVPower2k13, 2020-07-27:
Thank matt2005 for the build. It's working on my new raspberry pi 4.

My Samsung S9 seem to work when plug with the USB. The wifi android is not working.
The raspberry is connected to my cell and the android auto is not appering on the screen. So  using android auto wireless is not working. Can you please check it out?

My setting are all good because i used my old raspberry pi 3 and it work that way.

For now will use the cable but without cable will be awesome.

Thanks.

htruong, 2020-07-28:
Thanks @EVPower2k13,

As I cannot possibly take care of this repo due to various reasons and has been neglecting to update this repo for a while, I wonder if you (or someone who reads this) are interested in stepping up to maintain this project? Please email me htruong@tnhh.net

Thanks,
- Huan.

egisz, 2020-07-28:
Hi, I 100% support idea to have one primary repo, as now it's getting decentralized and unclear whitch fork to use. Thumbs up!

SuperUserockx, 2020-07-29:
I can also contribute in testing and bug fixing as well

matt2005, 2020-09-19:
I'm up for that, @htruong how should we proceed?

matt2005, 2020-10-01:
I've updated with my fixes so closing issue.
Please re-open a new issue if still not working with new release

closed 2020-10-01 by matt2005

EVPower2k13, 2020-10-15:
Hi everyone, sorry for the delay. I was really busy. I will not be able to continue on the project because I just sold my car and my new one has android auto. Thanks all for your effort for older car that need a good system like yours.