opencardev/crankshaft #417: Failed to start Crankshaft PulseAudio system server.

marcimix, 2020-10-03:
Hi! :)
I trying to run Crankshaft on the qemu but I got plenty errors
Anyone tried run virtual machine besides rpi3?
Thank you!

I use kubuntu 20.04 and I have only old rpi1b+ at my house. Still wonder if my newbie code run just for fun. 
Once again thank you! :+1: 

on the splash screen:
[image]

there is my dmesg of running  the img file:

````
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd034]
[    0.000000] Linux version 5.4.69-v8+ (dom@buildbot) (gcc version 9.3.0 (Ubuntu 9.3.0-10ubuntu1)) #1344 SMP PREEMPT Fri Oct 2 16:06:51 BST 2020
[    0.000000] Machine model: Raspberry Pi 3 Model B
[    0.000000] earlycon: pl11 at MMIO 0x000000003f201000 (options '')
[    0.000000] printk: bootconsole [pl11] enabled
[    0.000000] efi: Getting EFI parameters from FDT:
[    0.000000] efi: UEFI not found.
[    0.000000] cma: Reserved 8 MiB at 0x000000003b800000
[    0.000000] On node 0 totalpages: 245760
[    0.000000]   DMA zone: 3840 pages used for memmap
[    0.000000]   DMA zone: 0 pages reserved
[    0.000000]   DMA zone: 245760 pages, LIFO batch:63
[    0.000000] percpu: Embedded 31 pages/cpu s89304 r8192 d29480 u126976
[    0.000000] pcpu-alloc: s89304 r8192 d29480 u126976 alloc=31*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Detected VIPT I-cache on CPU0
[    0.000000] CPU features: detected: ARM erratum 845719
[    0.000000] CPU features: detected: ARM erratum 843419
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 241920
[    0.000000] Kernel command line: rw earlycon=pl011,0x3f201000 console=ttyAMA0 loglevel=8 root=/dev/mmcblk0p2 fsck.repair=yes net.ifnames=0 rootwait memtest=1
[    0.000000] Dentry cache hash table entries: 131072 (order: 8, 1048576 bytes, linear)
[    0.000000] Inode-cache hash table entries: 65536 (order: 7, 524288 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 937556K/983040K available (9852K kernel code, 1124K rwdata, 3436K rodata, 1024K init, 1202K bss, 37292K reserved, 8192K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 32872 entries in 129 pages
[    0.000000] rcu: Preemptible hierarchical RCU implementation.
[    0.000000] rcu:     RCU restricting CPUs from NR_CPUS=256 to nr_cpu_ids=4.
[    0.000000]  Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 25 jiffies.
[    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=4
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] random: get_random_bytes called from start_kernel+0x320/0x4b0 with crng_init=0
[    0.000000] arch_timer: cp15 timer(s) running at 62.50MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x1cd42e208c, max_idle_ns: 881590405314 ns
[    0.000111] sched_clock: 56 bits at 62MHz, resolution 16ns, wraps every 4398046511096ns
[    0.008809] Console: colour dummy device 80x25
[    0.010801] Calibrating delay loop (skipped), value calculated using timer frequency.. 125.00 BogoMIPS (lpj=250000)
[    0.011284] pid_max: default: 32768 minimum: 301
[    0.012446] LSM: Security Framework initializing
[    0.015388] Mount-cache hash table entries: 2048 (order: 2, 16384 bytes, linear)
[    0.015640] Mountpoint-cache hash table entries: 2048 (order: 2, 16384 bytes, linear)
[    0.036444] Disabling memory control group subsystem
[    0.060474] ASID allocator initialised with 32768 entries
[    0.061802] rcu: Hierarchical SRCU implementation.
[    0.066155] EFI services will not be available.
[    0.069326] smp: Bringing up secondary CPUs ...
[    0.072983] Detected VIPT I-cache on CPU1
[    0.073646] CPU1: Booted secondary processor 0x0000000001 [0x410fd034]
[    0.079171] Detected VIPT I-cache on CPU2
[    0.079364] CPU2: Booted secondary processor 0x0000000002 [0x410fd034]
[    0.082186] Detected VIPT I-cache on CPU3
[    0.082317] CPU3: Booted secondary processor 0x0000000003 [0x410fd034]
[    0.082945] smp: Brought up 1 node, 4 CPUs
[    0.084463] SMP: Total of 4 processors activated.
[    0.084764] CPU features: detected: 32-bit EL0 Support
[... 602 lines cut ...]
         Starting Crankshaft PulseAudio system server...
[  OK  ] Started Crankshaft Startup Service.
         Starting Crankshaft User startup Service...
[  OK  ] Started dhcpcd on all interfaces.
[  OK  ] Reached target Network.
         Starting Permit User Sessions...
         Starting OpenBSD Secure Shell server...
         Starting Crankshaft Wifi Monitor Service...
[  OK  ] Started Permit User Sessions.
[  OK  ] Started OpenBSD Secure Shell server.
[  OK  ] Started Crankshaft User startup Service.
         Starting /etc/rc.local Compatibility...
[  OK  ] Started Crankshaft Day Mode Service.
[  OK  ] Started /etc/rc.local Compatibility.
         Starting LSB: set CPUFreq kernel parameters...
[  OK  ] Started Crankshaft Update Check.
[  OK  ] Reached target Timers.
         Starting Terminate Plymouth Boot Screen...
         Starting Hold until boot process finishes up...
         Starting Crankshaft Bluetooth Service...

Welcome to Crankshaft CarOS (2019-08-08 / Build E2B068B)

CRANKSHAFT-NG login: [  148.234071] reboot: Restarting system
```

matt2005, 2020-10-03:
Are you using the latest release?
Please detail exactly how you are running it and I'll try to replicate.

marcimix, 2020-10-03:
> Are you using the latest release?
I use 2019-08-08 build version cause I got more progress on dmesg if i try put newest release I got mess on log:
```
[  OK  ] Reached target Network.
ERROR: ld.so: object '\>�' from /etc/ld.so.preload cannot be preloaded (cannot open shared object file): ignored.
�I͇�▒ȏ�d>�>xj��5��*▒����.D�CRf' from /etc/ld.so.preload cannot be preloaded (cannot open shared object file): ignored.
ERROR: ld.so: object '�r*d▒Ћ\��O�f��I��' from /etc/ld.so.preload cannot be preloaded (cannot open shared object file): ignored.
ERROR: ld.so: object 'dqW�<@8a��֩pW��▒Ey���Mvw' from /etc/ld.so.preload cannot be preloaded (cannot open shared object file): ignored.
��m�����JŊE�P�fI܈y�' from /etc/ld.so.preload cannot be preloaded (cannot open shared object file): ignored.
ERROR: ld.so: object '��T-X�ե��~�ciwmP�o�Žb��y▒<�ʆ��f}���ùh�-W4�z' from /etc/ld.so.preload cannot be preloaded (cannot open shared object file): ignored.
ERROR: ld.so: object '��1��6�g_�����2ōV��.���.uTC��6w���zC}c�"X��|dG愨6#��n��G����' from /etc/ld.so.preload cannot be preloaded (cannot open shared object file): ignored.
ERROR: ld.so: object '\>�' from /etc/ld.so.preload cannot be preloaded (cannot open shared object file): ignored.
�I͇�▒ȏ�d>�>xj��5��*▒����.D�CRf' from /etc/ld.so.preload cannot be preloaded (cannot open shared object file): ignored.
ERROR: ld.so: object '�r*d▒Ћ\��O�f��I��' from /etc/ld.so.preload cannot be preloaded (cannot open shared object file): ignored.
ERROR: ld.so: object 'dqW�<@8a��֩pW��▒Ey���Mvw' from /etc/ld.so.preload cannot be preloaded (cannot open shared object file): ignored.
��m�����JŊE�P�fI܈y�' from /etc/ld.so.preload cannot be preloaded (cannot open shared object file): ignored.
ERROR: ld.so: object '��T-X�ե��~�ciwmP�o�Žb��y▒<�ʆ��f}���ùh�-W4�z' from /etc/ld.so.preload cannot be preloaded (cannot open shared object file): ignored.
ERROR: ld.so: object '��1��6�g_�����2ōV��.���.uTC��6w���zC}c�"X��|dG愨6#��n��G����' from /etc/ld.so.preload cannot be preloaded (cannot open shared object file): ignored.
You are in emergency mode. After logging in, type "journalctl -xb" to view
system logs, "systemctl reboot" to reboot, "systemctl default" or "exit"
to boot into default mode.
ERROR: ld.so: object '\>�' from /etc/ld.so.preload cannot be preloaded (cannot open shared object file): ignored.
�I͇�▒ȏ�d>�>xj��5��*▒����.D�CRf' from /etc/ld.so.preload cannot be preloaded (cannot open shared object file): ignored.
ERROR: ld.so: object '�r*d▒Ћ\�O�f��I��' from /etc/ld.so.preload cannot be preloaded (cannot open shared object file): ignored.
ERROR: ld.so: object 'dqW�<@8a��֩pW��▒Ey���Mvw' from /etc/ld.so.preload cannot be preloaded (cannot open shared object file): ignored.
��m�����JŊE�P�fI܈y�' from /etc/ld.so.preload cannot be preloaded (cannot open shared object file): ignored.
ERROR: ld.so: object '��T-X�ե��~�ciwmP�o�Žb��y▒<�ʆ��f}���ùh�-W4�z' from /etc/ld.so.preload cannot be preloaded (cannot open shared object file): ignored.
ERROR: ld.so: object '��1��6�g_�����2ōV��.���.uTC��6w���zC}c�"X��|dG愨6#��n��G����' from /etc/ld.so.preload cannot be preloaded (cannot open shared object file): ignored.
Give root password for maintenance
(or press Control-D to continue): 

```
> Please detail exactly how you are running it and I'll try to replicate.
https://github.com/dhruvvyas90/qemu-rpi-kernel/issues/100 it's a @guysoft script from last comment. 

finger crossed for scp :)
Thank you!

matt2005, 2020-10-03:
the latest release should work in qemu as that's how I built it.
I've a made a quick typo fix so there will be a new release in the next few days

matt2005, 2021-02-10:
Does the latest release work in QEMU?
The Audio error is due to PA not finding a MIC.

closed 2021-11-27 by github-actions[bot]