opencardev/crankshaft #418: autoapp crashes on Rpi4, latest image CarOS (2020-10-03 / Build )

egisz, 2020-10-05:
I believe autoapp is looking for older libraries (1.62, while system has newer 1.67).

If you need support: Please post on https://www.reddit.com/r/crankshaft/ as this is a bug tracker, not a support forum.

Note: Reporting issues not based on official published builds will be closed without any comment.

- What phone do you have?
Samsung S10

- What OS version?
Android 10

- What crankshaft version?
 CarOS (2020-10-03 / Build ), from releases page.

- Steps to reproduce the bug, if you're planning to report a bug. Please indicate whether the bug is always repoducible.
Boot system.

- Please provide any further information that you might find helpful if available.

```
pi@CRANKSHAFT-NG:~ $ /usr/local/bin/autoapp
/usr/local/bin/autoapp: error while loading shared libraries: libboost_system.so.1.62.0: cannot open shared object file: No such file or directory

pi@CRANKSHAFT-NG:~ $ ldd /usr/local/bin/autoapp
        linux-vdso.so.1 (0xbec62000)
        /usr/lib/arm-linux-gnueabihf/libarmmem-${PLATFORM}.so => /usr/lib/arm-linux-gnueabihf/libarmmem-v7l.so (0xb6faf000)
        libboost_system.so.1.62.0 => not found
        libboost_log.so.1.62.0 => not found
        libboost_unit_test_framework.so.1.62.0 => not found
        libboost_date_time.so.1.62.0 => not found
        libboost_log_setup.so.1.62.0 => not found
        libboost_filesystem.so.1.62.0 => not found
        libboost_thread.so.1.62.0 => not found
        libboost_regex.so.1.62.0 => not found
        libboost_chrono.so.1.62.0 => not found
        libboost_atomic.so.1.62.0 => not found
        libpthread.so.0 => /lib/arm-linux-gnueabihf/libpthread.so.0 (0xb6f83000)
        libQt5MultimediaWidgets.so.5 => /usr/local/qt5/lib/libQt5MultimediaWidgets.so.5 (0xb6f5f000)
        libQt5Bluetooth.so.5 => /usr/local/qt5/lib/libQt5Bluetooth.so.5 (0xb6e6b000)
        libusb-1.0.so.0 => /lib/arm-linux-gnueabihf/libusb-1.0.so.0 (0xb6e46000)
        libprotobuf.so.10 => not found
        libbcm_host.so => /opt/vc/lib/libbcm_host.so (0xb6e1c000)
        libvcos.so => /opt/vc/lib/libvcos.so (0xb6e03000)
        libvchiq_arm.so => /opt/vc/lib/libvchiq_arm.so (0xb6deb000)
        librtaudio.so.5 => not found
        libtag.so.1 => /lib/arm-linux-gnueabihf/libtag.so.1 (0xb6d28000)
        libblkid.so.1 => /lib/arm-linux-gnueabihf/libblkid.so.1 (0xb6cd0000)
        libaasdk_proto.so => /usr/local/lib/libaasdk_proto.so (0xb6ae5000)
        libaasdk.so => /usr/local/lib/libaasdk.so (0xb68aa000)
        libQt5Multimedia.so.5 => /usr/local/qt5/lib/libQt5Multimedia.so.5 (0xb67b4000)
        libQt5Network.so.5 => /usr/local/qt5/lib/libQt5Network.so.5 (0xb661f000)
        libQt5Widgets.so.5 => /usr/local/qt5/lib/libQt5Widgets.so.5 (0xb60fb000)
        libQt5Gui.so.5 => /usr/local/qt5/lib/libQt5Gui.so.5 (0xb5b59000)
        libQt5Core.so.5 => /usr/local/qt5/lib/libQt5Core.so.5 (0xb5630000)
        libstdc++.so.6 => /lib/arm-linux-gnueabihf/libstdc++.so.6 (0xb54e7000)
        libm.so.6 => /lib/arm-linux-gnueabihf/libm.so.6 (0xb5465000)
        libgcc_s.so.1 => /lib/arm-linux-gnueabihf/libgcc_s.so.1 (0xb5438000)
        libc.so.6 => /lib/arm-linux-gnueabihf/libc.so.6 (0xb52ea000)
        /lib/ld-linux-armhf.so.3 (0xb6fc4000)
        libpulse-mainloop-glib.so.0 => /lib/arm-linux-gnueabihf/libpulse-mainloop-glib.so.0 (0xb52d6000)
        libpulse.so.0 => /lib/arm-linux-gnueabihf/libpulse.so.0 (0xb5280000)
        libglib-2.0.so.0 => /lib/arm-linux-gnueabihf/libglib-2.0.so.0 (0xb5170000)
        libQt5OpenGL.so.5 => /usr/local/qt5/lib/libQt5OpenGL.so.5 (0xb511e000)
        libGLESv2.so.2 => /lib/arm-linux-gnueabihf/libGLESv2.so.2 (0xb50ff000)
        libQt5Concurrent.so.5 => /usr/local/qt5/lib/libQt5Concurrent.so.5 (0xb50e9000)
        libQt5DBus.so.5 => /usr/local/qt5/lib/libQt5DBus.so.5 (0xb5055000)
        libudev.so.1 => /lib/arm-linux-gnueabihf/libudev.so.1 (0xb5025000)
        libdl.so.2 => /lib/arm-linux-gnueabihf/libdl.so.2 (0xb5012000)
        librt.so.1 => /lib/arm-linux-gnueabihf/librt.so.1 (0xb4ffb000)
        libicuuc.so.63 => /lib/arm-linux-gnueabihf/libicuuc.so.63 (0xb4e75000)
        libz.so.1 => /lib/arm-linux-gnueabihf/libz.so.1 (0xb4e4a000)
        libuuid.so.1 => /lib/arm-linux-gnueabihf/libuuid.so.1 (0xb4e33000)
        libprotobuf.so.17 => /lib/arm-linux-gnueabihf/libprotobuf.so.17 (0xb4bcf000)
        libboost_system.so.1.67.0 => /lib/arm-linux-gnueabihf/libboost_system.so.1.67.0 (0xb4bbb000)
        libboost_log.so.1.67.0 => /lib/arm-linux-gnueabihf/libboost_log.so.1.67.0 (0xb4ae9000)
        libboost_unit_test_framework.so.1.67.0 => /lib/arm-linux-gnueabihf/libboost_unit_test_framework.so.1.67.0 (0xb4a30000)
        libboost_date_time.so.1.67.0 => /lib/arm-linux-gnueabihf/libboost_date_time.so.1.67.0 (0xb4a10000)
        libboost_log_setup.so.1.67.0 => /lib/arm-linux-gnueabihf/libboost_log_setup.so.1.67.0 (0xb4978000)
        libboost_filesystem.so.1.67.0 => /lib/arm-linux-gnueabihf/libboost_filesystem.so.1.67.0 (0xb4951000)
        libboost_thread.so.1.67.0 => /lib/arm-linux-gnueabihf/libboost_thread.so.1.67.0 (0xb491e000)
        libboost_regex.so.1.67.0 => /lib/arm-linux-gnueabihf/libboost_regex.so.1.67.0 (0xb4827000)
        libboost_chrono.so.1.67.0 => /lib/arm-linux-gnueabihf/libboost_chrono.so.1.67.0 (0xb4810000)
        libboost_atomic.so.1.67.0 => /lib/arm-linux-gnueabihf/libboost_atomic.so.1.67.0 (0xb47fd000)
        libssl.so.1.1 => /lib/arm-linux-gnueabihf/libssl.so.1.1 (0xb477e000)
        libcrypto.so.1.1 => /lib/arm-linux-gnueabihf/libcrypto.so.1.1 (0xb4565000)
        libpng16.so.16 => /lib/arm-linux-gnueabihf/libpng16.so.16 (0xb4529000)
        libdouble-conversion.so.1 => /lib/arm-linux-gnueabihf/libdouble-conversion.so.1 (0xb4503000)
        libicui18n.so.63 => /lib/arm-linux-gnueabihf/libicui18n.so.63 (0xb42b5000)
        libicudata.so.63 => /lib/arm-linux-gnueabihf/libicudata.so.63 (0xb28b7000)
        libgthread-2.0.so.0 => /lib/arm-linux-gnueabihf/libgthread-2.0.so.0 (0xb28a5000)
        libpulsecommon-12.2.so => /usr/lib/arm-linux-gnueabihf/pulseaudio/libpulsecommon-12.2.so (0xb2824000)
        libcap.so.2 => /lib/arm-linux-gnueabihf/libcap.so.2 (0xb280f000)
        libdbus-1.so.3 => /lib/arm-linux-gnueabihf/libdbus-1.so.3 (0xb27be000)
        libpcre.so.3 => /lib/arm-linux-gnueabihf/libpcre.so.3 (0xb2747000)
        libGLdispatch.so.0 => /lib/arm-linux-gnueabihf/libGLdispatch.so.0 (0xb26c8000)
        libatomic.so.1 => /lib/arm-linux-gnueabihf/libatomic.so.1 (0xb26ad000)
        libboost_timer.so.1.67.0 => /lib/arm-linux-gnueabihf/libboost_timer.so.1.67.0 (0xb2699000)
        libX11-xcb.so.1 => /lib/arm-linux-gnueabihf/libX11-xcb.so.1 (0xb2687000)
        libX11.so.6 => /lib/arm-linux-gnueabihf/libX11.so.6 (0xb2565000)
        libxcb.so.1 => /lib/arm-linux-gnueabihf/libxcb.so.1 (0xb2536000)
        libICE.so.6 => /lib/arm-linux-gnueabihf/libICE.so.6 (0xb2510000)
        libSM.so.6 => /lib/arm-linux-gnueabihf/libSM.so.6 (0xb24f9000)
        libXtst.so.6 => /lib/arm-linux-gnueabihf/libXtst.so.6 (0xb24e4000)
        libsystemd.so.0 => /lib/arm-linux-gnueabihf/libsystemd.so.0 (0xb2447000)
        libwrap.so.0 => /lib/arm-linux-gnueabihf/libwrap.so.0 (0xb242d000)
        libsndfile.so.1 => /lib/arm-linux-gnueabihf/libsndfile.so.1 (0xb23b1000)
        libasyncns.so.0 => /lib/arm-linux-gnueabihf/libasyncns.so.0 (0xb239c000)
        libXau.so.6 => /lib/arm-linux-gnueabihf/libXau.so.6 (0xb2389000)
        libXdmcp.so.6 => /lib/arm-linux-gnueabihf/libXdmcp.so.6 (0xb2374000)
        libbsd.so.0 => /lib/arm-linux-gnueabihf/libbsd.so.0 (0xb234c000)
        libXext.so.6 => /lib/arm-linux-gnueabihf/libXext.so.6 (0xb232d000)
        libXi.so.6 => /lib/arm-linux-gnueabihf/libXi.so.6 (0xb230f000)
        liblzma.so.5 => /lib/arm-linux-gnueabihf/liblzma.so.5 (0xb22de000)
        liblz4.so.1 => /lib/arm-linux-gnueabihf/liblz4.so.1 (0xb22b2000)
        libgcrypt.so.20 => /lib/arm-linux-gnueabihf/libgcrypt.so.20 (0xb21d6000)
        libnsl.so.1 => /lib/arm-linux-gnueabihf/libnsl.so.1 (0xb21b2000)
        libFLAC.so.8 => /lib/arm-linux-gnueabihf/libFLAC.so.8 (0xb2152000)
        libogg.so.0 => /lib/arm-linux-gnueabihf/libogg.so.0 (0xb213d000)
        libvorbis.so.0 => /lib/arm-linux-gnueabihf/libvorbis.so.0 (0xb2106000)
        libvorbisenc.so.2 => /lib/arm-linux-gnueabihf/libvorbisenc.so.2 (0xb2074000)
        libresolv.so.2 => /lib/arm-linux-gnueabihf/libresolv.so.2 (0xb2050000)
        libgpg-error.so.0 => /lib/arm-linux-gnueabihf/libgpg-error.so.0 (0xb2026000)
```

- Starting with pre4: Boot into dev mode (can be selected in settings).
  Reproduce the possible bug. Create the debug logs by pushing the grey button on bottom of screen.
  Add the created file (/boot/debug.zip) to the issue.

  Warning: if you have configured wifi passwords they will be blanked by script during creation.
  We can't make absolutly sure you passwords are removed so do a personal check of the files inside the zip.

matt2005, 2020-10-05:
i'll be recompiling the binaries after add a couple of fixes this week.

matt2005, 2020-10-11:
I've rebuilt the prebuilts can you try before I promote to the opencardev repo. [Please try newer prebuilts](https://github.com/opencardev/crankshaft/issues/422#issuecomment-706664254)

matt2005, 2020-10-15:
Please try this release [alpha 5.5](https://github.com/opencardev/crankshaft/releases/tag/csng-alpha5.5)

matt2005, 2020-10-16:
Now working

closed 2020-10-16 by matt2005