opencardev/crankshaft #420: Openauto won't start because it has screen problems.

Yokosa, 2020-10-05:
If you need support: Please post on https://www.reddit.com/r/crankshaft/ as this is a bug tracker, not a support forum.

Note: Reporting issues not based on official published builds will be closed without any comment.

- What phone do you have?
Xiaomi Mi A2 Lite

- What OS version?
Android 10
- What crankshaft version?
crankshaft-csng-alpha53
- Steps to reproduce the bug, if you're planning to report a bug. Please indicate whether the bug is always repoducible.
Boot system
- Please provide any further information that you might find helpful if available.

Openauto won't start because it has screen problems. I have the same result on rpi3b and rpi4b. I'm using the waveshare 7 display c. https://www.waveshare.com/wiki/7inch_HDMI_LCD_(C)

```
lsusb from rpi4:
pi@CRANKSHAFT-NG:~ $ lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

lsusb from rpi3:
Bus 001 Device 004: ID 0eef:0005 D-WAV Scientific Co., Ltd
Bus 001 Device 003: ID 0424:ec00 Standard Microsystems Corp. SMSC9512/9514 Fast Ethernet Adapter
Bus 001 Device 002: ID 0424:9514 Standard Microsystems Corp. SMC9514 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

`Oct  5 15:30:25 CRANKSHAFT-NG systemd-modules-load[104]: Inserted module 'exfat'
Oct  5 15:30:25 CRANKSHAFT-NG systemd-sysctl[119]: Couldn't write '10' to 'kernel-panic', ignoring: No such file or directory
Oct  5 15:30:25 CRANKSHAFT-NG fake-hwclock[107]: Mon  5 Oct 15:30:24 UTC 2020
Oct  5 15:30:25 CRANKSHAFT-NG kernel: [    0.000000] Booting Linux on physical CPU 0x0
Oct  5 15:30:25 CRANKSHAFT-NG kernel: [    0.000000] Linux version 5.4.51-v7l+ (dom@buildbot) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611)) #1333 SMP Mon Aug 10 16:51:40 BST 2020
Oct  5 15:30:25 CRANKSHAFT-NG systemd-fsck[133]: e2fsck 1.44.5 (15-Dec-2018)
Oct  5 15:30:25 CRANKSHAFT-NG kernel: [    0.000000] CPU: ARMv7 Processor [410fd083] revision 3 (ARMv7), cr=30c5383d
Oct  5 15:30:25 CRANKSHAFT-NG kernel: [    0.000000] CPU: div instructions available: patching division code
Oct  5 15:30:25 CRANKSHAFT-NG kernel: [    0.000000] CPU: PIPT / VIPT nonaliasing data cache, PIPT instruction cache
Oct  5 15:30:25 CRANKSHAFT-NG kernel: [    0.000000] OF: fdt: Machine model: Raspberry Pi 4 Model B Rev 1.2
Oct  5 15:30:25 CRANKSHAFT-NG systemd-fsck[133]: Superblock last mount time is in the future.
Oct  5 15:30:25 CRANKSHAFT-NG kernel: [    0.000000] Memory policy: Data cache writealloc
Oct  5 15:30:25 CRANKSHAFT-NG kernel: [    0.000000] Reserved memory: created CMA memory pool at 0x000000001ec00000, size 256 MiB
Oct  5 15:30:25 CRANKSHAFT-NG kernel: [    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
Oct  5 15:30:25 CRANKSHAFT-NG kernel: [    0.000000] On node 0 totalpages: 966656
Oct  5 15:30:25 CRANKSHAFT-NG kernel: [    0.000000]   DMA zone: 2304 pages used for memmap
Oct  5 15:30:25 CRANKSHAFT-NG kernel: [    0.000000]   DMA zone: 0 pages reserved
Oct  5 15:30:25 CRANKSHAFT-NG systemd-fsck[133]: #011(by less than a day, probably due to the hardware clock being incorrectly set)
Oct  5 15:30:25 CRANKSHAFT-NG kernel: [    0.000000]   DMA zone: 196608 pages, LIFO batch:63
Oct  5 15:30:25 CRANKSHAFT-NG kernel: [    0.000000]   HighMem zone: 770048 pages, LIFO batch:63
Oct  5 15:30:25 CRANKSHAFT-NG kernel: [    0.000000] percpu: Embedded 20 pages/cpu s49856 r8192 d23872 u81920
Oct  5 15:30:25 CRANKSHAFT-NG kernel: [    0.000000] pcpu-alloc: s49856 r8192 d23872 u81920 alloc=20*4096
Oct  5 15:30:25 CRANKSHAFT-NG kernel: [    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
Oct  5 15:30:25 CRANKSHAFT-NG kernel: [    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 964352
Oct  5 15:30:25 CRANKSHAFT-NG kernel: [    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=1 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 snd_bcm2835.enable_headphones=1 video=HDMI-A-1:1024x600M@60 smsc95xx.macaddr=DC:A6:32:5F:36:92 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=tty3 root=/dev/mmcblk0p2 ro rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait quiet logo.nologo loglevel=0 vt.global_cursor_default=0 noswap splash plymouth.ignore-serial-consoles consoleblank=0 ipv6.disable=1
Oct  5 15:30:25 CRANKSHAFT-NG kernel: [    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes, linear)
Oct  5 15:30:25 CRANKSHAFT-NG kernel: [    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes, linear)
Oct  5 15:30:25 CRANKSHAFT-NG kernel: [    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
Oct  5 15:30:25 CRANKSHAFT-NG kernel: [    0.000000] software IO TLB: mapped [mem 0x17cc0000-0x1bcc0000] (64MB)
Oct  5 15:30:25 CRANKSHAFT-NG kernel: [    0.000000] Memory: 3473460K/3866624K available (10240K kernel code, 725K rwdata, 2740K rodata, 2048K init, 853K bss, 131020K reserved, 262144K cma-reserved, 3080192K highmem)
Oct  5 15:30:25 CRANKSHAFT-NG kernel: [    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
Oct  5 15:30:25 CRANKSHAFT-NG kernel: [    0.000000] ftrace: allocating 30051 entries in 59 pages
Oct  5 15:30:25 CRANKSHAFT-NG kernel: [    0.000000] rcu: Hierarchical RCU implementation.
Oct  5 15:30:25 CRANKSHAFT-NG kernel: [    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
Oct  5 15:30:25 CRANKSHAFT-NG kernel: [    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
Oct  5 15:30:25 CRANKSHAFT-NG kernel: [    0.000000] GIC: Using split EOI/Deactivate mode
Oct  5 15:30:25 CRANKSHAFT-NG kernel: [    0.000000] random: get_random_bytes called from start_kernel+0x340/0x518 with crng_init=0
Oct  5 15:30:25 CRANKSHAFT-NG kernel: [    0.000004] sched_clock: 32 bits at 1000kHz, resolution 1000ns, wraps every 2147483647500ns
Oct  5 15:30:25 CRANKSHAFT-NG kernel: [    0.000011] clocksource: timer: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 1911260446275 ns
[... 884 lines cut ...]
Oct  5 15:37:18 CRANKSHAFT-NG systemd[1]: Created slice User Slice of UID 1000.
Oct  5 15:37:18 CRANKSHAFT-NG systemd[1]: Starting User Runtime Directory /run/user/1000...
Oct  5 15:37:18 CRANKSHAFT-NG systemd[1]: Started User Runtime Directory /run/user/1000.
Oct  5 15:37:18 CRANKSHAFT-NG systemd[1]: Starting User Manager for UID 1000...
Oct  5 15:37:18 CRANKSHAFT-NG systemd[2154]: gpgconf: running /usr/bin/gpg-agent failed (exitcode=2): General error
Oct  5 15:37:18 CRANKSHAFT-NG systemd[2154]: gpgconf: fatal error (exit status 1)
Oct  5 15:37:18 CRANKSHAFT-NG service_openauto.sh[1225]: Could not queue DRM page flip on screen HDMI1 (Invalid argument)
Oct  5 15:37:18 CRANKSHAFT-NG systemd[2154]: Listening on GnuPG network certificate management daemon.
Oct  5 15:37:18 CRANKSHAFT-NG systemd[2154]: Listening on Sound System.
Oct  5 15:37:18 CRANKSHAFT-NG systemd[2154]: Listening on GnuPG cryptographic agent and passphrase cache.
Oct  5 15:37:18 CRANKSHAFT-NG systemd[2154]: Listening on GnuPG cryptographic agent (ssh-agent emulation).
Oct  5 15:37:18 CRANKSHAFT-NG systemd[2154]: Reached target Timers.
Oct  5 15:37:18 CRANKSHAFT-NG systemd[2154]: Listening on GnuPG cryptographic agent and passphrase cache (restricted).
Oct  5 15:37:18 CRANKSHAFT-NG systemd[2154]: Reached target Paths.
Oct  5 15:37:18 CRANKSHAFT-NG systemd[2154]: Listening on GnuPG cryptographic agent and passphrase cache (access for web browsers).
Oct  5 15:37:18 CRANKSHAFT-NG systemd[2154]: Starting D-Bus User Message Bus Socket.
Oct  5 15:37:18 CRANKSHAFT-NG systemd[2154]: Listening on D-Bus User Message Bus Socket.
Oct  5 15:37:18 CRANKSHAFT-NG systemd[2154]: Reached target Sockets.
Oct  5 15:37:18 CRANKSHAFT-NG systemd[2154]: Reached target Basic System.
Oct  5 15:37:18 CRANKSHAFT-NG systemd[2154]: Reached target Default.
Oct  5 15:37:18 CRANKSHAFT-NG systemd[2154]: Startup finished in 188ms.
Oct  5 15:37:18 CRANKSHAFT-NG systemd[1]: Started User Manager for UID 1000.
Oct  5 15:37:18 CRANKSHAFT-NG systemd[1]: Started Session c1 of user pi.
Oct  5 15:37:19 CRANKSHAFT-NG service_openauto.sh[1225]: Could not queue DRM page flip on screen HDMI1 (Invalid argument)
`

matt2005, 2020-10-05:
Have you modified the config.txt? If so please provide a copy.

Yokosa, 2020-10-06:
Here is my config.txt:

```
# For more options and information see
# http://rpf.io/configtxt
# Some settings may impact device functionality. See link above for details

# uncomment if you get no picture on HDMI for a default "safe" mode
#hdmi_safe=1

# uncomment this if your display has a black border of unused pixels visible
# and your display can output without overscan
#disable_overscan=1

# uncomment the following to adjust overscan. Use positive numbers if console
# goes off screen, and negative if there is too much border
#overscan_left=16
#overscan_right=16
#overscan_top=16
#overscan_bottom=16

# uncomment to force a console size. By default it will be display's size minus
# overscan.
#framebuffer_width=1280
#framebuffer_height=720

# uncomment if hdmi display is not detected and composite is being output
#hdmi_force_hotplug=1

# uncomment to force a specific HDMI mode (this will force VGA)
#hdmi_group=1
#hdmi_mode=1

# uncomment to force a HDMI mode rather than DVI. This can make audio work in
# DMT (computer monitor) modes
#hdmi_drive=2

# uncomment to increase signal to HDMI, if you have interference, blanking, or
# no display
#config_hdmi_boost=4

# uncomment for composite PAL
#sdtv_mode=2

#uncomment to overclock the arm. 700 MHz is the default.
#arm_freq=800

# Uncomment some or all of these to enable the optional hardware interfaces
#dtparam=i2c_arm=on
#dtparam=i2s=on
#dtparam=spi=on

# Uncomment this to enable infrared communication.
#dtoverlay=gpio-ir,gpio_pin=17
#dtoverlay=gpio-ir-tx,gpio_pin=18

# Additional overlays and parameters are documented /boot/overlays/README

# Enable audio (loads snd_bcm2835)
dtparam=audio=on

[pi4]
# Enable DRM VC4 V3D driver on top of the dispmanx display stack
dtoverlay=vc4-fkms-v3d
max_framebuffers=2

[all]
dtoverlay=vc4-fkms-v3d

# Custom power settings
max_usb_current=1

# Disable the PWR LED.
dtparam=pwr_led_trigger=none
dtparam=pwr_led_activelow=off

# Disable Rainbow splash
disable_splash=1

# GPU Mem
gpu_mem=256

# Overscan fix
overscan_scale=1

# Enable watchdog
dtparam=watchdog=on

# Boot time improvements
boot_delay=0
initial_turbo=30
start_cd=1
#dtoverlay=sdtweak,overclock_50=100

# Bluetooth
dtoverlay=pi3-disable-bt

hdmi_group=2
hdmi_mode=87
hdmi_cvt 1024 600 60 6 0 0 0
hdmi_drive=1
```

I haven't had such a problem yet. Previous crankshaft versions would normally detect the screen under hdmi, so would a clean buster.
i saw on the internet that it's a qt problem
https://www.raspberrypi.org/forums/viewtopic.php?t=252614

matt2005, 2020-10-06:
I'm currently rebuilding at, as I see the same issue.
If you comment out the extra hdmi lines in the config you will be able to boot successfully however the screen will be the incorrect size and the touch button will not match.
```text
#hdmi_group=2
#hdmi_mode=87
#hdmi_cvt 1024 600 60 6 0 0 0
#hdmi_drive=1
```
Please confirm this is the case.

Yokosa, 2020-10-06:
@matt2005 Yes, you are right, when I commented on these links, the application launched. Various stripes and characters appear on the screen at startup, then the application starts correctly but there is also a problem with subtitles, etc.

[image]

matt2005, 2020-10-06:
Thanks, I'm in the process of rebuilding Qt. I'm hopeing to have a fix soon.

matt2005, 2020-10-11:
I've rebuilt the prebuilts can you try before I promote to the opencardev repo. [Please try newer prebuilts](https://github.com/opencardev/crankshaft/issues/422#issuecomment-706664254)

matt2005, 2020-10-15:
Please try this release [alpha 5.5](https://github.com/opencardev/crankshaft/releases/tag/csng-alpha5.5)
Download and write the image to sd.
Edit the config.txt to add the following to the bottom of the file
```
hdmi_group=2
hdmi_mode=87
hdmi_cvt 1024 600 60 6 0 0 0
hdmi_drive=1
```

Yokosa, 2020-10-16:
In the latest version alpha 5.5 everything works on my waveshare display. Thanks!

closed 2020-10-16 by Yokosa