opencardev/crankshaft #421: aasdk fix not present in latest build (alpha53)

egisz, 2020-10-08:
Hi,

Tested latest build, but it seems that this fix is not incorporated: 
https://github.com/opencardev/aasdk/commit/76294c53517dbde5126e2b73422698da8892b2fe
Rotate buttons do not work.

I haven't figured 100% how build process works, but I believe prebuilds repo is out of date.
this repo is pointing to [prebuilds](https://github.com/matt2005/prebuilts/tree/cef40eca2bac23ccf33e138d3ea199838e7bcffb) which autoapp is 8 months old...

matt2005, 2020-10-10:
Yeah prebuilts out of date, which has caused several issues. I'm in process of rebuilding the prebuilts. I had a few issues with the builds so am about to update them.

matt2005, 2020-10-11:
I've rebuilt the prebuilts can you try before I promote to the opencardev repo. [Please try newer prebuilts](https://github.com/opencardev/crankshaft/issues/422#issuecomment-706664254)

matt2005, 2020-10-15:
Resolved in Alpha 5.5 [alpha 5.5](https://github.com/opencardev/crankshaft/releases/tag/csng-alpha5.5)